MarlinFirmware/Marlin #5620: Z height issue after Firmware Flash after Auto Level kit install

JIPSTERR, 2016-12-28:
Hello,

I just installed the Auto Level Kit on my printer. Afterwards I flashed my 3DP08 to 3DP12.

My printer is http://www.hic3dprinter.com/productshow.asp?ArticleID=0&id=25&cid=001

Now my Z axis height will no longer go all the way up. In the menu it shows it went up 200 but in actuality it only goes up about 3 inches max.
This is ruining prints and I need this fixed asap as I am now having a backlog of prints.

Can someone please help me?

Tannoo, 2016-12-28:
Steps/mm are incorrect for the Z axis.

JIPSTERR, 2016-12-28:
yeah, sounds about right. When I move 1mm at a time X & Y move properly and quickly, Z does not.

Do you know the fix?

Tannoo, 2016-12-29:
Configuration.h, line 489

I don't know how to figure out what you need aside from trial an error.

JIPSTERR, 2016-12-29:
I'm using a Hictop Prusa i3. Do you know a program illiterate nerd would go about modifying the .cpp.hex files I was provided. Additionally, I used Cura to flash the firmware.

I greatly appreciate the assistance you've been providing me.

Tannoo, 2016-12-29:
Not me... I can only count to 10.

JIPSTERR, 2016-12-30:
Hahaha well damn :-)

Tannoo, 2016-12-30:
Does that printer have an ATMEGA2560 and a RAMPS 1.4 board on it?

Tannoo, 2016-12-30:
- Download the Marlin firmware here: https://github.com/MarlinFirmware/Marlin.git
- Then extract it to it's own folder.
- Download and install the latest Arduino IDE (1.8.0 at the time of this writing): https://www.arduino.cc/en/Main/Software

- Open the Arduino IDE and open the `Marlin.ino` file in the zip archive you extracted earlier.
- Then select the `configuration.h` tab at the top of the IDE. This is the first config file for options.
- Once your are done with that one, select the `configuration_adv.h` tab and go through that one. This is the second config file for more options.

When you are done with those, connect a USB cable from your computer to the controller on your printer. Then click the "arrow" (upload) at the top of the IDE.

There is a sanity check that will happen (coded in the firmware) to let you know if you need to enable or disable some options. Because not everything plays well together.

There are other guides on the internet that will help with getting the proper a basic settings to get the printer moving and spitting.

This is not decoding HEX files. This is starting from something you can configure at will like the rest of us.

Tannoo, 2016-12-30:
Doing it this way will be easier to "adjust" your firmware in any way you need to.

Tannoo, 2016-12-30:
You could install a HEX editor and find the steps/mm in both of the firmware HEXs you got and change the new one to match the old one.

That is and will be a pita, but possible.

JIPSTERR, 2016-12-30:
[image]
[image]

I don't know the board info but hopefully these pictures might be of help. Let me know if you need a pic of the back of them or anything specific to look for. 

I'm looking thru the configuration.h file from time to time when I'm able here at work.

JIPSTERR, 2016-12-30:
Here's the Z axis max height for visual

[image]
[image]

Roxy-3D, 2016-12-30:
Part of the license agreement for Marlin is this:  A commercial entity (the company you bought your printer from) is allowed to use (and even modify) Marlin and ship it embedded in a 'For Profit' product.   But this privilege comes at the cost of providing THE FULL SOURCE CODE of what they embedded and shipped to the customers of that product.

My suggestion would be to ask the manufacturer for the source code of what was loaded on your printer when it shipped.   Using that Configuration.h file...   You can cross all of those old settings into the new Configuration.h file for RC-8 (Or even UBL!!!) and get running on a current code base.

JIPSTERR, 2016-12-30:
Yeah, having the file from them to just re-flash would be great! I'm lazy. I've contacted them multiple times, commented on their YouTube video regarding the Auto Level moments after they uploaded it and sent them the link of my YouTube video of the issue....aaannnnnd nothing. It'll probably be a few weeks until they reply.

And just for shiggles and because I'm slightly bored and almost off lunch, here's that video
https://www.youtube.com/watch?v=99BZSuG2ui4﻿

Tannoo, 2016-12-31:
Well, lazy has nothing to do with it. Sounds like you are more impatient.

JIPSTERR, 2016-12-31:
Haha. Not lazy, they need to know that what they're pushing out to their customers is breaking printers

JIPSTERR, 2016-12-31:
*not more important

Tannoo, 2017-01-03:
Did you get the files from the manufacturer, or are you going to start from scratch?

I can give you my config files to use as a base starting point.

Tannoo, 2017-01-03:
I just don't know what board that is.

JIPSTERR, 2017-01-03:
They have yet to get back to me with the proper files.

FHeilmann, 2017-01-03:
Its a Hictop MPX board. Looks very similar to a MKS Base to me. The printer itself looks like a clone similar to the Migbot Prusa i3 (for which ample discussion exists on the internet)

Tannoo, 2017-01-04:
I found this:

http://reprap.org/wiki/HICTOP_Prusa_i3

lrpirlet, 2017-01-06:
@FHeilmann. If only Z calibration is wrong, have a look at http://reprap.org/wiki/Triffid_Hunter's_Calibration_Guide to calculate the number of step per mm... 

Then use G92 Z(result)... (no need to recompile anything, nor request sources...)

**hint** the result should be an integer for a prusa I3 with a metric Z threaded rod....

lrpirlet, 2017-01-06:
@JIPSTERR @FHeilmann 
Sorry, I got it all wrong please disregard previous entry... (overwritten)

If only Z calibration is wrong, have a look at http://reprap.org/wiki/Triffid_Hunter's_Calibration_Guide to calculate the number of step per mm...

Then use **M92 Z(result)** (no need to recompile anything, nor request sources...)

**Hint:** the result should be an integer for a prusa I3 with a metric Z threaded rod...

Enjoy

lrpirlet, 2017-01-06:
Later, if that M92 Zvalue works AND if it does not stay across reset (reboot), then you'll need to recompile with that value in the configuration.h (that would mean that eeprom was not enabled)

JIPSTERR, 2017-01-11:
They just replied with the files broken down. Here's the section starting at 489.

//// MOVEMENT SETTINGS
#define NUM_AXIS 4 // The axis order in all axis related arrays is X, Y, Z, E
#define HOMING_FEEDRATE {50*60, 50*60, 4*60, 0}  // set the homing speeds (mm/min)

// default settings

#define DEFAULT_AXIS_STEPS_PER_UNIT   {80,80,398,94.4962144}  // default steps per unit for ultimaker //78.7402, 78.7402,1070,865.88
#define DEFAULT_MAX_FEEDRATE          {500, 500, 5, 25}    // (mm/sec)
#define DEFAULT_MAX_ACCELERATION      {9000,9000,100,10000}    // X, Y, Z, E maximum start speed for accelerated moves. E default values are good for skeinforge 40+, for older versions raise them a lot.

#define DEFAULT_ACCELERATION          3000    // X, Y, Z and E max acceleration in mm/s^2 for printing moves
#define DEFAULT_RETRACT_ACCELERATION  3000   // X, Y, Z and E max acceleration in mm/s^2 for retracts

Tannoo, 2017-01-13:
Did they give you the full zip file of the firmware?

JIPSTERR, 2017-01-13:
they did indeed (finally). They sent it to be checked out/fixed. How would one go about uploading the puppy?

Tannoo, 2017-01-13:
Same as I stated earlier in this post except use the files they gave you instead of downloading the latest Marlin.

Unzip the files into their own directory.
- Open the Arduino IDE and open the `Marlin.ino` file in the zip archive you extracted earlier.
- Then select the `configuration.h` tab at the top of the IDE. This is the first config file for options.
- Once your are done with that one, select the `configuration_adv.h` tab and go through that one. This is the second config file for more options.

When you are done with those, connect a USB cable from your computer to the controller on your printer.

1. Set the board you have "Arduino/Genuino Mega or Mega 2560"
2. Set the processor you have "ATmega2560 (Mega 2560)"
3. Set the active com port for the connected printer.
4. Click the "arrow" (upload) at the top of the IDE.

DannyRStreet, 2017-03-24:
Did you solve your issue?
I see by the pictures you have a MPX 1.3 HICTOP board.
Go to Thingivers and down load the code for your printer and install with Cura.
http://www.thingiverse.com/thing:1743047

closed 2018-02-06 by thinkyhead