MarlinFirmware/Marlin #5621: SAFE_HOMING Y axis not moving

lukeskymuh, 2016-12-28:
In RC8 BugFix the Y axis remains at the endstop when G28 is called instead of moving to the centre:
my marlin is attached. I only changed the configuration.h file
[Marlin-RCBugFix.zip](https://github.com/MarlinFirmware/Marlin/files/676430/Marlin-RCBugFix.zip)
.

Blue-Marlin, 2016-12-29:
Check your endstop configuration with M119.

lukeskymuh, 2016-12-29:
Works as it should: 

when nothing is pressed:
21:21:39.269 : Reporting endstop status
21:21:39.269 : x_min: open
21:21:39.269 : y_min: open
21:21:39.269 : z_min: open

when I press the y endstop:
21:20:48.236 : Reporting endstop status
21:20:48.236 : x_min: open
21:20:48.239 : y_min: TRIGGERED
21:20:48.239 : z_min: open

I have tested X and Z also.

lukeskymuh, 2016-12-29:
The configuration.h file I used was originallly for the RC8 (not Bug Fix)

lukeskymuh, 2016-12-29:
Issue is solved. My stepper current was to low.

closed 2016-12-29 by lukeskymuh