MarlinFirmware/Marlin #5622: Bilinear auto bed leveling height different than the z probe height

Froop666, 2016-12-29:
Hello everyone,
I have an problem with the Z-Probe and the bilinear auto bed leveling:
When i do the z Probe with an G28 I get a perfect height, but after the ABL the nozzle is 2-3mm over the maximum point of the bed.
I don't know if thats an bug or an configuration error?

[Configuration.h](https://github.com/MarlinFirmware/Marlin/files/676434/Configuration.h.txt)

Best regards,
Niclas

brainscan, 2016-12-29:
#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.2+1.6
Is this related?

Froop666, 2016-12-29:
It has the same effect as "#define Z_PROBE_OFFSET_FROM_EXTRUDER -0,6" or am I wrong?

manianac, 2017-01-01:
Just set it to -0.6 and try it.  I would bet thats your problem

Roxy-3D, 2017-01-02:
> Just set it to -0.6 and try it. I would bet thats your problem

Do this...  But don't forget to do an M502 and M500 also!

Froop666, 2017-01-02:
I tried this, but the problem stays the same :(

Whytehorse, 2017-01-17:
I'm affected by this bug too. On a delta with nearly perfect manual leveling, G29 returns the following:
Bilinear Leveling Grid:
      0     1     2     3     4     5     6     7     8
 0 +0.68 +0.93 +1.07 +0.55 +0.53 +0.49 +0.70 +0.24 -0.33
 1 +0.49 +0.64 +0.50 +0.55 +0.64 +0.50 +0.22 +0.05 -0.26
 2 +0.46 +0.43 +0.55 +0.64 +0.71 +0.63 +0.42 +0.10 -0.08
 3 +0.43 +0.53 +0.64 +0.69 +0.76 +0.75 +0.58 +0.36 +0.12
 4 +0.42 +0.51 +0.65 +0.71 +0.79 +0.85 +0.78 +0.62 +0.47
 5 +0.42 +0.50 +0.63 +0.72 +0.78 +0.85 +0.88 +0.81 +0.69
 6 +0.37 +0.42 +0.55 +0.67 +0.75 +0.79 +0.85 +0.87 +0.84
 7 +0.29 +0.37 +0.40 +0.49 +0.61 +0.65 +0.71 +0.79 +0.91
 8 +0.25 +0.51 +0.67 +0.28 +0.33 +0.49 +0.74 +0.78 +0.84

Changing z pro offset via m851 z0.5 haqs no effect. At the very last point in the g29 test, it is about 0.5mm too high. At the center it's about 0.8mm high. Tower A=0.3mm high. Tower B=-0.1mm low, Tower C=0.5mm high

thinkyhead, 2017-04-02:
Sorry there hasn't been more attention to this. How's it going?

Whytehorse, 2017-04-09:
I haven't tried the G29 since. Perhaps if there was some documentation as to what all the numbers mean I could troubleshoot it further. I've been using a manual calibration here: http://escher3d.com/pages/wizards/wizarddelta.php
and it works pretty good but I have to do it hundreds of times to get it  perfect. I'm now down to adjustments in the 0.005mm range and have probably reached the limits of that method. I still get first layer problems as it doesn't account for the bed being a little off.

thinkyhead, 2017-04-12:
The values reported by bilinear leveling represent the Z height at each probed point. The closer to zero the better. There's a new feature being added called "auto-calibration." It isn't super well-documented yet, but it should help you get all your delta parameters as close to correct as possible and deal with any curvature of the bed plane. Come on over to #6187 and give it a test to see if it helps.

closed 2017-10-31 by thinkyhead