MarlinFirmware/Marlin #5623: Print wont start after heating and display stops responding.

Yamomz, 2016-12-29:
I just finished my first build and i am about to try mi first print. I start the print, then hot end reach the set temperature then moves to the center to start printing, still a message saying Heating... (Hot end 180/180 Bed 0/17 room temperature i don't want to use the heated bed) Min bed temp is 5. Build corexy D-Bot marlin 11 RC7

brainscan, 2016-12-29:
Have you tried with RCBugFix? This may have already been corrected.

Yamomz, 2016-12-29:
Nope but as it seems was noob me and the lack of G29 at beginning of the code. Now sometimes auto bed leveling routine freeze marlin on point 2. I have to see if it only happens when using octapi or sd. Where do i get RCBugFix?

brainscan, 2016-12-29:
There's a drop down where you can switch between branches on GitHub, RCBugFix is the most up to date so if the problem is still there it's more likely to get attention than if you're using an older RC

closed 2017-03-04 by thinkyhead