MarlinFirmware/Marlin #5625: when the "DELTA_PRINTABLE_RADIUS" is bigger then 181

athlon789789, 2016-12-29:
The RC8 is working good on my small delta printer (DIA=250mm) .thanks for the author. but when I upload it to my big printer(DIA 400MM) ,the G29 can't work at all the printers doesnt move atall . later I find out the  DELTA_PRINTABLE_RADIUS cant set over 181MM I dont why,is there any one face the same issue?

when the DELTA_PRINTABLE_RADIUS setting over 181mm the arduino IDE report:

In file included from sketch\MarlinConfig.h:37:0,

                 from sketch\Marlin.h:36,

                 from sketch\Marlin_main.cpp:220:

sketch\Marlin_main.cpp: In function 'bool position_is_reachable(float*, bool)':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:99:19: warning: integer overflow in expression [-Woverflow]

 #define sq(x) ((x)*(x))

                   ^

sketch\Marlin_main.cpp:2923:30: note: in expansion of macro 'sq'

     return HYPOT2(dx, dy) <= sq(DELTA_PRINTABLE_RADIUS);

                              ^

In file included from sketch\MarlinConfig.h:37:0,

                 from sketch\Marlin.h:36,

                 from sketch\ultralcd.h:26,

                 from sketch\ultralcd.cpp:23:

sketch\ultralcd.cpp: In function 'void _lcd_move_xyz(const char*, AxisEnum)':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:99:19: warning: integer overflow in expression [-Woverflow]

 #define sq(x) ((x)*(x))

                   ^

sketch\ultralcd.cpp:1424:22: note: in expansion of macro 'sq'

           max = sqrt(sq(DELTA_PRINTABLE_RADIUS) - sq(current_position[Y_AXIS - axis]));

                      ^

Blue-Marlin, 2016-12-29:
Are you confusing radius and diameter?
Your 250mm diameter bed has a radius of 125mm.

Blue-Marlin, 2016-12-29:
Try 182 __.0__ !

asyring, 2017-01-03:
you can undefine the sq macro from the Arduino.h file.

`#undef sq(x)
#define sq(x) ((float)(x)*(float)(x))`

thinkyhead, 2017-02-12:
Our best solution is to make sure to cast defined values to `float` directly in the code that uses it. If used in `sq` then it should be `sq((float)VAL)`.

closed 2017-02-12 by thinkyhead