MarlinFirmware/Marlin #5626: Configuration RCBugfix

Schraubendreher, 2016-12-29:
Hello,

I would like to ask you something because I am slowly desperate and seek help.
I have problems configuring the Marlin RC8.
My Hardware:
- Board ICstep01 Atmega 2560
For actuation extruders
Pins.h is present and integrated
Http://shop.in-circuit.de/product_info.php?products_id=72

- 5 pcs power amplifiers TB6600 V1.2
For motor control 24V
Axis X two
Axis Y
Axis Z two
Https://en.aliexpress.com/item/CNC-Router-3-Axis-Kit-TB6600-3-Axis-Stepper-Motor-Driver-Controller-kit-4-5A-mach3/32291286690.html?scm= 1007.13338.69414.000000000000000 & pvid = e53e93ef-6481-458c-82ea-21d17d717497 & tpp = 1 & detailNewVersion = & categoryId = 100007158

My problem is the control of the motor drivers. Signals Enable, Step and DIR are tapped from the board.

The LED of the output stages flashes and the motors do not move.

Can someone help me? Thank you very much

Blue-Marlin, 2016-12-29:
Do you have activated
```
// Enable this option for Toshiba steppers
//#define CONFIG_STEPPERS_TOSHIBA
```
?

Schraubendreher, 2016-12-29:
I have just tried but it's all still the same

Schraubendreher, 2016-12-29:
I have light hum of engines, the engines want to start but do not move.

Schraubendreher, 2016-12-29:
this I would add,
Version RC4 is the last version that still works.

Blue-Marlin, 2016-12-29:
A bug in calculating accelerations has been fixed since RC4.
Try lower values for acceleration.

Schraubendreher, 2016-12-29:
/**
 * Default Max Feed Rate (mm/s)
 * Override with M203
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_FEEDRATE          { 30, 30, 5, 25 }
/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 300, 300, 100, 10000 }
/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          300    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  300    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   300    // X, Y, Z acceleration for travel (non printing) moves 
/**
 * Default Jerk (mm/s)
 *
 * "Jerk" specifies the minimum speed change that requires acceleration.
 * When changing speed and direction, if the difference is less than the
 * value set here, it may happen instantaneously.
 */
#define DEFAULT_XJERK                 9.6
#define DEFAULT_YJERK                 9.6
#define DEFAULT_ZJERK                  0.1
#define DEFAULT_EJERK                  5.0

I tried it, does nothing,sorry

Schraubendreher, 2016-12-29:
It is so that the motor drivers do not get a DIR signal.
As said, the RC4 version works.

Request for feedback

Schraubendreher, 2017-01-03:
Please configure external drivers tb6600 
http://www.progdron.com/ru/arduino-shield/arduino-shield/306-drajver-shagovogo-dvigatelya-tb6600-dc12-45v

in Configuration:
// For Inverting Stepper Enable Pins (Active Low) use 0, Non Inverting (Active High) use 1
// :{0:'Low',1:'High'}
#define X_ENABLE_ON 1
#define Y_ENABLE_ON 1
#define Z_ENABLE_ON 1

and

in Configuration_adv.h:
// By default pololu step drivers require an active high signal. However, some high power drivers require an active low signal as step.
#define INVERT_X_STEP_PIN true
#define INVERT_Y_STEP_PIN true
#define INVERT_Z_STEP_PIN true

now it works.

thinkyhead, 2017-02-12:
Issue resolved?

closed 2017-02-12 by thinkyhead