MarlinFirmware/Marlin #5636: LIN_Advance Support Question

psavva, 2016-12-30:
I have enabled LIN_Advance, and it's absolutely awesome to see such a big leap in the quality improvements in my test prints.

@Sebastianv650 I need some advice on the K value.
I started at 75 and worked my way up to 200.

I can't seem to get a good value for my printer. 
I have built a custom printer called the WeldingRod Bot.  You can find it on thingiverse...

It has a long Bowden tube with a RepRapDiscount Hexagon Hotend.

Here are some photos of my test prints.

[image]

K=75
[image]
[image]
[image]

K=90
[image]

[image]

[image]

K=150
[image]
[image]
[image]

K=200
[image]
[image]
[image]

Lin_Advance disabled
[image]
[image]
[image]

Yoda printed with Lin_Advance enabled at K=90
[image]
[image]
[image]

Sebastianv650, 2017-01-01:
Do you get a nice clean cube when you print it slow without LIN_ADVANCE? Just to be sure your esteps/mm value is good.

It's interesting to see that the difference between the K values seems to be not that clear on a bowden system. There might be other effects inside the tube (friction due to buckling and others) that are not covered by the advance algorithm.
Based on the given pictures, I would say the cubes look nicer with increasing K value, where K=200 is the best (compare corners and gaps in top infill). I would try another one with 300 and 400, just to see what will happen.

m0n5t3r, 2017-01-03:
I had a similar problem with underextrusion and it was because of E-steps being slightly off after moving from direct drive to bowden

Also, you may have to go higher, depending on the filament: using RC from a few days ago, on a ~60cm bowden the formula in Configuration_adv.h gives me 528 for ngen and 282 for PLA; 528 is definitely too much for ngen, the corners come out rather rounded at 60mm/s, and if I increase the speed (say, 150%, so 90mm/s in theory) they don't look like corners any more 

I am running with 350 now when I print ngen, and it seems to work fine at low-ish speeds (60-70mm/s); for flexible filaments I disable it completely (same for weird ones like {brass,bronze}fill or wood).

psavva, 2017-01-03:
I will firstly try with a slow speed print a cube.
Say around 30mm/s should be ok? 

I'll try adjust jerk and acceleration to get the best results. 

My esteps for my extruder are given by the manufacturer of the drive gear.
http://www.reprap.me/mk8.html

Sebastianv650, 2017-01-04:
@psavva 20-30mm/s for perimeters and top/bottom solid infill should be great to see if you printer is in a well calibrated state (without ADVANCE-things). For K calibration prints, I recommend a medium-high print speed. Sadly that's hard to give in numbers, because it depends on your extruder and hot end. On most printers, it's around 8mm³/s - you see it depends on the line width and layer height what that means in print speed. For a usual line width of around 0.5mm and 0.25mm layer height that means around maximum 65mm/s.

billyd60, 2017-01-05:
Really stupid question. Does Lin_Advance have to also be enabled in the LCD menu anywhere (like firmware retract) or just uncomment the function in .h and go?

psavva, 2017-01-11:
I managed to get the test cube look very similar to a good print with LIN_ADVANCE, by adjusting acceleration, jerk and max velocities without Lin_Advance

I will retune with Lin_Advance and see if I can get the full awesomeness of if it :)

Thank you for your advice and support.

closed 2017-01-11 by psavva