MarlinFirmware/Marlin #5638: Z axis steps per unit not correct.

kscharf, 2016-12-30:
There seems to be an issue with the new Marlin RC software. I put the exact same values for "DEFAULT_AXIS_STEPS_PER_UNIT" that I had for the current stable version (1.0.x)
#define DEFAULT_AXIS_STEPS_PER_UNIT {63.06,63.48224,2220,200} // default steps per unit.
The X and Y axis end up correct, (commanding two 50mm movements from the home position ends up in the middle of the 200x200 bed), but the Z axis moves TOO FAR. In 1.0.x I tried raising the Z axis (via G-code via repetier host, or from the panel knob and LCD) and it is accurate, I set it to 24 mm high and was able to just fit a 24 mm high object under the nozzle. With the RC version, the Z axis moves about three times too far! What's going on here?
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/679294/Configuration.h.txt)

Blue-Marlin, 2016-12-30:
Your EEPROM is active. Chances are high the printer is not using the configured, but the stored values.
Try if M502 helps.

brainscan, 2016-12-31:
I would try what's already been suggested but on a side point are you planning to use a fixed probe like an induction or capacitance sensor? If so you might need to look at the z offset, you currently have it as +2 rather than -2. If you're using a different method just ignore me.

kscharf, 2016-12-31:
I am using a fixed inductive probe.  The probe is mounted about 2mm higher than the nozzle, but it should fire when the nozzle is 2mm above the bed as it has a 4mm working distance (at least that's how I'm setting it up for starters).  I may have misunderstood the comments in the configuration.h file, but what I want is have the offset of the probe (when it fires) be 2mm BELOW the nozzle.  You may be right that means a negative offset.

I would now consider this issue closed as my problem was due to the RC8 software using the compiled in defaults, while my current 1.0x software was using stored eeprom values.  I had forgotten that I never changed the value for Z steps per unit when I upgraded the printer from 5/16-18 threaded rods to 8mm lead screws.  It seems I only modified this parameter in the eeprom, so I got snookered into entering the wrong value into the RC8 configuration file.  What I don't understand is WHY the old version used the eeprom values even when it was reloaded on top of the RC8 version as I went back and forth to debug this.  Is there a Gcode that will force using the compiled in "defaults"?  I know there is a command to copy the defaults to the eeprom (at least from the LCD panel).

Roxy-3D, 2016-12-31:
If you always want to use the Configuration.h setting, just turn the EEPROM off.

> Is there a Gcode that will force using the compiled in "defaults"?

Yes.  M502 will load the compiled in defaults.  Until you do something to change them, you will be using the defaults.

closed 2017-01-04 by kscharf

Sensemaya, 2017-02-01:
Hello I have the same issue with RC-8 after configuring the firmware with the right data uploading by Arduine IDE the #define DEFAULT_AXIS_STEPS_PER_UNIT upload the wrong data.

RC-8 before uploading "#define DEFAULT_AXIS_STEPS_PER_UNIT   {80.256, 80, 399, 100}  // default steps per unit for Ultimaker"
RC-8 After uploading using pronterface "echo:  "echo:  M92 X80.00 Y80.00 Z4000.00 E500.00
[Marlin 1.1.0-RC7.txt](https://github.com/MarlinFirmware/Marlin/files/745558/Marlin.1.1.0-RC7.txt)
[Marlin 1.1.0-RC8.txt](https://github.com/MarlinFirmware/Marlin/files/745559/Marlin.1.1.0-RC8.txt)

"
Doing the same with other RC ,6, 7, 3, the uploading show the correct data

thinkyhead, 2017-02-12:
@Sensemaya And after `M502` / `M500` ?

Sensemaya, 2017-02-12:
Hello 
I found that Pronterface somehow introduce the extra space in Z axis at print time.
Switch to repetir and all ok now