MarlinFirmware/Marlin #5640: Filament Change in LCD Menu in RC8 resets Marlin Board

matrix3dmachine, 2016-12-31:
Hi,
I'm using RC8 on a Ramps 1.4 board, using a single extruder.  When using the Filament Change Feature on the LCD, the board resets and cancels the current print.  However, if I configure for Dual Extruders:
These are the lines I change in Config.h:
#define MOTHERBOARD BOARD_RAMPS_14_EEB
#define EXTRUDERS 2
Add second temp sensor:
#define TEMP_SENSOR_1 1

Then, Filament Change Feature on LCD works just fine.

Can someone take a look at this bug?

Mike

timeshell, 2016-12-31:
Working with MIke on this we have determined the following:

1.  Also occurs when sending M600 from Pronterface
2.  Tried serial logging from beginning of the gcode_M600 routine.  Doesn't appear to log entry to the routine.
3.  Sending other commands from pronterface don't reset the printer

So, there appears to be an issue specifically with the issuing of M600 command and single extruders specifically.  This issue didn't exist in RCBugFix dated from 2016/11/02.

Tannoo, 2016-12-31:
I use Octopi normally.
I invoked filament change via the LCD and it worked better (flawless, in fact) than the last time I used it (over a year ago).
I also invoked filament change via M600 command in the Octopi terminal screen. Same. No issue....flawless.

I don't use Pronterface or Repetier anymore. I use Octopi due to windows updates have screwed prints soooo many times. My computer is not tied to any printing operations anymore.

Tannoo, 2016-12-31:
A lot of things have changed from RC8 to RC8-RCBugFix. Try the RCBugFix.

matrix3dmachine, 2016-12-31:
OK

Matrix 3D Machines
Mike Steffen
Director of Operations
6302 Crystal Brook Lane
Machesney Park, IL.  61115
815-519-1567
ne3print@gmail.com
www.matrix3dmachines.com <http://www.ne3print.com/>

Tannoo, 2016-12-31:
I am using Arduino IDE 1.8.0

Tannoo, 2016-12-31:
@timeshell, are you trying the exact same gcode file he is trying to print?
@matrix3dmachine, tried printing something else? Checked slicer settings?

Tannoo, 2016-12-31:
@matrix3dmachine, when was the last time that filament change worked for you? What changed? Printer firmware, slicer settings, or Pronterface version?

I created a very small print (60mm circle at .2mm high) for testing printing features. Print time is very short, but does the job as testing things like this. It takes more time to heat things up than to print this circle. lol

Tannoo, 2016-12-31:
BTW, the only time my computer is connected to the printer is during flashes and some menu debugging. Cannot flash through Octopi (Shucks).

timeshell, 2016-12-31:
The RC8 Bugfix ended up solving the problem for him.  Still not specific on why it works for us and not for him.

matrix3dmachine, 2016-12-31:
All, I tried RC8BUGFIX, and that fixes my issue. So, something got fixed or changed. However, if I revert back to RC8, it still resets. Not sure if its worth going back to find out what's wrong. 
Mike

Roxy-3D, 2016-12-31:
> All, I tried RC8BUGFIX, and that fixes my issue. So, something got fixed or changed. However, if I revert back to RC8, it still resets. Not sure if its worth going back to find out what's wrong. 

It is worth it because serious malfunctions like this need to be understood.  Without understanding the root cause, we can't be sure the problem is really fixed.

I don't see any changes to the M600 code between RC-8 and the latest RCBugFix.   That concerns me!

timeshell, 2016-12-31:
As far as we were able to determine, it wasn't even getting to the gcode_m600 routine (we added logging to the beginning of the sub and it didn't log anything).  It seemed to reset between the receiving of the instruction and the routine.

Roxy-3D, 2016-12-31:
That is very strange.   And in fact, once the MCode number is parsed, the M600 is handled the same as other MCodes.   If an M600 really isn't getting to the M600 code, we have something else very sinister happening.   (The MCode selector is just a giant switch() statement with a call to the M600 function.  
```cpp

      #if HAS_BED_PROBE
        case 851:
          gcode_M851();
          break;
      #endif // HAS_BED_PROBE

      #if ENABLED(FILAMENT_CHANGE_FEATURE)
        case 600: //Pause for filament change X[pos] Y[pos] Z[relative lift] E[initial retract] L[later retract distance for removal]
          gcode_M600();
          break;
      #endif // FILAMENT_CHANGE_FEATURE

      #if ENABLED(DUAL_X_CARRIAGE)
        case 605:
          gcode_M605();
          break;
      #endif // DUAL_X_CARRIAGE

      #if ENABLED(LIN_ADVANCE)
        case 905: // M905 Set advance factor.
          gcode_M905();
          break;
      #endif
```

timeshell, 2016-12-31:
Agreed.  I was starting to think it was a glitch in the compiler or corruption somewhere.  We tried a fresh copy of RC8, a working copy of RC8 (based on my working copy), I was actively debugging with Visual Micro.  Couldn't reproduce it.  RC8 just didn't want to work for him.

manianac, 2017-01-09:
Is there any worry about the overhead of Visual Micros debugging?

closed 2017-02-14 by thinkyhead