MarlinFirmware/Marlin #5642: Comment wrong units?

rewolff, 2016-12-31:
In  Configuration.h on line 1420, it says microseconds, but then the example number specifies 300 microseconds (if I accept the specified units), while the comment suggests 300 miliseconds.... 

https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/Configuration.h#L1420

psavva, 2016-12-31:
@rewolff, I didn't get what you're asking. 
The comment specified the units for the variable to be microseconds.

What is the confusion?

mperdue, 2016-12-31:
On line 1421 it says 300ms which would be milliseconds. It should be 300µs for microseconds.

rewolff, 2017-01-01:
To me 300ms sounds like the correct units/timeframe. But I'm not familiar enough with the code and what this part does to make that decisison final. 

Here is that part of the config file. 
```
// Delay (in microseconds) before the next move will start, to give the servo time to reach its target angle.
// 300ms is a good value but you can try less delay.
// If the servo can't reach the requested position, increase it.
#define SERVO_DELAY 300
```
Line 1421 suggests 300ms is a nice value, and then the default value is "300", which according to line 1420 has the units microseconds. If we believe line 1420 and 1421 then the number on line 1423 should be 300000, If we believe line 1420 and 1423, then the "300ms is a good value" is an odd remark if 300 microseconds is proposed as a default. But I suspect that line 1421 and 1423 are correct, and that line 1421 needs to read `Delay (in miliseconds)`

esenapaj, 2017-01-01:
Yes, miliseconds is correct.
https://github.com/MarlinFirmware/Marlin/blob/RC/Marlin/servo.cpp#L313

And, I've submitted a patch. https://github.com/MarlinFirmware/Marlin/pull/5645

closed 2017-01-12 by thinkyhead