MarlinFirmware/Marlin #5644: Manual Mesh Bed levelling hangs

DjDemonD, 2017-01-01:
Using RCbugFix 31-12-16 on a corexy. Migrating from RCBugFix 25-09-16 with no issues.

After selecting "level-bed" from the lcd it homes z then hangs showing "Homing XYZ" on the LCD, nothing on the console, it doesn't get as far as homing x and y. Homing works on this machine with this firmware otherwise, but I was using manual mesh levelling before and now am unable to do so.

[Configuration_adv.h.txt](https://github.com/MarlinFirmware/Marlin/files/680031/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/680032/Configuration.h.txt)

epatel, 2017-01-01:
Duplicate? https://github.com/MarlinFirmware/Marlin/issues/5592

alexxy, 2017-01-02:
Seems yes to me =)

DjDemonD, 2017-01-02:
I'd agree it is and I can close it if you wish. However that issue remains unresolved.

psavva, 2017-01-03:
@DjDemonD please close this issue as a PR is already submitted #5634

closed 2017-01-03 by DjDemonD