MarlinFirmware/Marlin #5651: G29 Probing does not respond to published GCODE

cbitterfield, 2017-01-03:
There are issues:

1. The G29 P5 T V4 does not seem to do anything.
2. The G29 command seems to skip certain points.
3. The G29 Topology Report does not match X,Y Coordinates.

Relevant Settings in firmware:
#define AUTO_BED_LEVELING_BILINEAR

/**
 * Enable detailed logging of G28, G29, M48, etc.
 * Turn on with the command 'M111 S32'.
 * NOTE: Requires a lot of PROGMEM!
 */
#define DEBUG_LEVELING_FEATURE

#if ENABLED(AUTO_BED_LEVELING_LINEAR) || ENABLED(AUTO_BED_LEVELING_BILINEAR)

  // Set the number of grid points per dimension.
  #define ABL_GRID_MAX_POINTS_X 3
  #define ABL_GRID_MAX_POINTS_Y ABL_GRID_MAX_POINTS_X

  // Set the boundaries for probing (where the probe can reach).
  #define LEFT_PROBE_BED_POSITION 40
  #define RIGHT_PROBE_BED_POSITION 200
  #define FRONT_PROBE_BED_POSITION 10
  #define BACK_PROBE_BED_POSITION 165

  // The Z probe minimum outer margin (to validate G29 parameters).
  #define MIN_PROBE_EDGE 40

I spent some time re-configuring my printer to match the configuration.h map.

[image]

On a 3 x 3 probe I would expect the following:
[image]

My Actual bed is configured like this:
[image]

I set M111 to S32 (for logging)
Executed a G28
Then a G29

This results in Test.1 (See attached file)

Running the G29 again but with G29 P5 V4 (I would expect 25 points 5^2)

This results in Test.2

In both cases the G29 skipped the front row (3rd Point) and second row(3rd Point) assuming the front of the printer to be the front row (aka Configuration,h)

<-Bilinear Leveling Grid:
<-      0     1     2
<- 0 -1.05 -0.81 +5.00
<- 1 -0.97 -0.75 +5.00
<- 2 -0.91 -0.83 -0.92
<-

The results imply that the back left is 0,0 when in reality the grid is printed with 0,2 transposed.

While I do appreciate the math problem on this, it creates issues with looking to trouble shoot.

In the case of Test 2, it still only sampled 3 points. 

I am not sure what the topology report should look like but I don't see any difference with the T flag on or off. 
[Test_1.txt](https://github.com/MarlinFirmware/Marlin/files/681710/Test_1.txt)
[Test_2.txt](https://github.com/MarlinFirmware/Marlin/files/681711/Test_2.txt)

manianac, 2017-01-04:
>The G29 P5 T V4 does not seem to do anything.
>#define AUTO_BED_LEVELING_BILINEAR

The `P` parameter is only for the linear bed leveling.  The bilinear grid array is hard-set by the configuration currently, although it wouldn't take much to change this.  As of those post it is a fixed size array, so you can't change the number of points through gcode.  The `T` parameter is useless as well with bilinear, as the grid printout _is_ your topology report.

>The G29 command seems to skip certain points.

It's not skipping points, its just always measuring a 3x3 grid as is set by the configuration.

>The G29 Topology Report does not match X,Y Coordinates.

It looks correct, its saying you have an extremely high point around (200,0) on your bed.  While the table is not printed in the way you expect, it does note where (0,0) is, you just have to follow it from there.

cbitterfield, 2017-01-04:
1. I'd recommend updating the Wiki page on G29..
2. The issue is that the grid report on bilinear implies the 0,0 to back left and the configuration says it is front left. I loved advanced math but shifting the matrix around in your head makes troubleshooting more difficult. (Recommend the bilinear leveling grid look more like the topology map and display using the 0,0 as front left.

manianac, 2017-01-04:
Honestly I think its just printing the array starting from 0, it could easily be swapped to put (0,0) on the last row first column of the printout.  (0,0) is still the front left by the way.

Blue-Marlin, 2017-01-04:
In Marlin_main.cpp
```diff
@@ -2463,10 +2463,11 @@ static void clean_up_after_endstop_or_probe_move() {
         if (x < 10) SERIAL_PROTOCOLCHAR(' ');
         SERIAL_PROTOCOL((int)x);
       }
       SERIAL_EOL;
       for (uint8_t y = 0; y < ABL_GRID_POINTS_VIRT_Y; y++) {
+//      for (uint8_t y = ABL_GRID_POINTS_VIRT_Y-1; y >= 0; y--) {
         if (y < 10) SERIAL_PROTOCOLCHAR(' ');
         SERIAL_PROTOCOL((int)y);
         for (uint8_t x = 0; x < ABL_GRID_POINTS_VIRT_X; x++) {
           SERIAL_PROTOCOLCHAR(' ');
           float offset = bed_level_grid_virt[x][y];
```
you can experiment with the commented line.

Blue-Marlin, 2017-01-04:
In Marlin_main.cpp
```diff
@@ -2430,10 +2430,11 @@ static void clean_up_after_endstop_or_probe_move() {
       if (x < 10) SERIAL_PROTOCOLCHAR(' ');
       SERIAL_PROTOCOL((int)x);
     }
     SERIAL_EOL;
     for (uint8_t y = 0; y < ABL_GRID_MAX_POINTS_Y; y++) {
+    //for (uint8_t y = ABL_GRID_MAX_POINTS_Y-1; y >= 0; y--) {
       if (y < 10) SERIAL_PROTOCOLCHAR(' ');
       SERIAL_PROTOCOL((int)y);
       for (uint8_t x = 0; x < ABL_GRID_MAX_POINTS_X; x++) {
         SERIAL_PROTOCOLCHAR(' ');
         float offset = bed_level_grid[x][y];
```
you could experiment with the commented line.

cbitterfield, 2017-01-04:
My perspective is that there is some code that could be consolidated. One of the issues is that the the origin (0,0) for any printer has at least 5 possibilities: Square (4) round (1). It might be better to make the following code changes:

1. a define that sets the origin; easier that what I did to move the origin.
2. using the G29 (T) flag to provide a common topology report (relative to the defined origin) for all types of bed leveling. 
3. add a debug under leveling that shows the Z correction in real time; 
4. add the feature to bilinear bed leveling to accept the G29 (P) switch
5. Under G29 V1-4 (Output debug, the parameters of the G29. (Number of Points = 4, Topology Report, etc). This would be under each type of bed leveling. (If a switch or flag is not used in that method; output "SWITCH P5 ignored" as the debug method.

Blue-Marlin, 2017-01-04:
Nice plan.
Good luck with programming that.

cbitterfield, 2017-01-04:
I spent a couple of hours today learning GIT hub and forking. "C" isn't that hard. The origin program part, I'll wait until I have a need. The other ones, I think I can do. Also, I forked the documentation so that I could make some updates.

manianac, 2017-01-05:
Don't re-invent the wheel, check out @Roxy-3D's UBL branch and see what she has put in.

cbitterfield, 2017-01-05:
I'll look perhaps, I can integrate it back to the main branch. (in my free time) :)

Blue-Marlin, 2017-01-05:
1. There is no need tho 'code' the origin - its a waste of code. The origin is, for Marlin, at the front left. If not, turn the printer, walk around the printer, or turn it in your mind.
2. Because bilinear (former delta) levelling is different from 3-point-grid-linear-leveling, it has no idealized plane. So no difference to it can exist. For sure i don't want two matrices full of zeros in the log. If it's only about the origin see my comments above. But reading it right, if you have the indices, should not be a bigger problem, even if the front and back row are switched.
3. 'real time'? On a display updated once per second? In the console window of a host - sending how much chars per second?
4. In opposite to the matrix based levellings we can't throw away the measured points away after building the matrix, where all needed information is in. As far as i can see you have to reserve all the RAM needed by the biggest possible grid in advance - even if you use only 3x3 points. For the matrix bsed levelling this is simply done on the stack.
5. I'd be surprised if you can find useful non redundant information not already in the log. 'Number of points is already in (just count or look at the indices). 'Topology Report' - see 1. 
6. You can always produce more output. If that makes sense is an other question. We already are on the way where the amount of text in the prom will be as large as the code. For now we want a minimal system to be usable on the 64K processors. All (new) features have to be as 'code size neutral' as possible when switched off. Hunting for more stability, correctness, speed and lower size should have a much higher priority then new features or 'beautifying' things - at the moment.

Please don't feel personally attacked. It's general tendency i observe since some month. The one wants to to make Marlin a sound system, the other a DMX system, or a weather station, a filament welder, a general purpose CNC-machine, whatever.

Up to now __Marlin is a operating system for 3D-Printers__ running on too slow and too tiny processors to have all possible features at once.

cbitterfield, 2017-01-09:
I get that. Part of the issue was that the documentation needs to be updated some. I have loaded that into my personal GitHub, and I plan on making some contributions. The main issue that lead me down this path was a defective probe.  Oddly, enough that got me interested in the code. I was about to look at a Arduio DUE for the bigger processor; only to find out; No EEPROM and 3.3V only. I am sure that will be a whole other issue using the SD card for what should be in an EEPROM. I am also working with a defunct Solidoodle 4; that has the case in common with the original printer.  My guess is that at somepoint there will need to be a Marlin lite or feature optimization. Let me know if I can help.

cbitterfield, 2017-01-09:
Issue is really related to updated documentation being needed.

closed 2017-01-09 by cbitterfield