MarlinFirmware/Marlin #5652: Z axis homed at end of SD print

kscharf, 2017-01-04:
I always add this gcode to the end of my slices

M104 S0 ; turn off temperature
G28 X0 ; home X axis
G1 Y190 F5000 ; get bed forward
M104 T0 S0 ; ALL heaters off
M140 S0
M84 ; disable motors

... as I want to leave the nozzle at the ending height, but move the bed to a position where I can remove the print.  However the RC8 code seems to add a G28 Z0 to this as the print head is homed at the end of my prints after the G28 X0, G1 Y190 F500, and M104 T0 S0 are executed.  This is BAD, it could cause the X carriage to crash into the side of a large print that extends to the edge of the bed!

kscharf, 2017-01-04:
[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/683644/Configuration.h.txt)

psavva, 2017-01-04:
Do you have the Gcode you used to print with?

kscharf, 2017-01-04:
[3DBenchy.gcode.txt](https://github.com/MarlinFirmware/Marlin/files/685203/3DBenchy.gcode.txt)

here is one example

muhammedozkan, 2017-01-05:
Can you try it?

end gcode

M104 S0 ;extruder heater off
M140 S0 ;heated bed heater off (if you have it)
G91 ;relative positioning
G1 E-3 F300  ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F9000 ;move Z up a bit and retract filament even more
G28 X0 Y0 ;move X/Y to min endstops, so the head is out of the way
M84 ;steppers off
G90 ;absolute positioning

kscharf, 2017-01-05:
This is what I ended up with for my ending gcode that seems to solve my problem:

M104 S0 ; turn off temperature
G92 Z0	;  fake z home
G1 Z5	; raise Z up a bit
G28 X0  ; home X axis
G1 Y190 F5000 ; get bed forward
M104 T0 S0 ; ALL heaters off
M140 S0
M84     ; disable motors

This moves the bed in a position where I can remove the part (actually at the far end of home for the Y axis and home at the X axis).  It also leaves the Z axis raised a fraction above the final print position.  Without the G92 and G1 commands the firmware would home the Z axis.

jaysonkelly, 2017-01-21:
I'm having the same problem with RC8. The machine moves z to home simultaneously with the movement that brings the bed forward. My x-axis gantry smashed into my print, ripped it off the bed and dragged it to the back of the build surface.

I'm running a RigidBot 2 with the reprapdiscount full graphic display.

thinkyhead, 2017-02-15:
Could you enable `DEBUG_LEVELING_FEATURE` and log the output from a tiny (perhaps do-nothing) print that improperly homes Z at the end. Just use `M111 S247` to enable full logging before doing the print. Paste the output here or on Pastebin for examination. It should reveal where `G28 X0` is getting confused.

Phaelz, 2017-08-02:
I don't know what happened to this thread but I'm having the same problem.

I installed Marlin Bugfix 1.1.4 before yesterday. After finishing every print, the bed was going down (Z axis up). It went down untill it crashed, surpassing the fisical limit on configuration.h

I checked the G-Code generated by Slicer Prusa, it was the same I used on another printer and it worked fine. Then I modified the End G-Code to move the bed to the last mm available. (Z max is 176mm, in the G-Code I put G1 Z176 F2400).

Now the bed goes down as expected.

closed 2018-02-06 by thinkyhead