MarlinFirmware/Marlin #5653: G29 no longer double touching probe points, no change in config

bmarks-mylo, 2017-01-04:
New printer build with a few successful calibration prints.  Until today, G29 has worked perfectly.  I used it to help manually level the bed and it has run in my start GCode for all calibration prints so far.  Today I tried to print again and I noticed that there is no double touch on any of the probe points for G29.  Trying to run G29 through pronterface produces the same behavior, no double touch.  My G28 still double touches correctly.  

I made no changes from last night's print to my attempts today.  I did try to re-flash the firmware to see if somehow there was something saved to eeprom but that has had no effect.

[Configuration.h](https://gist.github.com/bmarks-mylo/a383a2c11d55a88426180d390912657a)

Blue-Marlin, 2017-01-04:
If you really want?
```
// Use double touch for probing
//#define PROBE_DOUBLE_TOUCH
```
but 'Why double touch probing is not a good thing #4455'

bmarks-mylo, 2017-01-04:
Maybe I am misunderstanding exactly what double touch is.  The "normal" behavior I was observing was:

move to probe point -> lower till probe is active -> raise -> move to next point -> repeat

The "abnormal behavior I started seeing was:

move to probe point -> lower till probe is active -> move to next point -> move to next point -> ...

There was no raising of the Z axis to check the next point it just ran through the XY positions of all the points until it reached the final point.

After enabling the PROBE_DOUBLE_TOUCH it seems to be doing the first behavior again.

Blue-Marlin, 2017-01-04:
Probably on of the following is to small - the probe stays triggered.
```
/**
 * Z probes require clearance when deploying, stowing, and moving between
 * probe points to avoid hitting the bed and other hardware.
 * Servo-mounted probes require extra space for the arm to rotate.
 * Inductive probes need space to keep from triggering early.
 *
 * Use these settings to specify the distance (mm) to raise the probe (or
 * lower the bed). The values set here apply over and above any (negative)
 * probe Z Offset set with Z_PROBE_OFFSET_FROM_EXTRUDER, M851, or the LCD.
 * Only integer values >= 1 are valid here.
 *
 * Example: `M851 Z-5` with a CLEARANCE of 4  =>  9mm from bed to nozzle.
 *     But: `M851 Z+1` with a CLEARANCE of 2  =>  2mm from bed to nozzle.
 */
#define Z_CLEARANCE_DEPLOY_PROBE   10 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  5 // Z Clearance between probe points
```
```
#define Z_HOME_BUMP_MM 2
```

bmarks-mylo, 2017-01-04:
Thank you, I will try that next.

NiklasOlofsson, 2017-01-04:
It does that when the probe is triggered or go in error state while probing. First point will be ok but if triggered it will move to each point and pause for a short time and then move to next point until end.

NiklasOlofsson, 2017-01-04:
Yep. Your probe is triggered all the time or in error state in case of BLTouch.

bmarks-mylo, 2017-01-04:
I merged in the latest RCBugFix branch and bumped the values of the config suggested by @Blue-Marlin and it seems that it's back to normal again.  Thank you for the help.

closed 2017-01-04 by bmarks-mylo