MarlinFirmware/Marlin #5654: Printrboard testers needed - new Arduino IDE system proposed

Bob-the-Kuhn, 2017-01-04:
I've put together a system that (hopefully) allows the Printrboards to compile the latest mainstream Marlin firmware without any special handling.

I need some Printrboard owners to test the new system and give me feedback.

Attached are new/updated **pins_PRINTRBOARD.h** and **pins_PRINTRBOARD_REVF.h** files.
[pins_PRINTRBOARD.zip](https://github.com/MarlinFirmware/Marlin/files/683905/pins_PRINTRBOARD.zip)

Instructions on setting up & using the new system are in the files.

The areas I'm most worried about are:
- Did I get the pin definitions right
- Is there access to the header I selected for the spindle control signals when an LCD adapter board is used

If you think I've gotten a pin definition wrong then please do the following:
1. In Configuration_adv.h enable "PINS_DEBUGGING"
2. Compile & upload
3. Enter the command "M43 W1 I1".  This command will report the pin number and the name of any pin that changes state.
4. Using a 1k (approximately) resistor, pull the questionable pin(s) to ground and see what is reported.

The testing involves downloading a new Arduino IDE extension.  Arduino extensions have a bad habit of not playing well with each other.  I'm pretty sure that the new extension will work but it might break an already installed extension.  The only recourse I know of is to uninstall the new extension when finished with it and then reload your extension.  You might have to delete the **Arduino15** directory to get rid of the last vestiges of the new extension.  On my Windows machine the path is: C:\Users\Bob\AppData\Local\Arduino15

There'll be more red text flying by when compiling.  These are warnings & can be ignored.  I haven't figured out how to eliminate the warnings AND have the extension run no matter what other extension is installed.

Roxy-3D, 2017-01-04:
Wow!  3 weeks ago, I could have helped.   My first printer had a a PrintRboard on it.  And I really wanted to keep it just the way it was.  But that company refuses to support their hardware.  They tell everybody to use Arduino v.22      Come on...   That doesn't make any sense at all.

I spent $18 to get an Atmega-2560 and a Ramps board and said "I'm done with PrinterBot stuff."   If they can't even update their files to let the PrintRboard work with current versions of Arudino, I can't spend the time fighting it.   And besides...   I can't have one of my printers not running a modern version of Marlin just because they refuse to do a few hours of work.

Bob-the-Kuhn, 2017-01-05:
@Roxy-3D - if you still have the Printrboard I'd appreciate a sanity check.  Maybe enable the M43 command, compile, download and see if a few of the pins respond correctly to the M43 W1 I1 command.

fiveangle, 2017-01-05:
I'm compiling my Printrboard RevF with RC8 today by installing the Teensy dev kit extensions, but do not have LCD adapter (stock Printrbot Simple Metal).  Am I missing something ?  Is your intent here to remove the requirement to install the Teensy extensions into the Arduino IDE ?

Bob-the-Kuhn, 2017-01-05:
@fiveangle  - I hope I'm not fixing a problem that doesn't exist.

The intent is to provide a system that works with all the mainstream Marlin software and makes the X-STOP available.  Details are in the pins files attached to this issue.

I tried Teensyduino with the "#define AT90USBxx_TEENSYPP_ASSIGNMENTS" line in FASTIO.h commented out.  I ran into one function that no longer worked.  It compiled with no errors & downloaded fine.  It was 99% functional.

I've also done Teensyduino with the "#define AT90USBxx_TEENSYPP_ASSIGNMENTS" line in FASTIO.h enabled.  With the attached pins file everything worked except the X-STOP.

My understanding is that there are Marlin branches that are specifically targeted at the Teensyduino environment.  Do you know what they offer that Mainstream Marlin doesn't or what's different about them?

How are you doing your downloads?  I was over the 64k limit so the Teensy loader wouldn't run.  I ended up manually running AVRDUDE through an USBtinyISP programmer.  Rather cumbersome.  The "Printrboard"  extension from https://github.com/scwimbush/Printrboard-HID-Arduino-IDE-Support is a Teensyduino environment but with a much nicer bootloader.  It'll handle the full 128K and downloads it in only 3-5 seconds without the need for an external programmer..

fiveangle, 2017-01-06:
I'm currently running Marlin RC/RCBugFix branches on my stock Printrbot Simple Metal 1403 with Heated Bed by dropping in my 2 config files and compiling using latest Arduino 1.6.11 with Teensyduino extensions installed (although I am not sure if I loaded anything else, which I will have to check on).  Once hex file is generated, I flash to board with Atmel Flip after placing in DFU mode.  If my Printrbot is 99% functional, I don't know what 1% is missing, but it is likely components I am not using or don't have (LCD, for example).  I originally attempted using PlatformIO but the recipe included in Marlin appears to be broken (which may be PlatformIO rather than Marlin, but I didn't investigate further).

Bob-the-Kuhn, 2017-01-06:
As I understand it, you should have the board "TEENSY++ 2.0" selected and, in **fastio.h**, the  "#define AT90USBxx_TEENSYPP_ASSIGNMENTS" line enabled (not commented out).  Am I correct?

The one piece I know for sure that doesn't work is the M43 command (enabled by the PINS_DEBUGGING flag at the end of Configuratuion_adv.h).  This command is meant to tell you what the state is of the pins on the controller (name, input/output, high/low, analog/digital, ...).  Drove me bananas until I figured out that some pins used the Teensy pin mapping and others used the FASTIO non-Teensy pin mapping.

I'm also working on code for a CNC feature and it doesn't work with the current system.

So, I can't point to any problem Marlin code that you'll be running on a 3D printer after it's setup and debugged.

Could you do me a big favor and try a few simple things using the proposed system?  I'm thinking something like putting the machine at the origin, downloading the test image, moving X +10mm, then back 11, same on the Y, do a Z +10, -5, and maybe even a cold extrude.

Bob-the-Kuhn, 2017-01-06:
@Roxy-3D - can you chime in here with some of the problems you were having?

Roxy-3D, 2017-01-06:
OK.   At the highest level, we had problems with Arduino using buggy versions of GCC.   We used to have pieces of code that cause compiler bugs to show up.   We had no choice but to say Marlin needs something like Arduino v.1.6.8 (or something close to that) or we won't compile.  It is my belief that those problems have been addressed.   Most likely, the current version (RC-8) can be compiled and loaded successfully with Arduino v0.22     But it is totally unsupported.  

If I still had a PrintRboard, I have no way to compile RC-8 and get it loaded onto the board.   And really...   That isn't a Marlin problem.  That is a PrintRbot problem.   That company should be more responsive to its hardware users.   For crying out load.   Almost for sure most PrintRboards have Marlin loaded on them.  Right???   Come on....  Update your Arduino files so the Blessed user can compile an up to date Marlin for the board.

That is the problem I was having....

fiveangle, 2017-01-11:
Tight on time right now, but will try your request when i change FW next.  Attached are config files I drop in and compile with.
[Marlin 1.1.0 config.zip](https://github.com/MarlinFirmware/Marlin/files/700103/Marlin.1.1.0.config.zip)

fiveangle, 2017-01-12:
Well defining PINS_DEBUGGING causes compile to fail entirely, so I'll defer to your expertise on what's not working.  However, if I drop in your pin definitions, it still fails:

[...]
> 
> pinsDebug.h: In function 'void pwm_details(uint8_t)':
> pinsDebug.h:750: error: 'TIMER0A' was not declared in this scope
>        case TIMER0A:
> 
>             ^
> 
> pinsDebug.h:757: error: 'TIMER0B' was not declared in this scope
>        case TIMER0B:
> 
>             ^
> 
> pinsDebug.h:767: error: 'TIMER1A' was not declared in this scope
>        case TIMER1A:
> 
>             ^
> 
> pinsDebug.h:774: error: 'TIMER1B' was not declared in this scope
>        case TIMER1B:
> 
>             ^
> 
> pinsDebug.h:781: error: 'TIMER1C' was not declared in this scope
>        case TIMER1C:
> 
>             ^
> 
> pinsDebug.h:791: error: 'TIMER2A' was not declared in this scope
>        case TIMER2A:
> 
>             ^
> 
> pinsDebug.h:798: error: 'TIMER2B' was not declared in this scope
>        case TIMER2B:
> 
>             ^
> 
> pinsDebug.h:808: error: 'TIMER3A' was not declared in this scope
>        case TIMER3A:
> 
>             ^
> 
> pinsDebug.h:815: error: 'TIMER3B' was not declared in this scope
>        case TIMER3B:
> 
>             ^
> 
> pinsDebug.h:822: error: 'TIMER3C' was not declared in this scope
>        case TIMER3C:
> 
>             ^
> 
> Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire 
> 'TIMER0A' was not declared in this scope
[...]

And going back to pins debugging disabled, there are a lot of warnings:

[...]
"C:\Program Files (x86)\Arduino\hardware\teensy/../tools/avr/bin/avr-g++" -c -Os -g -Wall -ffunction-sections -fdata-sections -MMD -fno-exceptions -felide-constructors -std=c++0x -mmcu=at90usb1286 -DTEENSYDUINO=134 -DARDUINO_ARCH_AVR -DARDUINO=10801 -DF_CPU=16000000L -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy" "-IC:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire" "C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\watchdog.cpp" -o "C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\watchdog.cpp.o"
In file included from C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\pins.h:115:0,

                 from C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\MarlinConfig.h:33,

                 from C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\Marlin.h:36,

                 from C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\watchdog.cpp:23:

C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\pins_PRINTRBOARD_REVF.h:178:0: warning: "USBCON" redefined

 #define USBCON 1286  // Disable MarlinSerial etc.

 ^

In file included from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\iousb1286.h:38:0,

                 from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\io.h:192,

                 from c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\pgmspace.h:90,

                 from C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\Marlin.h:32,

                 from C:\Users\SPEEDS~1\AppData\Local\Temp\arduino_build_173693\sketch\watchdog.cpp:23:

c:\program files (x86)\arduino\hardware\tools\avr\avr\include\avr\iousbxx6_7.h:963:0: note: this is the location of the previous definition

 #define USBCON  _SFR_MEM8(0XD8)

 ^
[...]

Haven't investigated the reason.  This is on current RC branch.

fiveangle, 2017-01-12:
Okay, I didn't realize this was a "system" and not just an updated pins definition file.  I'm not sure the reason for a new bootloader but I've not done that and am probably not going to in the short term.  The stock DFU boot loader works and I'm not interested in the CDC or LUFA based HID bootloaders as they have proven flakey and/or problematic when trying to update FW from Raspberry Pi (dfu-uploader works fine with stock Printrbot bootloader).

I'm quite sure my X-stop works with RC and my configs as is today ;)

Is that expected due to the Printrbot Simple Metal 1403 stock config possibly using some other pins for that function ?  (I haven't checked).

Bob-the-Kuhn, 2017-01-12:
Running the new pins file with Teensyduino is a good test to see if I set it up right.  It supposed to default to the Rev A pin definitions if Teensyduino is being used.  The intent is that you'll end up with exactly the same images with both old & new pins files when TEENSY++ 2.0 is the selected board.

I'm really surprised to hear that you're seeing more warnings with the new pins file with TEENSY++ 2.0 selected.  You'll definitely be seeing more warnings if you use the Marlin_AT90USB Arduino IDE extension than when using the Teensyduino extension.

If you're using the TEENSY++ 2.0 board selection then PINS_DEBUGGING won't provide useful information and should be disabled.  

The **'TIMER0A' was not declared in this scope** errors are because Teensyduino does not define some of the "standard" functions/structures.  I need to modify the PINS_DEBUGGING code to handle that situation.

I agree that changing bootloaders is not a good thing to do since this is a short term change for you.   What I like about the new bootloader is the Arduino IDE will automatically upload to the board if the Marlin_AT90USB extension or the "Printrboard" extension is used.  What I don't like is it can't do a verify.

closed 2017-02-18 by Bob-the-Kuhn