MarlinFirmware/Marlin #5655: How to control E3D Cyclops extruder  to  realize  double  color  mixture  printing?

lizlovespurs, 2017-01-04:
I'm  modifying  firmware  to  control  the  E3D  Cyclops extruder.  I   want  to  make  the   printing model  with  double  color.  For  example,  I  want  to  print  a   cube  with  20  mm  side length. The first
10 mm  height  of  the  cube  is  green   and  the  rest  10  mm  is   red.  If   the  medium  part  of  the  cube  is  yellow,  it  is  better.  Because  I   also   want  to   realize   a   little  color  mixture.  
      Have   you  ever  used  this  extruder?  What  is  the optimum  temperature  of  the nozzel  to   print   with  this  extruder ? There are  two  extruder  steppers  for  two  color  filament.  And   how  many steps   per  millimeter  should I  set  for   the   stepper   of   this  extruder?  I   hope  some   programmers   or   3d-printing  fans  can  give   me   some  advice  in   programming  or  printing   parameter   setting.  This  is  very  important  to  me.  If  you   have  some  good  ideas , please  dicuss  with  me!  Thank  you  very  much!

psavva, 2017-01-05:
I suggest you try the 3D Printing community on Google+

https://plus.google.com/communities/117814474100552114108

lizlovespurs, 2017-01-07:
Thank  you   very  much! @psavva
And   I   have   made  some  progress   in   printing   one  model   with   double   color   recently.

Roxy-3D, 2017-01-07:
It is my thought that we might need a 'Color Mix Command'.    Right now we are setup to move filament through a single extruder.   But in the case where there are multiple filaments being fed into a single extruder, wouldn't it be nice to be able to specify how much each of the filaments should move?  We probably would still want to move the filament through that extruder with a single command specifying the distance and feedrate.   But this new command would automatically break that higher level move down into separate (concurrent) moves to create the specified 'color'.

lizlovespurs, 2017-01-09:
The  command  should  also   depends   on   slicer   software. @Roxy-3D

thinkyhead, 2017-04-02:
> Right now we are setup to move filament through a single extruder. 
> wouldn't it be nice to be able to specify how much each of the filaments should move?

I added `MIXING_EXTRUDER` _months and months_ ago. It's documented in the configuration file and on the marlin documentation website.

closed 2018-02-26 by thinkyhead