MarlinFirmware/Marlin #5656: Modifications to control a non-3D-printer

jerryerry, 2017-01-04:
Hi there

I have an unusual question. I am making a filament joiner that connects the filament from two spools onto one larger spool. 

I am using RAMPS 1.4 with pretty much all the same control components used in 3D printers, such as steppers (6 of them), limit switches, heater cartridge, fan, and thermistor. It joins the filament and then winds it up automatically. There is also a hot wire cutter that cuts the filament so that an alternative filament can be joined. 
(There is also a DC and servo motor, and I have an additional RAMPS shield that I thought I could use, since all the steppers won't fit on one)

Some issues I've had (because I've never used Arduino before) is that I don't know how to control the steppers, mechanical (and opto) switches, and turn the heater on and off.

I am finding it difficult to find a way to modify Marlin to be used to include a program for this joining project. I would so appreciate it if you could give me some pointers? If needed, I can send more info on the system I'm making.

Many thanks
Jerusha

[image]
[image]
[image]

bobc, 2017-01-04:
It's an interesting project, but I'm not sure Marlin is a good fit for it. Marlin is really designed around precise position control; in a filament winder you don't really care how far is traveled, but it needs to rotate continuously until an event occurs (etc). That's a bit like homing I suppose, but in Marlin is specialized for printers rather than general purpose.
There are many more details to see before I could get a really good idea. For example, is there a PC involved? That makes a huge difference.
To be honest, this is really out of scope for Marlin issue tracker. I would suggest asking at http://forums.reprap.org/index.php, they have a good cross-section of knowledgeable people, and are often willing to consider novel projects especially if they are printer related.

> I've never used Arduino before

That is going to be a problem! Whatever route you take. If you are learning Arduino, it might be better to start from scratch and build up. The control requirements do not look that complicated. Either way you will need to be a good learner or find someone local willing to get involved.

jerryerry, 2017-01-04:
Hi bobc, thanks for the quick response. 
Since I'm unfamiliar with Arduino I thought it would make life a bit easier using Marlin since it already has the LCD and temp control set up well, and I'm on a serious deadline (the project is my bachelors thesis).

I do need precise movement in certain parts of the operation, such as the point where two filament ends are brought together and then welded. 

I thought that perhaps I could just write the program using gcode?

jerryerry, 2017-01-04:
gcode seems to be working well

closed 2017-01-04 by jerryerry

bobc, 2017-01-04:
Even with a full implementation of Gcode (
https://en.wikipedia.org/wiki/G-code), you would not be able to write a
control program, but anyway Marlin implements a subset of commands, there
is no "programming" capability.

I guess if I was doing a quick hack I might rewrite the Marlin loop()
function and replace it with a state machine, and hope to find enough
functions in the rest of the Marlin code base to use directly or adapt to
minimise the amount of new code written.

jerryerry, 2017-01-04:
I meant that I had tried running the gcode in using marlin. It is all that I needed - the control is taken care of already in marlin.

I tried writing a procedure in the loop but it's so difficult trying to figure out what methods are accessible like that. I couldn't figure out a way to control any stepper movement because there's a subclass of the stepper library. I then tried writing to the pins manually so that I didn't have to use the stepper class, but that also didn't work. There's too many hidden settings I think.

Anyway, thanks for your comments. I am going to try write what I need in gcode, and anything else I will consult that forum you mentioned.