MarlinFirmware/Marlin #5661: RC8 ringing

MostTornBrain, 2017-01-05:
It looks like the code in #4997 has introduced some noticeable ringing on XY direction changes that wasn't present in earlier versions of planner.cpp.  If I use earlier firmware with the same configuration values, then the ringing is a lot less noticeable.

The ringing is directly related to the changes for the jerk calculations that were "Adopted from Prusa MKS firmware".  If I revert just that section of code in planner.cpp and use the old jerk calculation code the ringing goes away.
 
I'm not sure if this is a bug or expected behavior with the new code.  Maybe it is expected that with RC8 one needs to figure out new acceleration and jerk values, but I thought I should mention it since the ringing introduced is significant using the same settings going from version 1.02 to RC8.

cfelicio, 2017-01-06:
Did you try reducing the jerk values by 1/2? That seemed to help me. I also had issues with the change...

Sebastianv650, 2017-01-06:
In fact with this change you need new jerk values. With this PR, moves starts at jerk speed instead of jerk speed / 2. While it's something the user has to be aware of when updating to RC8, I like the new behaviour because jerk speed / 2 was a little bit random in my eyes..

MostTornBrain, 2017-01-06:
I tried using 1/2 the jerk value that I had been using in 1.02 and it does reduce the ringing, but it still isn't as good as the 1.02 behavior.  The previous jerk calculation wasn't always jerk / 2, it was a little more complicated than that.  It only used jerk / 2 if there was just one move queued.   If there was more than one move queued,  it computed:

`jerk = sqrt(pow((current_speed[X_AXIS]-previous_speed[X_AXIS]), 2)+pow((current_speed[Y_AXIS]-previous_speed[Y_AXIS]), 2));`

and then used that value to proportionally adjust the base jerk value.

Not sure if it would be useful, but I could do some debug logging and compare jerk values between the two revisions to see when the jerk / 2 is being used vs. the slightly more complicated calculation and how this maps to the RC8 jerk calculation.

cfelicio, 2017-01-06:
If you have time to figure it out, it would be excellent! I fiddled with the settings quite a bit, and for me it wasn't the ringing, but I was missing steps with RC8 and with 1.0.2 was totally fine.

What I ended up doing is to replace my drivers back to A4988, as they are stronger, but if you find a setting on RC8 that works similar to 1.0.2 that would be amazing! :-)

jlarsenwy, 2017-01-09:
I also started to see ringing issues with RC8 on my MakerFarm Pegasus.  I tried dropping the default X/Y values slowly from the default of 20 all the way down to 5 and didn't really see an improvement.  I'm interested to see if anyone has any suggestions on settings.

Sebastianv650, 2017-01-09:
Strange that lowering XY jerk values that much doesn't show a result in the print. Can you provide a test print RC7 vs RC8 (maybe also RC8 with default jerk values you also used in RC 7 and a third print at 5mm/s jerk).
I would be interested in the magnitude we are talking about. I have some ideas, but they don't fit if reducing the values doesn't change anything.

Blue-Marlin, 2017-01-09:
>  I tried dropping the default X/Y values slowly from the default of 20 all the way down to 5 and didn't really see an improvement.

Do you have the EEPROM on?

Kaibob2, 2017-01-09:
Is there a way to download the last version before #4997? Or some version right before RC8 release?  I'm experiencing ringing problems since some time too, but I believed they where caused by the massive 6mm Alu heatbed I installed for the Y-axis. As I only have increased ringing in the Y direction, this is more likely but I would like to verify. 
I changed the heatbed around the same time when RC8 came out, so there might be a coincidence. Never had ringing before with the 3mm Alu heatbed and Rc7 Bugfix

jlarsenwy, 2017-01-09:
> Do you have the EEPROM on?

I do, but I'm pretty good about reloading it when I re-flash.  It's very possible I forgot to though.  I'll try to redo my tests for comparison and post my findings.

jlarsenwy, 2017-01-11:
Ok, I tried to do some really basic side-by-side testing of RC7 and RC8 using different jerk speeds.  I indeed must have forgotten to clear EEPROM at some point before, as there is a noticeable improvement by dropping the X/Y jerk speed by half each time.  There's still a difference between the two versions using the default X/Y jerk speed of 20mm/s, but it's not major.

Pictures don't make it quite as apparent as in person and I'm a bad photographer, but here you go:  

http://imgur.com/a/6Shuv

All prints were done using the same gcode file at 50mm/s with acceleration at 1000mm/s.  I'm still kind of a newbie so I'm not sure if there's a better object or better settings to test with.

In my opinion, RC7 at 20mm/s is about comparable to RC8 at 10/mms.  This makes sense based on some of the calculation changes that someone listed above.  That makes me wonder though.  With those changes, doesn't it make sense to lower the default speeds in configuration.h?

MostTornBrain, 2017-01-12:
After double-checking my printer's calibration, I noticed the E steps were advancing the filament about 20% more than they should.  (I had previously calibrated E steps using this same spool of filament with version 1.02.) After tuning the E steps in RC8 to get an accurate extrusion rate, and then also enabling and calibrating LIN_ADVANCE, I am now getting BETTER prints with RC8 than I was with 1.02.  The ringing I was previously seeing is no longer present.

I am now quite happy with RC8.

I need to go back and do another test with 1.02 using the same E step value I'm using in RC8 to see if there is a firmware behavior difference or whether I just have a crappy spool of filament that was a different thickness when I did the original E step calibration.

shacal, 2017-01-12:
"filament width is inconsistent" - pretty common now days (china hit the market)
I use probe to measure that and works beautifully.

jlarsenwy, 2017-01-12:
> I use probe to measure that and works beautifully.
I hate to take this thread off-topic but that sounds really interesting.  Could you send some more information on how you set that up?

thinkyhead, 2017-04-02:
We'll have to remember to document the change in jerk for 1.1.

closed 2017-04-02 by thinkyhead