MarlinFirmware/Marlin #5662: Anyone used Stallguard2 on TMC2130 as touchprobe?

boelle, 2017-01-05:
as per subject i wonder if anyone has used the Stallguard2 with the 2 outputs on the driver as a touchprobe?

JustAnother1, 2017-01-06:
https://www.youtube.com/watch?v=Ec0rvkBLPYg

boelle, 2017-01-06:
touchprobe not endstop... there is a huge difference

@Wurstnase do you think its possible or would it not trigger as the motorload get lighter when the nozzle rest on the bed

ekozan, 2017-01-07:
you will damage your bed or nozzle :/

JustAnother1, 2017-01-07:
@ekozan did you test that? Do you know how much force is needed for the Stallguard2 to trigger?

Blue-Marlin, 2017-01-07:
The calculation is simple,
Stallguard detects stalls.
The motor stalls at its maximum torque.
Together with the mechanical translation you can calculate the force.

JustAnother1, 2017-01-07:
@Blue-Marlin it is not so easy. The TMC2130] as described here http://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2130_datasheet.pdf has an SPI Interface. The Stallguard Value is a number. The motor current can also be configured. and the current stall guard value can be read. You might want to watch this: https://www.youtube.com/watch?v=Prw7wNa20Gk

boelle, 2017-01-07:
to make it clear: has anyone done it before and can give a guide for us "dummies" to follow

Wurstnase, 2017-01-07:
Stall**Guard** not detection.
It's a bit tricky. But maybe it could be used for bedprobing. But the speed should not change like in the current setup. Stallguard don't work at low speeds.

boelle, 2017-01-07:
@Wurstnase i was thinking excatly about bed probing

teemuatlut, 2017-01-08:
Bed probing will be more difficult since often times the bed is a bit springy, you'll have a lot of torque due to the leadscrews and you'll need a lot more precision for z-height than for x/y -endstops. The sensorless endstop detection for Marlin has already been done tho, if you'd like to test bed probing as well.

boelle, 2017-01-08:
lets pretend the bed is not springy

and sensorless endstop detection.... maybe i'm blind or have been away for to long a period but i cant see any settings for that

teemuatlut, 2017-01-08:
It's largely untested since I only have one printer and have been doing work on the Teensy fork, and even then, the changes never made it to the official repo because Marlin went with the other TMC2130 library.

If you want to try it out for bed probing, there's nothing stopping you and it might just be a matter of finding the right threshold value.

closed 2017-01-08 by boelle

widjis, 2017-12-28:
why dont yo use this driver as probeless bed leveling?

teemuatlut, 2017-12-29:
In short, the torque multiplier from the lead screw is too much and the motor will not see much resistance when hitting the bed. The bed also often is at least slightly springy. You can search for older threads about the topic as it has been discussed before.

alfredanil, 2017-12-29:
@teemuatlut What if the motor is driven not by lead screw but with belt and the bed is "not a cantilever" (supported on four points) then would the stallguard2/sensorless probing work?

teemuatlut, 2017-12-29:
Perhaps but I don't know if the back EMF sensing (stallGuard) would work with multiple motors.

jammi, 2018-01-01:
What if you have two tmc2130's for Z, would it work then? Also let's assume you'd configure the Vref for the Z drivers really low when doing the probing, to prevent damage and make the motor stall very easily. Include @boelle's assumption of bed definitely not being springy but rather one of the most solid parts on the printer.

ToddBeaulieu, 2018-03-19:
I'm intensely curious about this topic I really dislike "it can't be done" as an answer. It's possible that it might not work reliably when the current and speed are set way low (low enough to be delicate), but it also seems that nobody's really tested this out. To me, figuring out the Z offset is a major PITA and in 2018 I find it hard to believe that this can't be automated. I got excited when I saw the code for M915, but sadly, it doesn't seem to be for such a feature. Thinking about trying to add it myself.

Roxy-3D, 2018-03-19:
>  I got excited when I saw the code for M915, but sadly, it doesn't seem to be for such a feature. Thinking about trying to add it myself.

If you get it working...   we can add it to the M915 command.

teemuatlut, 2018-03-19:
That's not what `M915` is for.

Z homing and/or probing was tested by @tonokip at Lulzbot and the conclusion was that it's not feasible because of the torque multiplier lead screws or threaded rods have.

ToddBeaulieu, 2018-03-19:
Could you please explain what 915 is trying to accomplish? Thank you.

teemuatlut, 2018-03-19:
On a typical cartesian machine it will level the X axis according to the frame. It works similar to how the Prusa printers do it but more automated.
See their (old) YouTube demonstration [here](https://youtu.be/JqH41K2vq0g?t=294).

ToddBeaulieu, 2018-03-19:
Ha! I did not see that coming. Thanks!