MarlinFirmware/Marlin #5664: Taz 5 getting thermal runaway error after stopping a print and restarting

billyd60, 2017-01-06:
Just like the subject says. Every time I stop a print in process from the LCD screen of my Taz 5 and restart a print, I will eventually get a thermal runaway error during the second print. I have to power down the printer before restarting a new print job to avoid the thermal runaway error (after an LCD stop print command).

I am using an E3Dv6 extruder and custom bed setup that never got thermal runaway errors on previous versions so I don't think it's a hardware problem. Let me know if you want to see any files.

Blue-Marlin, 2017-01-06:
What is your Marlin version? Feeld we had two dozen changes in the code and in the defaults of this feature during the last two years.
Is the runaway related to the hotend or the bed? Protocol?
Does the problem occur right at the beginning of the second print or somewhere in the middle?
Do you have a temperature graph (ideally RH)  or a log including the M105 responses?

If the error occurs directly when starting the second print, likely the temperature is falling fast and the heeater needs a bit more time to compensate the undershoot. Increase the time constant in Configuration_adv.h.

billyd60, 2017-01-06:
Sorry latest rcbugfix

The problem occurs at random points after a restart of a print but early on. And sometimes I will get a runaway error before I get a chance to restart at all. I also just had one after powering down (from a LCD stop print command) and then up while everything was still hot and got the error  while reheating back to printing temps. So I powered down again, and waited about 30 seconds and then printed without an issue.

Blue-Marlin, 2017-01-06:
Do your prints finish with the part-fan on? That may delay heating.
After waiting some time the temperature dos not fall that fast any more.

billyd60, 2017-01-06:
Print primarily in ABS so don't typically even use the part fan. I like your idea of increasing the time constant I will give that a try I feel like that might work. I don't know which part is the offender, whether it is the extruder or the bed I print without usb attached so I don't have any logging to this point. The error message on screen doesn't specify.

Sometimes the error is just Heating Failed instead of thermal runaway. This is evolving for me as I print with the new firmware. I am getting excellent prints btw, just don't stop a print from the LCD and it's fantastic.

psavva, 2017-01-07:
I had the same issues on my printer. 
To fix it, I upped the values of 
THERMAL_PROTECTION_PERIOD
 THERMAL_PROTECTION_HYSTERESIS 

In addition, check that your thermistor and heater are securely tightened/fixed. 
My heater somehow became lose which caused most of my issues. 
Make sure that the fan wires are not causing noise on your thermistor leads...
Make sure your fan is off during the initial heating. When you turn the fan on, check if there is a sudden drop in temperature, if so ensure your fan is not directed towards the heater block itself....
Lastly, a heater block silicone sock held too.

billyd60, 2017-01-29:
Psavva I did have a loose thermistor canister on my e3dv6 and tightening that fixed the issues with my extruder thermal runaways. I am still having trouble with my heated bed, but I am starting to think that may be a hardware problem as well. It seems it was either an unfortunate coincidence that I started having trouble just as I upgraded to the latest rcbugfix, or perhaps this newer firmware is less forgiving on "shaky" hardware.

The problem I am having now is an intermittent heating failed problem with my heat bed. It's either the silicone heater itself, the wiring or the controller. I am leaning toward a wiring issue. I am going to swap out the dc connectors (between the silicone heater and the wiring from the controller) with better ones to make sure I am getting perfect contact. If that doesn't work I will check the connection inside the electronics enclosure. I am doing the connectors first because I know they are not a good tight fit anyway and have been wanting to fix that even before the trouble showed up.

psavva, 2017-01-29:
If you got any soldering points, check that you have good contact.

billyd60, 2017-02-08:
Just for the record, the problem was a bad heat bed connector at the Taz 5 electronics enclosure.

I now have the heat bed and it's thermistor connected directly to the rambo board with no connectors in the run.. My heat bed has never worked so well. I think this might be a good mod for the Taz 5. The enclosure connector is weak and I think never had great contact and I was apparently losing alot of power to heat the bed through that connector. My heat bed now heats up in a few minutes.

closed 2017-02-08 by billyd60