MarlinFirmware/Marlin #5666: ABL causing Z positioning issues after M420/G29

chorca, 2017-01-06:
### Overview
Running the latest RCBugfix of RC8, I'm having problems getting ABL to work properly. I'm unable to get the printer to properly print after running a Bilinear Mesh level routine. The Z axis appears to either be zeroed incorrectly, or is stepping without checking it's real location.

### Environment
My printer is a custom CoreXY design. I'm currently using Megatronics 2.0 and the associated board file, along with a BLTouch to act as a Z homing probe as well as ABL probe. Current config files can be found here: [marlinconf.zip](https://github.com/MarlinFirmware/Marlin/files/689370/marlinconf.zip)

I have ABL enabled (attempted both with and without fade enabled), BLTouch configured to a custom pin on the Megatronics board (45), and EEPROM enabled on the board. I have tested the EEPROM facilities and they are working, as is the grid readout I'm seeing from the ABL routine.

### Issue
After powering on the printer, performing a G28 to home all axis, a M501 to load EEPROM settings, an M420 S1 to enable the grid leveling matrix, and attempting a print, the Z axis begins running upwards until it hits physical limits and continues to do so past the max size of the printer. This occurs at the first G1 Zxxx command sent to the machine after a M420 S1.

At the same time, checking the printer's assumed location of the head after issuing the M420 S1 command shows that it has reset the Z axis from it's usual after-homing safe height (17mm above zero) to 0.00. The printer is instructed to move 0.07mm upwards, and it begins to drive upwards forever.

Alternatively, attempting a print immediately after issuing a G29 to perform the ABL will not result in the Z axis running upwards forever, but it does reset Z's real 17mm height to 0 and cause the printer to air print.

### Example 1
ABL Grid already created and stored via G29, M500. After reset, attempting to print
```
Send: G28
Recv: X:311.00 Y:310.00 Z:17.00 E:0.00 Count A: 99360 B:160 Z:13600
Recv: ok
```
After homing, Z is at 'safe height'
```
Send: M420 S1
Recv: echo:Bed Leveling On
Recv: ok
```
Enabled Bed Leveling from M420. The same issue occurs whether or not an M501 is issued beforehand
```
Send: M114
Recv: X:311.00 Y:310.00 Z:0.00 E:0.00 Count A: 99360 B:160 Z:13600
Recv: ok
```
Here the machine shows Z has been reset to 0, even though the motors have not moved.
```
Send: M104 S190 T0
Recv: ok
Send: M109 S190 T0
Recv:  T:187.3 /190.0 B:50.9 /50.0 @:65 B@:0 W:9
Recv:  T:187.8 /190.0 B:50.9 /50.0 @:60 B@:0 W:8
Recv:  T:188.5 /190.0 B:50.8 /50.0 @:53 B@:0 W:7
Recv:  T:189.1 /190.0 B:50.7 /50.0 @:46 B@:0 W:6
Recv:  T:189.5 /190.0 B:50.7 /50.0 @:42 B@:0 W:5
Recv:  T:190.0 /190.0 B:50.7 /50.0 @:37 B@:0 W:4
Recv:  T:190.5 /190.0 B:50.7 /50.0 @:32 B@:0 W:3
Recv:  T:191.1 /190.0 B:50.7 /50.0 @:27 B@:0 W:2
Recv:  T:191.6 /190.0 B:50.7 /50.0 @:20 B@:0 W:1
Recv:  T:191.6 /190.0 B:50.7 /50.0 @:23 B@:0 W:0
Recv: ok
Send: G92 E0
Recv: X:311.00 Y:310.00 Z:17.00 E:0.00 Count A: 99360 B:160 Z:13600
Recv: ok
Send: M114
Recv: X:311.00 Y:310.00 Z:17.00 E:0.00 Count A: 99360 B:160 Z:13600
Recv: ok
Send: G1 E-2.0000 F1500
Recv: ok
Send: M114
Recv: X:311.00 Y:310.00 Z:17.00 E:-2.00 Count A: 99360 B:160 Z:13600
Recv: ok
Send: G1 Z0.070 F1002
Recv: ok
```
Here, printer begins running Z full speed positive, I send a few M114s to see where it thinks it is
```
Send: M114
Recv: X:311.00 Y:310.00 Z:0.07 E:-2.00 Count A: 99360 B:160 Z:68659
Recv: ok
Send: M114
[... 41 lines cut ...]
Recv: 11 +0.36481 +0.34033 +0.31584 +0.28468 +0.24019 +0.18902 +0.14449 +0.11281 +0.08776 +0.06398 +0.03644 +0.01017 -0.00505 -0.00441 +0.00729 +0.02537 +0.04741 +0.07585 +0.11333 +0.16378 +0.22328 +0.28278
Recv: 12 +0.35375 +0.33213 +0.31051 +0.28250 +0.24162 +0.19435 +0.15375 +0.12648 +0.10588 +0.08500 +0.05815 +0.03102 +0.01375 +0.01046 +0.01704 +0.03125 +0.05125 +0.07889 +0.11750 +0.17185 +0.23718 +0.30250
Recv: 13 +0.34278 +0.32251 +0.30224 +0.27657 +0.24065 +0.19932 +0.16181 +0.13154 +0.10508 +0.08130 +0.05837 +0.03812 +0.02486 +0.01926 +0.02066 +0.03139 +0.05172 +0.08138 +0.12185 +0.17724 +0.24343 +0.30963
Recv: 14 +0.33403 +0.31395 +0.29387 +0.26981 +0.23929 +0.20480 +0.16986 +0.13290 +0.09551 +0.06593 +0.04841 +0.03870 +0.03236 +0.02524 +0.02147 +0.02903 +0.05110 +0.08449 +0.12759 +0.18311 +0.24834 +0.31356
Recv: 15 +0.33250 +0.31148 +0.29046 +0.26625 +0.23769 +0.20593 +0.17125 +0.12898 +0.08380 +0.05000 +0.03565 +0.03269 +0.03125 +0.02431 +0.01889 +0.02625 +0.05134 +0.08921 +0.13625 +0.19431 +0.26153 +0.32875
Recv: 16 +0.34144 +0.31813 +0.29483 +0.26755 +0.23416 +0.19680 +0.15782 +0.11367 +0.06790 +0.03361 +0.01794 +0.01376 +0.01273 +0.00878 +0.00800 +0.02028 +0.04999 +0.09277 +0.14546 +0.21032 +0.28509 +0.35986
Recv: 17 +0.35759 +0.33088 +0.30417 +0.27204 +0.23039 +0.18332 +0.13773 +0.09306 +0.04986 +0.01667 -0.00255 -0.01177 -0.01440 -0.01364 -0.00629 +0.01389 +0.04950 +0.09794 +0.15759 +0.23166 +0.31694 +0.40222
Recv: 18 +0.37625 +0.34569 +0.31514 +0.27875 +0.23148 +0.17838 +0.12875 +0.08389 +0.04250 +0.01000 -0.01199 -0.02509 -0.02875 -0.02398 -0.00977 +0.01750 +0.05926 +0.11407 +0.18125 +0.26468 +0.36046 +0.45625
Recv: 19 +0.39667 +0.36192 +0.32718 +0.28810 +0.24077 +0.18910 +0.14051 +0.09553 +0.05363 +0.02065 -0.00206 -0.01586 -0.01894 -0.01188 +0.00588 +0.03792 +0.08580 +0.14794 +0.22315 +0.31538 +0.42068 +0.52597
Recv: 20 +0.41958 +0.38022 +0.34086 +0.29968 +0.25492 +0.20835 +0.16338 +0.11860 +0.07543 +0.04157 +0.01892 +0.00559 +0.00366 +0.01230 +0.03234 +0.06833 +0.12257 +0.19277 +0.27657 +0.37776 +0.49256 +0.60736
Recv: 21 +0.44250 +0.39852 +0.35454 +0.31125 +0.26907 +0.22759 +0.18625 +0.14167 +0.09722 +0.06250 +0.03991 +0.02704 +0.02625 +0.03648 +0.05880 +0.09875 +0.15935 +0.23759 +0.33000 +0.44014 +0.56444 +0.68875
Recv: 
Recv: X:556.00 Y:605.00 Z:0.00 E:0.00 Count A: 185760 B:-7840 Z:13600
Recv: ok
```
This shows that even though it had ended the sensing at a 'safe height' from the probe stow, Z is still set to 0, although the nozzle is actually 17mm away from the bed.
Subsequently attempting a print will cause the printer to begin printing at that height (17mm + 0.07mm for first layer)

### Tests
I have attempted to enable and disable fade, and noticed that with fade ENABLED in configuration.h and issuing a fade parameter to the M420 command, (M420 Z2 S1), Z would NOT be reset to zero (as seen via M114), but the head would still 'fly away' upon issuing a G1 Zxxx command.

Printing directly after a G29 command has different results than printing after a M420 S1 command, but both do not work.

### EEPROM settings
[eeprom.txt](https://github.com/MarlinFirmware/Marlin/files/689425/eeprom.txt)

manianac, 2017-01-06:
When you compile, how much minimum memory usage does it report?  7x7 grid w/ subdividing is pretty massive.

chorca, 2017-01-06:
```
Sketch uses 77674 bytes (30%) of program storage space. Maximum is 253952 bytes.
Global variables use 5623 bytes (68%) of dynamic memory, leaving 2569 bytes for local variables. Maximum is 8192 bytes.
```
The printer bed is 600x600mm, and has a bit of a dip to it, so that's why I'm using a large number of points.

I tried lowering the points from 8 to 7, and that seemed to change behavior:
```
Sketch uses 77734 bytes (30%) of program storage space. Maximum is 253952 bytes.
Global variables use 4995 bytes (60%) of dynamic memory, leaving 3197 bytes for local variables. Maximum is 8192 bytes.
```
Now printing is working correctly, but with bilinear enabled, it does not seem to be compensating enough.. some areas are far too low, others are too high, even after enabling leveling. Z motors are running during print moves but it does not seem like enough.
For this issue though, it seems that the memory might be the cause, will do more testing on lingering unevenness.

manianac, 2017-01-07:
Try disabling bed subdivision, even at 7 you could be running out of memory (you could check with the M100 gcode if you enable it).

Try upping your points to 10/11 after disabling subdivision as a test.

Koshu, 2017-01-08:
Same Problem here with a 5x5 Grid, biliniar bed leveling but with subdivisions disabled. M420 S1 sets Z to zero regardless of its position. Issuing a another homeing after that disables the bed leveling again.

chorca, 2017-01-08:
I'm actually not able to compile with more than 8 points, it alerts for low memory in Arduino and so I haven't attempted it.
Even at 7 points I'm running into odd issues.. M501 + M420 S1 still sets Z to 0 wherever it's at, and runs the bed up forever on printing, but doing a manual G29 and then printing seems to work.

I can try and lower it even further and see if that helps, but may not very accurately map my bed at this size.

Koshu, 2017-01-08:
Just tested again with AUTO_BED_LEVELING_LINEAR which seems fine. It only occurs if bilinear is used.

chorca, 2017-01-08:
Even more oddness. If I restart the board, do an G28 to home, G29 to probe the bed, then attempt a print, it will start printing in the air (Z reset to 0). If I then stop the print, and try and do a G29, it does not try and find safe home first, but it starts running Z negative into the bed, without moving to a safe probe deploy distance first. If Z was actually reset, it should have gone +15mm (as set in the configuration.h file) but it just instantly starts running negative.

Koshu, 2017-01-08:
Read a bit of the code. I never did any coding for marlin and i am completely unfamiliar with it. So these are not more than guesses but I found i few things seems weird to me. 

in Marlin_main.cpp function set_bed_leveling_enabled line 2240 and following:
 
```
#elif HAS_ABL

      if (enable != planner.abl_enabled) {
        planner.abl_enabled = enable;
        if (!enable)
          set_current_from_steppers_for_axis(
            #if ABL_PLANAR
              ALL_AXES
            #else
              Z_AXIS
            #endif
          );
        else
          planner.unapply_leveling(current_position);
      }

    #endif
```

The enable variable enables the ABL, however because of  if (!enable) it runs planner.unapply_leveling(current_position) the moment ABL gets enabled. Is this correct? I would had assumed that unapply_leveling should be called the moment ABL gets disabled.

Further in planner.cpp function unapply_leveling line 627 and following:
```
 #elif ENABLED(AUTO_BED_LEVELING_BILINEAR)

      #if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
        const float c = bilinear_z_offset(logical);
        logical[Z_AXIS] = (z_fade_height * (RAW_Z_POSITION(logical[Z_AXIS]) - c)) / (z_fade_height - c);
      #else
        logical[Z_AXIS] -= bilinear_z_offset(logical);
      #endif

    #endif
```

If ENABLE_LEVELING_FADE_HEIGHT is not defined (disabled) the calculated offset gets subtracted from the current position (is "logical" the current position?). However, if ENABLE_LEVELING_FADE_HEIGHT is defined (enabled) the logical position is completely overwritten. Shouldn't there be an "-=" as well? Especially if z_fade_height is set to zero, this would zero the z_position. Also i don't understand the (z_fade_height - c) part of that line. Couldn't this result in an devide by zero error?

manianac, 2017-01-08:
>The enable variable enables the ABL, however because of if (!enable) it runs planner.unapply_leveling(current_position) the moment ABL gets enabled. Is this correct? I would had assumed that unapply_leveling should be called the moment ABL gets disabled.

I could be way off, but I believe the intention there was to update current position with the leveling offset.  On my printer I have an endstop switch on the Z axis that stops above the bed, but use the probe to measure below this switch.  So when a `G28` is issued, Z0 is above the bed.  But once ABL is enabled with `M420`, `unapply_leveling` updates the logical Z position (in this case as a positive number, since we're above the bed).  That way if a move to Z0 is issued, we will now actually end up at the bed.

>Shouldn't there be an "-=" as well? Especially if z_fade_height is set to zero, this would zero the z_position. Also i don't understand the (z_fade_height - c) part of that line. Couldn't this result in an devide by zero error?

I don't think it should be a "-=" as the Z position is already in the equation, otherwise you would be subtracting the entire Z axis height from itself.  Consider this example:
```
fade height = 10
bilinear offset = 0.2
At a Z axis height of 5 it is updated to ~4.89
At a Z axis height of 9 it is updated to ~8.97
```
So the math is sound, it is fading as it should.  A "-=" operator would kill this.

I am interested in the potential divide by zero, but curious why the cpu doesn't reset or the watchdog catch it.  While I haven't found a situation for the abl grid to have the same value as the z_fade_height, it would be catastrophic.

@Koshu and @chorca, I suppose you could try with fading disabled, and see if it still occurs?

manianac, 2017-01-08:
>Even more oddness. If I restart the board, do an G28 to home, G29 to probe the bed, then attempt a print, it will start printing in the air (Z reset to 0). If I then stop the print, and try and do a G28, it does not try and find safe home first, but it starts running Z negative into the bed, without moving to a safe probe deploy distance first. If Z was actually reset, it should have gone +15mm (as set in the configuration.h file) but it just instantly starts running negative.

I was experiencing the same thing not too long back when the ABL saving to EEPROM PR was being merged.  While I never found the cause, what are you compiling with?  I had a sneaking suspicion I had a gcc bug or something odd, and had to update.

Koshu, 2017-01-08:
Unfortunatly I can't test for now because i have a problem with my probe (not related to this bug). 

However, I still don't quite understand the working of the ABL planner. 

`(z_fade_height * (RAW_Z_POSITION(logical[Z_AXIS]) - c)) / (z_fade_height - c);`

What if z_fade_height is set to zero? The result will always be zero. Especially since it is initialized as such:

```
#if ENABLED(ENABLE_LEVELING_FADE_HEIGHT)
  float Planner::z_fade_height = 0.0,
        Planner::inverse_z_fade_height = 0.0;
#endif
```

Which in regards to the documentation in the code is completely acceptable:

```
  /**
   * M420: Enable/Disable Bed Leveling and/or set the Z fade height.
   *
   *       S[bool]   Turns leveling on or off
   *       Z[height] Sets the Z fade height (0 or none to disable)
   */
```

This also fits the observation of @chorca :

> I have attempted to enable and disable fade, and noticed that with fade ENABLED in configuration.h and issuing a fade parameter to the M420 command, (M420 Z2 S1), Z would NOT be reset to zero (as seen via M114), but the head would still 'fly away' upon issuing a G1 Zxxx command.

About the divide by zero problem: It might by unlikely to happen and that's why nobody noticed it until now. However, in my experience equations that could result in a divide by zero error are a very bad idea. Because not only could it crash everything (literally), the problem already start if it gets close to zero. The result would tend to infinite. With limited precision of float this often has very weird side effects.

manianac, 2017-01-08:
I'm still unable to duplicate any of these concerns, with varying Z fade heights (including zero) and movements.  Maybe @thinkyhead can chime in if he has a zero check somewhere else that I'm missing, I only see one in apply_bed_leveling.

@Koshu why not add a zero-check and see if the problem still occurs?

Koshu, 2017-01-08:
Would like to but first i need a new probe :/

manianac, 2017-01-08:
My mistake, forgot you mentioned that.  You could still test it by switching the probe endstop yourself, to fill in some bogus values to the table?

Koshu, 2017-01-08:
I was able to test something. Just noticed that the manual_bed_leveling uses the same line of code (using the manual method for now). I could reproduce the error here as well but MBL needs to be activated by M420:

```
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:1600
>>> G1 Z15
SENDING:G1 Z15
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:15.00 E:0.00 Count X: 0 Y:0 Z:7214
>>> M420 S0
SENDING:M420 S0
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:18.03 E:0.00 Count X: 0 Y:0 Z:7214
>>> M420 S1
SENDING:M420 S1
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:7214
```

Even so Z was at position 15, after disabling and re-enabling MBL it is set to zero. So enabling via M420 only works if Z is at zero to start with.

Koshu, 2017-01-08:
Update: Repeated the test with fade of 1.0. Everything works fine:

```
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0 Y:0 Z:1600
>>> G1 Z15
SENDING:G1 Z15
>>> M420 S0 Z1
SENDING:M420 S0 Z1
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:18.03 E:0.00 Count X: 0 Y:0 Z:7214
>>> M420 S1 Z1
SENDING:M420 S1 Z1
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:18.03 E:0.00 Count X: 0 Y:0 Z:7214
```

Koshu, 2017-01-08:
Thought about it for a moment. In my opinion there is something wrong with that line in ABL as well as in MBL. While the calculation of the fading is quite straight forward to apply it is anything but trivial to reverse, because the result of that calculation depends on the result of the same equation. That's what that line is trying to do. My gut tells me that should be more than a one-liner. One simple solution i can think of is to do another homing after M420 (like it happens at startup). 

But of course i might be completely wrong.

chorca, 2017-01-08:
> I suppose you could try with fading disabled, and see if it still occurs?
I have had fading disabled the whole time I've been testing, as I noticed the same thing @Koshu did, that with fade enabled and unset, Z was always reset to 0, however, sending the fade parameter even with fade disabled (!) fixed that problem and enabled it to work. However, I still experienced other issues like air printing.

> what are you compiling with?
Latest arduino package, 1.8.0, running on Win64.
Compiling using the "Export compiled binary" function.
I've included output from the compiler, nothing here seems bad.
[compile.txt](https://github.com/MarlinFirmware/Marlin/files/692388/compile.txt)

Including the Configuration.h again, just so this is for sure my current one:
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/692389/Configuration.txt)

Koshu, 2017-01-08:
> I have had fading disabled the whole time I've been testing, as I noticed the same thing @Koshu did, that with fade enabled and unset, Z was always reset to 0, however, sending the fade parameter [...]

I think @manianac meant to disable it in the configuration.h

chorca, 2017-01-08:
That's what I meant, it's disabled for bilinear ABL in configuration.h.

Koshu, 2017-01-08:
In MBL if fading is disabled in the configuration.h everything works fine. Just tested:

```
>>> G28
SENDING:G28
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:0.97 E:0.00 Count X: 0 Y:0 Z:1600
>>> G1 15
SENDING:G1 15
>>> G1 Z15
SENDING:G1 Z15
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:15.00 E:0.00 Count X: 0 Y:0 Z:7214
>>> M420 S0
SENDING:M420 S0
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:18.03 E:0.00 Count X: 0 Y:0 Z:7214
>>> M420 S1
SENDING:M420 S1
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:15.00 E:0.00 Count X: 0 Y:0 Z:7214
```

chorca, 2017-01-08:
Just did another test, this time with fade ENABLED in configuration.h, and specifying the fade height in M420, but Z just runs away from the bed forever.

chorca, 2017-01-09:
Doing some more debugging tonight, noticed this:
During the Z axis runaway, the Z 'distance to go' jumps suspiciously close to the max of a signed int:
```
Recv:   Planner FR:133.33 X:311.00 (0) Y:310.00 (0) Z:17.10 (0)
Send: G1 X-1 F9000
Recv:   Planner FR:150.00 X:310.00 (-160) Y:310.00 (0) Z:0.00 (2147469968)
```
Looks like something may be rolling over. The bilinear_z_offset routine is always returning 0.00, and i'm doing some more digging.

manianac, 2017-01-09:
I've noticed as well, as issuing `M420 S1` resetting Z to zero is the only thing I've duplicated thus far.  But its only for display, issuing a
```gcode
G28
G1 Z5
M420 S1
G1 Z5
```
causes it to only drop slightly for my abl grid, so it has not lost Z position.

chorca, 2017-01-09:
Disabling bilinear subdivision seems to be the issue here.

`bilinear_z_offset`, called by `planner.unapply_leveling`, uses the `ABL_BG_SPACING` macro, which is defined as `bilinear_grid_spacing_virt` in Marlin_main.cpp.
This variable is initalized to an int array with its items set to 0. The only place it is set is in a G29 command, where it is set to `xGridSpacing/(BILINEAR_SUBDIVISIONS)`.
`bilinear_grid_spacing` is set in EEPROM along with other values, but `bilinear_grid_spacing_virt` does NOT get stored or recalled it seems, since it is not mentioned in configuration_store.cpp.
Because of this, it is never set (other than being initialized to 0), and therefore all offset calculations done without running a G29 prior to issuing will be dividing `x = RAW_X_POSITION(cartesian[X_AXIS]) - bilinear_start[X_AXIS]` by zero, like the one in `bilinar_z_offset`: `float ratio_x = x / ABL_BG_SPACING(X_AXIS)`
`bilinear_grid_spacing_virt` is only used by subdivision, not by plain bilinear mesh.

chorca, 2017-01-10:
Not sure if this is at all related, but after installing RC8, i'm getting MINTEMP errors at random times in the print, the earliest within about 10 minutes of starting. I've hooked an oscilloscope up to the thermistor lines and at 20Ms/s I'm not seeing any sort of transients or disconnects that would indicate the thermistor is the problem, not to mention this hasn't been an issue before RC8.
Looking through the code a little, but not seeing anything obvious.

thinkyhead, 2017-04-02:
Thanks, @chorca. I've just patched the issue.

closed 2017-10-31 by thinkyhead