MarlinFirmware/Marlin #5671: With eeprom enable and 2 extruders, fail compile

fbarcenas, 2017-01-08:
When you configure:
```
#define EXTRUDERS 2

#define TEMP_SENSOR_0 -1 // or the sensor than you want
#define TEMP_SENSOR_1 -1 // or the sensor than you want
#define TEMP_SENSOR_2 0
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1 // or the sensor than you want

#define EEPROM_SETTINGS
```

happen this:
```
Arduino:1.8.0 (Mac OS X), Tarjeta:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch/temperature.cpp: In function 'manage_heater':
temperature.cpp:797: error: unable to find a register to spill in class 'POINTER_REGS'
 }
 ^
temperature.cpp:797: error: this is the insn:
(insn 109 106 111 5 (set (reg:SF 110 [ D.2333 ])
        (mem:SF (post_inc:HI (reg:HI 111 [ ivtmp.126 ])) [7 MEM[base: _141, offset: 0B]+0 S4 A8])) sketch/temperature.cpp:684 100 {*movsf}
     (expr_list:REG_INC (reg:HI 111 [ ivtmp.126 ])
        (nil)))
sketch/temperature.cpp:797: confused by earlier errors, bailing out
lto-wrapper: /Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc returned 1 exit status
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
unable to find a register to spill in class 'POINTER_REGS'
```

happen in RC8 and RCBUGFIX

some idea?

emartinez167, 2017-01-08:
Try a different version of Arduino. Known issue.

Regards,

Ernesto

fbarcenas, 2017-01-08:
What version are you use?
I test
1.8.0 fail
16.13 fail

emartinez167, 2017-01-08:
If a different version won't work, try reshuffling some code. I am using
the latest. I assume you are using the one from Arduino.cc?

emartinez167, 2017-01-08:
https://www.arduino.cc

fbarcenas, 2017-01-08:
I use yes of the original website of arduino.

Nobody more happen this?

Blue-Marlin, 2017-01-08:
#5004, #5030, #5459, #5529, ...

fbarcenas, 2017-01-08:
With version 1.6.8 and WITHOUT update the boards, WORKS.

closed 2017-01-08 by fbarcenas

emartinez167, 2017-01-08:
As Makaira points out, it is a well known issue. Try downloading earlier
versions of Arduino or shuffling some code around.

gregoryfenton, 2017-02-05:
No need to jump through hoops or download earlier versions. I am compiling Marlin-1.1.0-RC8 on Windows 10, Arduino 1.8.1.

To fix this, find the function giving the error. In my case it was temperature.cpp:manage_heater()

Simply copy the function declaration and paste it directly above.
Remove the trailing { and add  
`__attribute__((__optimize__("O2")));` to the end of the line.

Before:
```
void Temperature::manage_heater() {
```
after
```
void Temperature::manage_heater() __attribute__((__optimize__("O2")));
void Temperature::manage_heater() {
```

Recompile and all is well.

I hope that helps someone.

Greg

fbarcenas, 2017-02-22:
WORKS PERFECTLY!!!!!
Problem solved, also with the last RCBUGFIX works!

I think than this change must apply in the actual rcbugfix!

pmbroth, 2017-04-24:
I tried the fix
void Temperature::manage_heater() __attribute__((__optimize__("O2")));
void Temperature::manage_heater() {

and get this:

sketch\temperature.cpp:665:69: warning: declaration of 'static void Temperature::manage_heater()' outside of class is not definition [-fpermissive]

 void Temperature::manage_heater() __attribute__((__optimize__("O2")));

ANy opinions:  Using latest ARduino and rc 1.1.0 rc8

BR

thinkyhead, 2017-04-26:
This is already fixed in `RCBugFix` (due to be released as 1.1.0 in a few days). So I suggest upgrading.