MarlinFirmware/Marlin #5673: Compile error in UBL.

Tannoo, 2017-01-09:
I am trying to give UBL a shot and have run into a snag.

If the `Z_MIN_PROBE_ENDSTOP` and the `DISABLE_Z_MIN_PROBE_ENDSTOP` are defined, this causes a sanity check to give the error:
"A probe needs a pin! Use `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` or `Z_MIN_PROBE_PIN`."

This decription doesn't imply any errors will occur:

// This option disables the use of the `Z_MIN_PROBE_PIN`
// To enable the Z probe pin but disable its use, uncomment the line below. **This only affects a
// Z probe switch if you have a separate Z min endstop also and have activated `Z_MIN_PROBE_ENDSTOP` above.**
// If you're using the Z MIN endstop connector for your Z probe, this has no effect.
`#define DISABLE_Z_MIN_PROBE_ENDSTOP`

I am using a z_min endstop switch and have a probe configured on D40 (AUX2->Pin 6) in `pins_RAMPS_14.h` file (D32 is in the AUX4 connector, which is being occupied by the LCD).

Do I need this `DISABLE_Z_MIN_PROBE_ENDSTOP` enabled or not? If I disabled this, it compiles just fine.

Note: RC8_Bugfix doesn't have this issue. I can use both with no issues. Of course, the settings are a little different.

Roxy-3D, 2017-01-09:
All of the Z-Probe and End Stop stuff was inherited from RC-7 on July 4th.   

Can you just turn off the Z-Min End Stop?  You don't really need the Z-Min End Stop if the you have a Z-Probe.

I think when ThinkyHead gets back on the grid he will know the correct way to configure things.   

Alternatively...   If we can't figure this out easily...   UBL will work without a Z-Probe.   But if we were going to configure the system without one or the other, I would turn off the Z-Min End Stop.

Tannoo, 2017-01-09:
Ok, yes I can.

Bob-the-Kuhn, 2017-01-10:
Here's the code for the Z_MIN endstop checking:
```
            #if HAS_BED_PROBE && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
              if (z_probe_enabled) UPDATE_ENDSTOP(Z, MIN);
            #else
              UPDATE_ENDSTOP(Z, MIN);
            #endif

          #endif // !Z_DUAL_ENDSTOPS

        #endif // HAS_Z_MIN

        #if HAS_BED_PROBE && ENABLED(Z_MIN_PROBE_ENDSTOP) && DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)
          if (z_probe_enabled) {
            UPDATE_ENDSTOP(Z, MIN_PROBE);
            if (TEST_ENDSTOP(Z_MIN_PROBE)) SBI(endstop_hit_bits, Z_MIN_PROBE);
          }
        #endif
```

If you want to use a pin f HAS_BED_PROBE && ENABLED(Z_MIN_PROBE_ENDSTOP) && DISABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)

Roxy-3D, 2017-01-10:
We should cut and paste that post into the Documentation !

Tannoo, 2017-01-10:
Ok, so why have the `DISABLE_Z_MIN_PROBE_ENDSTOP` option at all?

Wouldn't `//define Z_MIN_PROBE_ENDSTOP` suffice?

I also saw that defining the `Z_MIN_PROBE_ENDSTOP` will also define `Z_MIN_PROBE_PIN`.

So, if `Z_MIN_PROBE_PIN` is diabled by `DISABLE_Z_MIN_PROBE_ENDSTOP`, then you should not have a probe defined at all.

Again, that's not what this says:

// To enable the Z probe pin but disable its use, uncomment the line below. This only affects a
// Z probe switch if you have a separate Z min endstop also and have activated Z_MIN_PROBE_ENDSTOP above.
// If you're using the Z MIN endstop connector for your Z probe, this has no effect.

Tannoo, 2017-01-10:
If UBL is not fully setup for handling a Z_MIN_ENSTOP && Z_MIN_PROBE_ENDSTOP,  that's ok.

I was only asking. I don't have to run both. I do on RCBugFix. It is easy enough to move the probe's output to the Z_MIN plug and call it good for now.

Roxy-3D, 2017-01-11:
> If UBL is not fully setup for handling a Z_MIN_ENSTOP && Z_MIN_PROBE_ENDSTOP, that's ok.
> I was only asking. I don't have to run both. I do on RCBugFix. It is easy enough to move the probe's output to the Z_MIN plug and call it good for now.

No...  It should be setup for that.   RC-7 could handle that just fine.  And UBL uses all the probe functionality (unchanged) from RC-7.    I think the root of the issue is the boards silk screen is what is used to determine pin numbers and such.   And if you put stuff on a connector that is not silk screened for the function that it is connected to...  It is a little bit confusing to get things setup right.

I'm not an expert at getting the endstops and probes configured.   ThinkyHead is going to be able to explain this very easily.    But in the mean time...   Just put the Z-Probe on the Z-Min endstop and every thing should be easy (and obvious) to setup.

closed 2017-03-04 by thinkyhead