MarlinFirmware/Marlin #5675: Feature: Hotend Shutdown/Restart While Waiting for Filament Change

timeshell, 2017-01-11:
Title kind of says it all.  On a long print, the printer may come to a filament change point in code and need to wait some time before the user comes along to change the filament.  There's no point in the hotend staying on for hours until one comes along to change the filament.  In the m600 code, perhaps there could be code to shut down the hotend in when #defined HOTEND_DOWN_DURING_FILAMENT_CHANGE.

Tannoo, 2017-01-11:
Ooooh... That would be nice.

Roxy-3D, 2017-01-11:
That would be almost trivial to add.   Maybe a 5 minute default until the nozzle is turned off?

Tannoo, 2017-01-11:
I wouldn't think so. 5 minutes is a long time with no filament movement. I would be more worried about the heat traveling past the heatbreak and causing issues.

Tannoo, 2017-01-11:
Just the option to use or not use would be nice.

Roxy-3D, 2017-01-11:
> I wouldn't think so. 5 minutes is a long time with no filament movement. I would be more worried about the heat traveling past the heatbreak and causing issues.

Actually....  to your point...   It is worse than that.   The M600 retracts the filament out of the nozzle so the only filament left is the stuff that is stuck to the inside wall of the nozzle.   

I suspect my nozzle would be OK with a 5 minute delay.   After all, the nozzle is going to hold temperature at the 'correct' number during that time.   But still....   it would be easy to make it user configurable (probably in Configuration_adv.h).

I am out of town without my printer.   I have no way to test code right now.   But I can take a look at it and see how bad the changes are to do this.   If it isn't too complex, I might be able to post the changes for you to try out and verify.

Tannoo, 2017-01-11:
Ok. I am eagerly waiting. ;)

Roxy-3D, 2017-01-11:
Turning it off is easy...    We just modify this loop to also watch for the default time out for the nozzle:
```cpp

    while (wait_for_user) {
      #if HAS_BUZZER
        millis_t ms = millis();
        if (ms >= next_buzz) {
          BUZZ(300, 2000);
          next_buzz = ms + 2500; // Beep every 2.5s while waiting
        }
      #endif
      idle(true);
    }
```
But then...   If the nozzle has been turned off, we will need to add a loop (much like UBL's G26 nozzle and bed preheat) to get it back up to temperature prior to continuing.  This is the UBL code to get the bed and nozzle up to temperature.   Obviously, the bed code can be removed.   And probably the messages and confirmation should be changed.

```cpp
bool turn_on_heaters() {
#if HAS_TEMP_BED
  #if ENABLED(ULTRA_LCD)
    if ( bed_temp > 25 ) {
      lcd_setstatus( "G26 Heating Bed.  ", true);
      lcd_quick_feedback();
  #endif
      UBL_has_control_of_LCD_Panel++;
      thermalManager.setTargetBed( bed_temp );
      while( abs(thermalManager.degBed()-bed_temp) > 3 ) {
  	if ( G29_lcd_clicked() ) {
		strcpy( lcd_status_message, "Leaving G26    "); // We can't do lcd_setstatus() without having it continue;
  		while ( G29_lcd_clicked() )			// Debounce the switch
			idle();
            	lcd_setstatus( "Leaving G26     ", true);	// Now we do it right.
		return 1;
	}
	idle();
      }
  #if ENABLED(ULTRA_LCD)
    }
    lcd_setstatus( "G26 Heating Nozzle. ", true);
    lcd_quick_feedback();
  #endif
#endif
 
//
// Now turn on the hotend (nozzle) and wait for it to come up to 
// temperature.

  thermalManager.setTargetHotend( hotend_temp , 0 );
  while( abs(thermalManager.degHotend(0)-hotend_temp) > 3 ) {
  	if ( G29_lcd_clicked() ) {
		strcpy( lcd_status_message, "Leaving G26    "); // We can't do lcd_setstatus() without having it continue;
  		while ( G29_lcd_clicked() )			// Debounce the switch
			idle();
            	lcd_setstatus( "Leaving G26     ", true);	// Now we do it right.
		return 1;
	}
	idle();
  }

#if ENABLED(ULTRA_LCD)
  lcd_setstatus( "                  ", true);
  lcd_quick_feedback();
#endif
  return 0;
}
```

timeshell, 2017-01-11:
Great!  Thank you.

Roxy-3D, 2017-01-19:
OK!    I have the first pass done.   I'm sorry it took so long.   I'm skiing and by the end of the day I'm so tired I can barely layout my cloths for the next day.   Right now I have the nozzle time out set for 5 minutes (in Configuration_adv.h)   You change that to 10 seconds to verify correct operation if you don't want to wait that long.

It is very likely it won't work the first time.   But it should be close.  If you tell me how it fails, we can probably get it working pretty quickly.   You can download the code at:  https://github.com/Roxy-3D/Marlin

timeshell, 2017-01-19:
Very much appreciated.  I'll give it a try and let you know.   Thank you!

timeshell, 2017-01-19:
Ok, this is what we have so far.  It turns off the heaters after the timeout.  It would be nice if it showed the temperature during this process.  However, when I click to reheat, the machine just shuts down within seconds.

Roxy-3D, 2017-01-20:
> However, when I click to reheat, the machine just shuts down within seconds.

Yes...   I realized  this morning I was missing a call to idle(); in the warm up loop.  I didn't have any way to get to a computer and fix that until just now.  Without the call to idle(); the watchdog timers will trigger and shut down the machine.   

I believe it should be showing the temperature during both the wait (and cool down) as well as the re-heat period.   I have the idle() call inserted.   If you can give it another try...   It might just work for you now.

@Tannoo   Are you following this thread still?   It might be ready for you to jump in and test it.

Tannoo, 2017-01-20:
Yes I am. And I will.

timeshell, 2017-01-20:
I'm using the RepRapDiscountSmartGraphicsController.  When M600 is sent, the whole screen goes to the changing filament message.  It doesn't show any stats on the heaters or position or anything else during that time.

I'll add the idle() call and try again.

Roxy-3D, 2017-01-20:
> When M600 is sent, the whole screen goes to the changing filament message. It doesn't show any stats on the heaters or position or anything else during that time.

Oh!   I understand what you are saying...   I added a couple of messages:
```cpp
      lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_CLICK_TO_HEAT_NOZZLE);
      lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_WAIT_FOR_NOZZLES_TO_HEAT);
```
I think what you are saying is the Filament Change code takes over the entire LCD Panel and right now we are losing the temperature and position numbers.    It is only being used to display simple instructions to the user and not display the machine's status.  When I get home and can look at things on a live printer, I suspect we can easily get those numbers back on the display.   But I wouldn't dare try to do that without having a printer to look at.

Please see if the new messages make sense and accurately reflect the state of the machine.

Tannoo, 2017-01-20:
Hopefully we don't loose the stepper positioning during this event.

Tannoo, 2017-01-20:
Wow.. M600 is not working at all for me.

Roxy-3D, 2017-01-20:
In Configuration_adv.h you can set the default stepper deactivation time.   You may want to bump it up.
```cpp
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
```
But...  I wonder if the M600 filament change command should over ride this.   If we are waiting for the user to do a filament change, should we keep the steppers locked in position?   It would seem to me that would make sense.

Does anybody have any strong feelings on this thought?

Tannoo, 2017-01-20:
I have a stronger feeling of where to put `tmps` as:

`thermalManager.setTargetHotend( tmps[iii] , iii );`

tmps was not declared in this scope.

timeshell, 2017-01-20:
I would strongly say yes, lock the steppers while the filament change is occuring.  My client can go for very prolonged periods before getting to change the filament.  It would be very bad if the position changed because the motors were no longer locked.

timeshell, 2017-01-20:
@Tannoo It's supposed to be "temps"

Tannoo, 2017-01-20:
ok...Roxy must have been typing too fast. lol

timeshell, 2017-01-20:
You said that already....  :)

timeshell, 2017-01-20:
Also Tannoo, don't forget to enable:

  #define FILAMENT_CHANGE_FEATURE             // Enable filament exchange menu and M600 g-code (used for runout sensor too)

timeshell, 2017-01-20:
Well, seems to work now.  I haven't used it mid print yet, but so far so good.  Looking forward to display data during the filament change.

Roxy-3D, 2017-01-20:
>> But... I wonder if the M600 filament change command should over ride this. If we are waiting for the user to do a filament change, should we keep the steppers locked in position? It would seem to me that would make sense.

> You said that already.... :)

I'm kind of soliciting feedback about how badly people will react if 
```cpp
#define DEFAULT_STEPPER_DEACTIVE_TIME 120
```
doesn't always do what it is specified as.   If we over ride this timeout, we aren't doing what we say we will do.   But to your point....    I don't think it makes sense to lose position just because the user takes 10 minutes to change the filament.

If somebody has a strong feeling about this, PLEASE SPEAK UP NOW!   Please explain why it gives you heartburn!

timeshell, 2017-01-20:
Answers are above after you asked the first time...

timeshell commented 26 minutes ago
I would strongly say yes, lock the steppers while the filament change is occuring. My client can go for very prolonged periods before getting to change the filament. It would be very bad if the position changed because the motors were no longer locked.

Tannoo, 2017-01-20:
A little humor never hurts anything. :P

timeshell, 2017-01-20:
Very true.  Some of us aren't as good at humor than others...(like me)

Roxy-3D, 2017-01-20:
> A little humor never hurts anything. :P
Agreed...   

I'll start looking at how complicated it will be to add the control:
```cpp
#define STEPPER_MOTORS_DONT_TIMEOUT_DURING_FILAMENT_CHANGE
```
That's not a joke!

Tannoo, 2017-01-20:
I sent `M18 S1000` at the start of the print.
I went to watch some other programming on a TV.
I sent `M600` near the end of the print.
I went to finish watching the programming I started earlier while listening to an annoying beep. :P
I came back (6 mins?) to check the nozzle temp by a piece of filament first...it was cold.
Steppers were still active.
Changed the filament and clicked the screen.
Waited and it started extruding and display was just as this was not an issue.
So...it works for me also. It works flawless.

Adding the hotend temp on the filament change screen would be a good thing.

Plus:
- I didn't see the message that it was unloading any nozzle.
- I did see the message `Click to heat nozzle`. I changed that to say `Click to heat nozzle(s)`.
- I did see the message `Wait to heat nozzle`. I changed to say `Wait to heat nozzle(s)`.

Roxy-3D, 2017-01-20:
I'm not sure it is valid to change multiple nozzle filaments with a single M600.   The reason is it is very typical to lower the temperature of all nozzles that are not the active tool at the moment.   I'm not sure the lower temperature is appropriate for retracting and removing the filament and it probably is not suitable for starting the new filament.   

> I didn't see the message that it was unloading any nozzle.
```cpp
   stepper.synchronize();
    lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_UNLOAD);

    // Unload filament
  ...
    // Synchronize steppers and then disable extruders steppers for manual filament changing
    stepper.synchronize();
    disable_e0();
    disable_e1();
    disable_e2();
    disable_e3();
    delay(100);
...
    // Wait for filament insert by user and press button
    lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_INSERT);
```
The retraction is very short in time.   Are you sure you didn't just miss it?   But I wonder if we need an idle(); call right after the 
```cpp
    lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_UNLOAD);
```
to get it up onto the display before it gets over written by the next  lcd_filament_change_show_message() call ????

Tannoo, 2017-01-20:
stepper.synchronize();
    lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_UNLOAD);

    // Unload filament
    if (code_seen('L')) destination[E_AXIS] += code_value_axis_units(E_AXIS);
    #if defined(FILAMENT_CHANGE_UNLOAD_LENGTH) && FILAMENT_CHANGE_UNLOAD_LENGTH > 0
      else destination[E_AXIS] -= FILAMENT_CHANGE_UNLOAD_LENGTH;
    #endif

    RUNPLAN(FILAMENT_CHANGE_UNLOAD_FEEDRATE);

    // Synchronize steppers and then disable extruders steppers for manual filament changing
    stepper.synchronize();
    disable_e0();
    disable_e1();
    disable_e2();
    disable_e3();
    delay(100);

I can see it being really quick if the `FILAMENT_CHANGE_UNLOAD_LENGTH` is really short.
When I used filament change in RCBugFix, it took about 10 secs or so to unload.

Dunno what happened to this one yet.

timeshell, 2017-01-20:
I saw the message Wait for filament unload until the timeout occurred and then it switched to Click to heat nozzle.

Tannoo, 2017-01-20:
I will have to look again with my good eye.

Roxy-3D, 2017-01-20:
Well....  It is harmless to put a call to idle(); right after the 
```cpp
    lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_UNLOAD);
```
I'll change that just to eliminate any possible loss of the message.   But remember, the LCD Panel only gets refreshed every 150 ms or so.   And it takes time for the LCD crystals to get dark.   I think it is possible to miss a fast message.

I'll start the work to optionally disallow stepper motor timeouts (and power downs) during filament change.  That might be ready in a few days for you guys to test.   Do either of you have a dual nozzle machine to test the current code on?   It should work.  But I would feel better if somebody actually verified it works on multiple extruder machines.

Roxy-3D, 2017-01-20:
> I have a stronger feeling of where to put tmps as:
> thermalManager.setTargetHotend( tmps[iii] , iii );
> tmps was not declared in this scope.

Now fixed!

Tannoo, 2017-01-21:
>Do either of you have a dual nozzle machine to test the current code on?

Not I.

timeshell, 2017-01-21:
I do not, yet.

Roxy-3D, 2017-01-21:
@Tannoo  @timeshell
     
> I would strongly say yes, lock the steppers while the filament change is occuring. My client can go for very prolonged periods (hours up to a day maybe) before getting to change the filament. It would be very bad if the position changed because the motors were no longer locked.

OK!   There is a new configuration option in Configuration_adv.h called:
```cpp
    #define STEPPER_MOTORS_DONT_TIMEOUT_DURING_FILAMENT_CHANGE    // Enable to make stepper motors hold position during filament change even if it 
                                                                  // takes longer than DEFAULT_STEPPER_DEACTIVE_TIME
```

Unfortunately, I've been doing this in the RC branch of the fork instead of RCBugFix!  Or well, you can get it here:  https://github.com/Roxy-3D/Marlin/tree/RC

Tannoo, 2017-01-21:
I can this evening.

Roxy-3D, 2017-01-22:
Please test using the RCBugFix version of the code.   I've moved all the changes over to RCBugFix now.

You can get it here:  https://github.com/Roxy-3D/Marlin/tree/RCBugFix

Roxy-3D, 2017-01-22:
> Ok, it works as intended. 

:)

> The only thing left to make it perfect, is to display the hotend temp during cooldown and heatup.

And that can't happen until I have a printer to look at.    But there is a chance it is as easy as grabbing the current Status Screen code lines and having them do their magic at a specified location of the new screens.

Tannoo, 2017-01-22:
> And that can't happen until I have a printer to look at.

I can send you a picture of mine. :)

Roxy-3D, 2017-01-23:
Pull request for first phase has been submitted:    https://github.com/MarlinFirmware/Marlin/pull/5737

Tannoo, 2017-01-23:
I can't seem to clone or fork your repo, @Roxy-3D, maybe because I already have a fork of marlin? Anyway, I changed these in `ultralcd.cpp`:

      void lcd_filament_change_wait_for_nozzles_to_heat() {
      START_SCREEN();
      STATIC_ITEM(MSG_FILAMENT_CHANGE_HEADER, true, true);
      STATIC_ITEM("Wait to heat nozzle");
      STATIC_ITEM("");
      STATIC_ITEM("   Nozzle On: ", false, true, itostr3left(thermalManager.degHotend(0)));
      END_SCREEN();
    }

      void lcd_filament_change_heat_nozzle() {
      START_SCREEN();
      STATIC_ITEM(MSG_FILAMENT_CHANGE_HEADER, true, true);
      STATIC_ITEM(MSG_FILAMENT_CHANGE_INSERT_1);
      #ifdef MSG_FILAMENT_CHANGE_INSERT_2
        STATIC_ITEM(MSG_FILAMENT_CHANGE_INSERT_2);
      #endif
      #ifdef MSG_FILAMENT_CHANGE_INSERT_3
        STATIC_ITEM(MSG_FILAMENT_CHANGE_INSERT_3);
      #endif
      STATIC_ITEM("   Nozzle Off: ", false, true, itostr3left(thermalManager.degHotend(0)));
      END_SCREEN();
    }

This in `marlin_main.cpp`:

      while (wait_for_user) {
      millis_t ms = millis();
      if (nozzle_timed_out == true)
        lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_CLICK_TO_HEAT_NOZZLE);
      #if HAS_BUZZER
        if (ms >= next_buzz) {
          BUZZ(300, 2000);
          next_buzz = ms + 2500; // Beep every 2.5s while waiting
        }

And this:

    KEEP_CHECKING_TEMPS:
    idle();
    for( iii=0; iii<HOTENDS; iii++){
      if (abs(thermalManager.degHotend(iii)-temps[iii]) > 3 ) {
        lcd_filament_change_show_message(FILAMENT_CHANGE_MESSAGE_WAIT_FOR_NOZZLES_TO_HEAT);
        goto KEEP_CHECKING_TEMPS;
      }
    }

This will only show one hotend. I'm not sure how to show the current hotend on muli-hotend setups.
I'm sure you have a more elegant way of implementing this. So, by all means...

Roxy-3D, 2017-01-23:
I've added your changes to https://github.com/Roxy-3D/Marlin/tree/RCBugFix

You should be able to go to that page and do a "Clone or Download" and at worst case pull down a .ZIP file of it.   But if you delete your fork of it, you should be able to re-fork and start over.

I made a small change to your code.  Right now, in the code you posted, you have it hard coded to be the first nozzle:  thermalManager.degHotend(0).   Instead of displaying extruder 0's temperature, I changed it to display the active extruder's temperature.     Probably...  if we are only going to show one nozzle, we should be displaying the active nozzle.

Tannoo, 2017-01-23:
Yeah, I was going to PR your branch....I just couldn't get that to work. I have several PR's on my branch. That would screw things up if I deleted it, correct?

I'm sure that I'm done modding things for a little while. I have to get back to printing... I have plastic that is getting cold!

Roxy-3D, 2017-01-23:
> Yeah, I was going to PR your branch....I just couldn't get that to work. I have several PR's on my branch. That would screw things up if I deleted it, correct?

Yeah...   Not to worry...   I moved those changes you posted over to my branch with the M600 work.   I'll probably re-issue the Pull Request with the thermal stuff included.   But I need confirmation that it works as expected prior to doing that.   If you could change the 0 to active_extruder in those two places, that would speed up the confirmation process.

Tannoo, 2017-01-23:
I did....plus, you need to disable the filament change feature for travis.

Roxy-3D, 2017-01-23:
> plus, you need to disable the filament change feature for travis.

I don't understand.  It is turned off by default in Configuration_adv.h    Are you saying I need to change something in the Travis configuration?

Tannoo, 2017-01-23:
Latest travis-build failed.

Marlin_main.cpp:7256: error: unterminated #if
 #if ENABLED(FILAMENT_CHANGE_FEATURE)
 ^
exit status 1

Tannoo, 2017-01-24:
I had been appling your commits to the version I am running. As I have all of my PRs applied and running.

Tannoo, 2017-01-24:
>If you could change the 0 to active_extruder in those two places, that would speed up the confirmation process.

I have and verified that the screens still display the temps as before. YAY!

Tannoo, 2017-01-24:
In`ultralcd.cpp`, I changed this:

      void lcd_filament_change_wait_for_nozzles_to_heat() {
      START_SCREEN();
      STATIC_ITEM(MSG_FILAMENT_CHANGE_HEADER, true, true);
      STATIC_ITEM("Wait to heat nozzle");
      STATIC_ITEM("");
      STATIC_ITEM("");
      STATIC_ITEM("   Nozzle On: ", false, true, itostr3left(thermalManager.degHotend(0)));
      END_SCREEN();
    }

to this to center the message on the screen:

      void lcd_filament_change_wait_for_nozzles_to_heat() {
      START_SCREEN();
      STATIC_ITEM(MSG_FILAMENT_CHANGE_HEADER, true, true);
      STATIC_ITEM("");
      STATIC_ITEM("Wait to heat nozzle");
      STATIC_ITEM("");
      STATIC_ITEM("   Nozzle On: ", false, true, itostr3left(thermalManager.degHotend(0)));
      END_SCREEN();
    }

Roxy-3D, 2017-01-24:
Yeah...  but I think we want the (0) to be (active_extruder) so we are messing with the 'correct' extruder.

Tannoo, 2017-01-24:
Corrected.

Your PR is missing the #endif for the buzzer.

Tannoo, 2017-01-24:
The `#endif // FILAMENT_CHANGE_FEATURE` becomes the #endif //HAS_BUZZER instead and no #endif for the filament change feature.

Roxy-3D, 2017-01-24:
Yeah...   I messed up the #ifdef nesting when I cut and pasted your code.   I'm generating a new Pull Request right now.

Roxy-3D, 2017-01-24:
Travis seems to be happy again:

> All checks have passed
> 1 successful check

> continuous-integration/travis-ci/pr — The Travis CI build passed
> Details
> This branch has no conflicts with the base branch
> Merging can be performed automatically.

[65 more replies]

closed 2017-10-31 by thinkyhead