MarlinFirmware/Marlin #5677: M110 acting strange

duisenberg, 2017-01-11:
```
M110 N24
ok
N25 M105*16
Error:Line Number is not Last Line Number+1, Last Line: 1
Resend: 2
ok
M110 N25
ok
N25 M105*16
Error:Line Number is not Last Line Number+1, Last Line: 1
Resend: 2
ok
N2 M115*36
FIRMWARE_NAME:Marlin 1.1.0-RC8 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:ANYCUBIC EXTRUDER_COUNT:1 UUID:8112eb85-d15b-460e-8089-f83b108660bf
ok
M105
ok T:14.4 /0.0 B:15.6 /0.0 @:0 B@:0
N0 M110 N1*124
ok
N1 M105*38
Error:Line Number is not Last Line Number+1, Last Line: 1
Resend: 2
ok
```
Something seems to be broken with M110, the short command w/o checksum seems to reset to line 1 regardless of its argument, the checksum version seems to set to line+1.

manianac, 2017-01-13:
What version are you running?  You may wish to try RCBugfix.

I just tested this, and its working fine for me.

```gcode
Send: N50 M110 N50*125
[...]
Send: N51 M105*19
Recv: ok T:25.1 /0.0 B:25.0 /0.0 @:0 B@:0
```

manianac, 2017-01-13:
Alright I'm pretty positive the beginning of the `get_serial_commands` is picking up the N parameter as an actual line number, and trying to read after the parameter (usually \0 outside the buffer) [here](https://github.com/MarlinFirmware/Marlin/blob/0c0c840ba8fbff410e2e9add31d8f4fe23565641/Marlin/Marlin_main.cpp#L1012)

Only ideas I have are drop support for M110 without a line number/checksum, or rewrite the serial handler for a specific case of M110 without a line number.  Probably should wait for @AnHardt or @thinkyhead to chime in.

duisenberg, 2017-01-13:
The issue might be related more to [gcode_M110](https://github.com/MarlinFirmware/Marlin/blob/0c0c840ba8fbff410e2e9add31d8f4fe23565641/Marlin/Marlin_main.cpp#L5683) where M110 sets gcode_N but leaves gcode_LastN untouched. 

Changing the code at 5683 to
```if (code_seen('N')) { gcode_N = code_value_long(); gcode_LastN = gcode_N-1; }```
should do.

manianac, 2017-01-13:
Ah my mistake, I completely misunderstood the flow.  Write up a PR so it can get tested / approved.

AnHardt, 2017-01-13:
Some features are used so rarely, to make it possible, a bug survives more than 10 month. ;-)
#5691

duisenberg, 2017-01-18:
Sorry for the delay. Did not realize what PR meant, never done such a thing until now :) Hope the RCBugfix branch was the one it should go into. Do I close the issue if the pull request is done?

closed 2017-03-04 by thinkyhead