MarlinFirmware/Marlin #5687: Auto Bed Levelling RC8 / Bugfix Support

leecuk, 2017-01-12:
Hi All,

Hopefully someone can help me please. I've been playing around with Auto Bed Levelling in RC8 on which I've been having issues. I've just tried the latest BugFix branch and still having the same problem.

The issue is that the compensation does seem correct for the bed and always seems to be too close to the bed on the far right of the bed - although you can see Z altering thoughout the move.  I'm using a borosilicate glass bed with a capacitive sensor.

I've enabled debug and attached results from G28 and G29. 

Can someone point me in the right direction please? :-)

Thanks
Lee

[image]
[results.txt](https://github.com/MarlinFirmware/Marlin/files/702889/results.txt)

Tannoo, 2017-01-13:
Give the UBL branch a try, see if you have the same issues there.

strips, 2017-01-13:
Interesting. I did not notice the UBL branch before. Thanks @Tannoo !

Do you know if UBL will work with dual z endstops at max? Have a IR probe as well.

strips, 2017-01-13:
Hmm... devel-ubl branch is old. Latest commit was 22. November. I'll stick to RC8 for now.

Tannoo, 2017-01-13:
Sounds like she may have it nailed down. She needed more testing though.

Roxy-3D, 2017-01-14:
> Sounds like she may have it nailed down. She needed more testing though.

I'm working on getting UBL to use RC-8 as the base.   But it isn't done yet.   And I'm out of town so progress isn't fast.

Foxandxss, 2017-01-21:
For what is worth, I have this cheap chinese ripoff with crap hard with a modified marlin for my board (pretty much the same, minimum changes) and I was always noticing the same issue, right side of the bed is closer than the left part. I can lower the bed on the right but after G29, it will make it close again.

I thought it was me only, but unsure if related, due to bad hard in my part.

tcm0116, 2017-02-04:
I have a similar setup and noticed the same issue. Z is moving during an X movement from left to right, but the nozzle ends up closer to the bed on the right side than on the left. I wonder if I raised up the left side of the bed if the issue would switch to the left, indicating that there may be some sort of scaling error in the mesh.

Roxy-3D, 2017-02-04:
> Do you know if UBL will work with dual z endstops at max? Have a IR probe as well.

Yes.   I'm in the process of moving the UBL code base to RC-8 right now.

b3niup, 2017-02-07:
I have similar issue, but it only affects front of the bed. On it's back correction seems to be fine.

benlye, 2017-02-07:
Same issue for me - I also get the nozzle closer to the bed on the right side, while the left is almost perfect.

On RC8-BugFix and using bilinear levelling with 7 points, and I can see that Z correction is definitely doing _something_ but it doesn't seem to provide enough correction.  

Happy to provide as much info as is useful or run any tests.

tcm0116, 2017-02-08:
I took a look through the Bilinear Leveling code in RCBugFix, and I think I've identified a few issues with the computation of the z offset. Basically, it was not calculating the z offset correctly for points outside of the probed area or even for points within the most significant grid boxes. I've put together a commit that I think should address those issues and hopefully result in a better result. Can anyone give this a try and see if it makes any improvements?

https://github.com/tcm0116/Marlin/commit/7575ae3a569b5d8dd5f9b6d37d8db3d2299cd8e5

b3niup, 2017-02-08:
It's a bit better now but still nozzle on the front left bed side is closer to the bed (too close) then on the right side (too far). 

But this issue not only regards bilinear mode because it's not working properly for linear too.

Blue-Marlin, 2017-02-08:
@b3niup 
In this case it could be a twisted axis what messes up the measurement.

b3niup, 2017-02-08:
@Blue-Marlin
Whole printer is pretty square now. But since leveling sensor is fixed with extruder block measurements should always be consistent with real distance from extruder to the bed, right?

benlye, 2017-02-08:
@tcm0116 I just tried your patch and I'm sorry to say it seemed to make the imbalance I see slightly worse rather than better.

cosmith71, 2017-02-09:
I'm having a similar/same problem.  Facing the front of the machine, my LEFT is closer to the bed when using bed leveling.  The right side is fine, and then some imaginary line is crossed and the nozzle is too close to the bed to even extrude.  I can see the Z axis changing, but the compensation must be off.  The frustrating thing is, it did this with the last official release but seemed to work correctly with RC8, until suddenly it didn't anymore.

I'm using a Geeetech i3 Pro C, a dual extrusion Prusa clone with a 3D Touch sensor (BLTouch clone).

Thanks for all your hard work on this!

tcm0116, 2017-02-09:
@cosmith71 can you give the change I posted above a try?

cosmith71, 2017-02-09:
I'm willing to give it a shot, but I have an issue with BUGFIX.  It doesn't deploy the sensor probe on a G28 command.  Am I missing something?  RC8 works fine in this regard.

cosmith71, 2017-02-10:
@tcm0116 , your fix seemed to fix my problem, but so did the latest RCBugfix.  In my limited testing I can't tell much of a difference.

tcm0116, 2017-02-13:
@cosmith71 if you bed is relatively flat, then I don't expect that you'd really notice a difference.

I went ahead and made a pull request (#5822) since I think this does at least correct some errors in the computation of the bilinear z-offset.

cosmith71, 2017-02-13:
I figured that was the case.  I use glass, and it probably wasn't a good test.  I'm glad you can help others with it, though.

closed 2017-10-31 by thinkyhead