MarlinFirmware/Marlin #5693: Auto leveling issues

caladrien, 2017-01-13:
Hi there,

I was working my way through the internet to find a solution for my problem. Even many people having the some sort of same issue I couldn’t resolve it until now.

I’m using the latest Marlin build and I want to integrate my Z prob.

I did the installation according the Marlin official discretion: Putting the offset value via the Menu into the machine. My offset was -2.75. After running “Home all” I now have 12.75 Z on my display. Fine until now. When I’m doing a G0 Z0 my paper sticks a little bit. Perfect. But when I print the first layer will start at Z 0.02. So nozzle is a bit of bed and display is saying: Z 0.2.
 
I read that it could have to do with only using G28 without G29 so I tried that as well with no result. 
My reworked start code. It is more or less the curas’ default. Just added a G29:

`G21        ;metric values
G90        ;absolute positioning
M107       ;start with the fan off
G28  
G29     
G1 Z15.0 F{travel_speed} ;move the platform down 15mm
G92 E0                  ;zero the extruded length
G1 F200 E3              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F{travel_speed}
;Put printing message on LCD screen
M117 Printing...`

I thought also about just removing those start printing z 0.2 from my offset. I actually did try and it work - at least a bit. But this seems to be ceating on the machine and not what I was expecting of a auto leveling engine. 

Whatever I do, in the beginning of any print my display is telling me that z is at 0.2. Any idea why? Is there a "TAPE_ON_BED" mode? 

For any help I would be relay glad.

Roxy-3D, 2017-01-15:
is your layer height set at .2mm ?   That is where the nozzle will be for the first layer.

caladrien, 2017-01-15:
When you say that it totaly makes sense. Dumm question.

But sill, I'm why to far away from the bed. More than 0.2mm.

Roxy-3D, 2017-01-15:
> But sill, I'm why to far away from the bed. More than 0.2mm.

You need to adjust the machine's Z_PROBE_OFFSET_FROM_EXTRUDER to be correct.  You can do that in Configuration.h or you can use the M851 command.    Either way...  You should probably do a M500 when you get the settings correct.   (If you modify it in Configuration.h do a M502 to force those numbers to be loaded prior to testing and doing the M500)

caladrien, 2017-01-15:
But the z_offset, which I enter over the Printer UI, is a diffrent than in the config.h? I'll try later. Thank very much you!

Foxandxss, 2017-01-21:
What you write in here is completely fine. The offset will be the difference between your nozzle tip and the probe tip. Also if you have a glass, you will add the glass thickness to that.

So even if you get `-2.75` you will probably need to bump it a little. Perhaps `.2.90` or something like that, until you get that perfect first layer.

The fact that the print starts at 0.2 is completely normal, that is the height of your first layer and it can't be 0, 0 would be completely touching the bed and no PLA would get out of there.

thinkyhead, 2017-04-02:
> between your nozzle tip and the probe tip

After homing move the Z axis down slowly until the probe triggers. (You can check this with `M119`) Take the Z coordinate reported in `M114`, negate it, and that is your probe offset.

closed 2017-04-02 by thinkyhead