MarlinFirmware/Marlin #5694: Manual bed level nozzle crash in RC8

strips, 2017-01-14:
I tried to redo the config file from scratch and still get an crash. 

I have dual Z max endstop leveling working fine when doing a G28. G30 single probe also works fine with IR probe. 

I have enabled MESH_BED_LEVELING and MANUAL_BED_LEVELING.

When I run Level Bed from display menu it will do a full YX min and Z max homing. Then Z goes all the way to ~0 and the IR probe triggers. Then the carriage moves a tad towards bed center and nozzle crashes. Nozzle is about 0.9mm from bed when IR probe triggers and I have Z offset at 0.8mm.

[Configuration.h.txt](https://github.com/MarlinFirmware/Marlin/files/705654/Configuration.h.txt)

psavva, 2017-01-14:
If you have a z probe, then use Bilinear bed leveling.

It's basically mesh bed leveling with probing automatically with the probe

strips, 2017-01-14:
Think I'll give it a try. Isn't there a problem with saving the mesh with Bilinear on RC8?

psavva, 2017-01-14:
Use RCBugFix, some issues were resolved that existed in RC8

strips, 2017-01-17:
Will give RCBugFix a go before trying Bilinear. manual mesh level should work even if Bilinear might work better.

psavva, 2017-01-17:
Yes, give it a try, and if the issue persists we will look into it further.

Many bugs are fixed in RcBugFix for the bed levelling routine.

thinkyhead, 2017-04-02:
Should be in pretty good shape now in the current code-base. Re-open if the problem recurs.

closed 2017-04-02 by thinkyhead