MarlinFirmware/Marlin #5700: need advice on AT90USB startup messages - delayed printing?

Bob-the-Kuhn, 2017-01-14:
While playing with my AT90USB1286 based Teensylu I noticed that the startup messages displayed in Repetier Host were missing a large chunk of what I see on my Atmega2560.

I traced it down to Marlin printing to the serial port before the USB connection was established.  On the non-AT90USB boards there's a chip between the host and the processor.  The chip handles all the USB link startup stuff BEFORE the main chip is reset.

How do I delay the prints on the AT90USB boards?

I'm hoping that there's a semi-easy way to redirect the prints to a buffer and then, when the link is up,  send them out the serial port.  

The only other option I see is to modify some existing code.

Since this is mostly a cosmetic item that's been there forever I don't want to put in a lot of time fixing it.

======================================================

Here's an overview of the Marlin startup process
```
RESET ISR activates STARTUP()
init serial port
print some messages
some setup
Config_RetrieveSettings();
    print source of config info
    print config info
enable interrupts
more setup
```

There's a function in most AT90USB libraries I can query to determine when the link is up.
- If I put it before the interrupts are enabled then the board will hang if using an LCD because the USB link is never established.
- If I put it at the end of STARTUP (after the interrupts are enabled) and move the print statements there then the board should be functional if using an LCD.  I'm pretty sure the **Config_RetrieveSettings();** routine can't be moved which means it's print statements will be lost if the USB link is ever established.  Hmmm ... maybe I could **Config_RetrieveSettings();** to add a flag that'll tell me which message to print.  I'd like to avoid modifying other code if I can help it.  The code for this option would probably look like this:

```
RESET ISR activates STARTUP()
init serial port
some setup
Config_RetrieveSettings(STARTUP_MODE);
    set message flag
    if !STARTUP_MODE
        print as usual
enable interrupts
more setup
if AT90USB 
   wait for link
print messages
```

closed 2017-01-20 by Bob-the-Kuhn

thinkyhead, 2017-02-12:
Still an extant issue?

Bob-the-Kuhn, 2017-02-12:
It's a "nice to do" item.  

Unless you have an elegant solution I think it's best just to leave things as they are.

thinkyhead, 2017-02-14:
Sounds good.

> ======================================================

Four dashes "----" is easier to type and makes a nice line like this:

----

Cool, eh?