MarlinFirmware/Marlin #5703: RC8 BugFix 1.1.0 LCD 12864 RepRap Full Graphic Smart Controller shifted

lukeskymuh, 2017-01-15:
My display is sometimes sifted to the right afer a few minutes. Is this a bug, a hardware problem or anything else?

configuration.h file:
//
// RepRapDiscount FULL GRAPHIC Smart Controller
// http://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller
//
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

[image]

NiklasOlofsson, 2017-01-15:
I would say hardware. I use the same as you (bigtree) and it works as it should after cutting away the tab to be able to insert the flat cables the other way around. ;) Using GT2560 as controller board.

shacal, 2017-01-15:
I have seen this on some controllers, and was fixed by hand tuning delays.

#define ST7920_DELAY_1 DELAY_0_NOP
#define ST7920_DELAY_2 DELAY_1_NOP
#define ST7920_DELAY_3 DELAY_2_NOP

lukeskymuh, 2017-01-15:
Thank you shacal. Looks good up to now.

closed 2017-01-15 by lukeskymuh

jarnoburger, 2017-03-15:
i thought that i only had to uncomment one define st920-blabla.. but you need to uncomment all of them

trikko, 2017-10-04:
I have exactly the same problem with my LCD. How did you solve the issue?

Thanks in advance,
Andrea

marcio-ao, 2018-02-13:
For some reason the delays in ST7920_WRITE_BYTE in  "ultralcd_st7920_u8glib_rrd.h" were removed in Marlin 2.0. I fixed the timing issue by reintroducing them:

```
static void ST7920_WRITE_BYTE(uint8_t val) {
  for (uint8_t i = 0; i < 8; i++) {
    WRITE(ST7920_DAT_PIN, val & 0x80);
    ST7920_DELAY_2;
    WRITE(ST7920_CLK_PIN, HIGH);
    ST7920_DELAY_1;
    WRITE(ST7920_CLK_PIN, LOW);
    ST7920_DELAY_3;
    val <<= 1;
  }
}
```
The problem is that the datasheet for the ST7920 specifies a minimum SCLK period and hold times and this function is just toggling the hell out SCLK without regards for timing. Here are the required values:

- Serial clock cycle : 400 ns
- SCLK high pulse width: 200 ns
- SCLK low pulse width: 200 ns
- SID data setup time: 40 ns
- SID data hold time: 40 ns
- CS setup time: 60 ns
- CS hold time: 60 ns

An instruction on a 16MHz AVR takes 62ns, so at minimum about four instructions must elapse before toggling the ST7920_CLK_PIN. Using 4 for all the delays would be guaranteed to work, but I guess the various constants were meant to allow the delays to be adjusted as to squeeze more performance out of the write loop. These delays got removed from Marlin 2.0, even though they are still needed.

AnHardt, 2018-02-13:
Thanks for finding that.

Right. But times differ with Vdd. Above 4.5V is a bit faster than 2.7V. See ST7920 datasheet V4.0 page 43. Consequently the 3V ARMs need a relaxed timing. This code wasn't made for them anyway. Likely they have enough power to call a normal delay function.
This function is worth to be optimised as far as possible. It's called >2000 times per full screen refresh.

For AVR 8bit processors cycles have been counted and timing was calculated for 16 and 20MHz, for the well defined optimisation level in the disassembled code. For 5V only one 'nop' in delay 3 is needed. Most RRDFGSDs run pretty well completely without any nop, if cables and voltages are adequate.

AC Characteristics (TA = -30℃ ~ 85℃, VDD = 4.5V) Serial Mode Interface
Symbol Characteristics Test Condition Min. Typ. Max. Unit
Internal Clock Operation
fOSC OSC Frequency R = 33KW 470 530 590 KHz
External Clock Operation
fEX External Frequency - 470 530 590 KHz
Duty Cycle - 45 50 55 %
TR,TF Rise/Fall Time - - - 0.2 ms
TSCYC Serial clock cycle Pin E 400 - - ns
TSHW
SCLK high pulse
width Pin E 200 - - ns
TSLW SCLK low pulse width Pin E 200 - - ns
TSDS SID data setup time Pins RW 40 - - ns
TSDH SID data hold time Pins RW 40 - - ns
TCSS CS setup time Pins RS 60 - - ns
TCSH CS hold time Pins RS 60 - - ns

AC Characteristics (TA = -30℃ ~ 85℃, VDD = 2.7V) Serial Mode Interface
Symbol Characteristics Test Condition Min. Typ. Max. Unit
Internal Clock Operation
fOSC OSC Frequency R = 18KW 470 530 590 KHz
External Clock Operation
fEX External Frequency - 470 530 590 KHz
Duty Cycle - 45 50 55 %
TR,TF Rise/Fall Time - - - 0.2 ms
TSCYC Serial clock cycle Pin E 600 - - ns
TSHW
SCLK high pulse
width Pin E 300 - - ns
TSLW SCLK low pulse width Pin E 300 - - ns
TSDS SID data setup time Pins RW 40 - - ns
TSDH SID data hold time Pins RW 40 - - ns
TCSS CS setup time Pins RS 60 - - ns
TCSH CS hold time Pins RS 60 - - ns

AnHardt, 2018-02-13:
@Bob-the-Kuhn 
I explicitly warned to modify that. #8262
This loop was unroled by purpose (https://github.com/MarlinFirmware/Marlin/pull/8262/files#diff-a3acb5fe7daf75c60548289a5afd49b7L106).
Please restore the AVR 8 bit code.

AnHardt, 2018-02-13:
Where is the optimisation phragma gone?

Bob-the-Kuhn, 2018-02-14:
@AnHardt - let me see what is going on.  It's been a while & I just don't remember the details.

I'll move the AVR 8 bit code over & do some testing.

leytanelectronics2, 2019-01-17:
> #define ST7920_DELAY_1 DELAY_0_NOP

ok where are these located?