MarlinFirmware/Marlin #5704: RC8 / Bugfix REPRAP_DISCOUNT_SMART_CONTROLLER Rotary Encoder Menus inaccurate

leecuk, 2017-01-15:
Whilst attempting to resolve the ABL issues i'm having, I've noticed that quite often I'm not entering the menu item I'm trying to select and get the one above it - it might be that it is detecting movement as part of the button push, but the behaviour isn't as accurate as it was in 1.0.x.

I'm using REPRAP_DISCOUNT_SMART_CONTROLLER.

Anyone notice this?

Tannoo, 2017-01-16:
Have you played with these settings?

`#define ENCODER_PULSES_PER_STEP`
`#define ENCODER_STEPS_PER_MENU_ITEM`

leecuk, 2017-01-16:
Thanks, the default settings have been changed to 
```
#define ENCODER_PULSES_PER_STEP 1
#define ENCODER_STEPS_PER_MENU_ITEM 5
```

With no override for the REPRAP_DISCOUNT_SMART_CONTROLLER

Changing to your/original settings are much better and better match the detents on the encoder.

Should an override be added?

Tannoo, 2017-01-17:
The settings I posted were not default, but what works for my Reprap Full Graphic LCD.

leecuk, 2017-01-17:
I believe, although I could be wrong, the settings you posted were the default back in 1.0.x. I had a quick look and that's what they seem to be. They work much better than the new ones, at least for me. I was just suggesting that there are already redefines of some of the encoder steps for certain interface boards that override the default settings. Possibly a new one could be added for this very popular board?

closed 2017-01-17 by leecuk

tetraDB, 2017-01-21:
I experience the same issue with stock settings from the same REPRAP_DISCOUNT_SMART_CONTROLLER. I messed with the ENCODER_PULSES_PER_STEP and it's behaving better.
Seem the selected menu item index (if that how it works) is becoming an invalid value, -1 or something. I notice I can scroll all the way to the top, then I have to scroll down twice to get the highlight to move. Sometimes when I select a menu item, it actually clicks the one above. It's difficult to reproduce and this is the best way I can describe it sorry!
Where I notice it a lot is when I go to preheat the bed instead and instead it clicks the one above so both bed and hotend heatup.

thinkyhead, 2017-02-12:
So it looks like we should add an override for the `REPRAP_DISCOUNT_SMART_CONTROLLER` as with some of the other displays.