MarlinFirmware/Marlin #5707: millis_t issues

PsyBorg1968, 2017-01-15:
I'm not sure if these beauties are something I can comment out or if I need to figure out how to give defined data.

Arduino: 1.8.0 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\Marlin.h:60:0,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\thermistortables.h:26,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\Configuration.h:1417,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\MarlinConfig.h:30,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\Marlin.h:36,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\M100_Free_Mem_Chk.cpp:43:

stopwatch.h:45: error: 'millis_t' does not name a type

     millis_t accumulator;

     ^

stopwatch.h:46: error: 'millis_t' does not name a type

     millis_t startTimestamp;

     ^

stopwatch.h:47: error: 'millis_t' does not name a type

     millis_t stopTimestamp;

     ^

stopwatch.h:104: error: 'millis_t' does not name a type

     millis_t duration();

     ^

In file included 
                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\thermistortables.h:26:0,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\Configuration.h:1417,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\MarlinConfig.h:30,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\Marlin.h:36,

                 from C:\...\AppData\Local\Temp\arduino_build_896576\sketch\M100_Free_Mem_Chk.cpp:43:

Marlin.h:248: error: 'millis_t' does not name a type

 extern millis_t previous_cmd_ms;

        ^

C:\...\AppData\Local\Temp\arduino_build_896576\sketch\Marlin.h: In function 'void refresh_cmd_timeout()':

Marlin.h:249: error: 'previous_cmd_ms' was not declared in this scope

 inline void refresh_cmd_timeout() { previous_cmd_ms = millis(); }

                                     ^

Marlin.h:249: error: 'millis' was not declared in this scope

 inline void refresh_cmd_timeout() { previous_cmd_ms = millis(); }

                                                              ^

C:\...\AppData\Local\Temp\arduino_build_896576\sketch\Marlin.h: At global scope:

Marlin.h:286: error: variable or field 'update_software_endstops' declared void

 void update_software_endstops(AxisEnum axis);

                               ^

Marlin.h:286: error: 'AxisEnum' was not declared in this scope

Using library LiquidCrystal at version 1.0.5 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
'millis_t' does not name a type

Tannoo, 2017-01-15:
hmm... I think you have a missing `}` somewhere.

PsyBorg1968, 2017-01-15:
This millis keeps coming up as an error, but even it has its { }

extern millis_t previous_cmd_ms;                           <<<===== Error on this line
inline void refresh_cmd_timeout() { previous_cmd_ms = millis(); }

Tannoo, 2017-01-16:
You can compare your files with a fresh copy of Marlin by using something like WinMerge.

Roxy-3D, 2017-01-16:
> You can compare your files with a fresh copy of Marlin by using something like WinMerge.

Agreed.   Rather than lose all of your changes, it would make more sense to compare your current files against what you started with.   I've never used WinMerge but I probably should give it a try.  Right now I'm using ExamDiff Pro.    But it really doesn't matter...  Any good visual diff program will let you see the changes and you will find the missing } or ) or ; very easily.

Tannoo, 2017-01-16:
WinMerge is free.

PsyBorg1968, 2017-01-16:
Thank you all so much.  Big help.  Gonna start the WinMerge run right now.

Tannoo, 2017-01-16:
Winmerge can also show what files are different when comparing directories.

Tannoo, 2017-01-17:
If all is good now, this issue can be closed.

PsyBorg1968, 2017-01-17:
I found several lines I had commented out by accident.   The missing } must have been amongst it.  Thank you so much for the guidance and WinMerge.

closed 2017-01-17 by PsyBorg1968