MarlinFirmware/Marlin #5714: Separate config file for hardware layer

Angel996, 2017-01-17:
Hello.

I have recently upgraded from an earlier Marlin to the latest RC. I noticed some parameters have different names, user has to go thru the entire config file and use an old one as a sample, because hardware parameters are scattered throughout the config file. I have two different 3D printers, so I had to do it twice. :)

Maybe it would be a good idea to have a separate config file for hardware parameters ONLY? And keep it unchanged throughout the development? It would need to have the following, for example:

- Control board type MOTHERBOARD, CUSTOM_MACHINE_NAME, etc
- Stepper coefficients DEFAULT_AXIS_STEPS_PER_UNIT, DEFAULT_MAX_FEEDRATE
- Stepper and endstop polarity X_MIN_ENDSTOP_INVERTING, INVERT_X_DIR, etc
- Number of extruders, hotbed present
- Hotbed size, homing points, bed leveling end points
- Heaters parameters Kp, Ki, Kd
- LCD/Keyboard type

Whenever one uprades ANY piece of software, they don't make them go through the entire settings file every time. :)

And thanks for your great work, guys!

los7702, 2017-01-18:
My suggestion is to create an empty preset_configuration.h and insert a row

#include <preset_configuration.h>
Early in configuration.h

And change all defines in configuration.h to 
#ifndef xyz
#define xyz

los7702, 2017-01-18:
This would make it very easy to provide simple configuration for developer/manufacturers using marlin.

Tannoo, 2017-01-18:
> This would make it very easy to provide simple configuration for developer/manufacturers using marlin.

Most developers set the configurations files once for the machine they are using.
Most manufacturers set the configurations files once for the machine they are making and "cookie-cut" from there.

There are already "sections" in the configuration files, eg:
`//===========================================================================`
`//============================= Getting Started =============================`
`//===========================================================================`

`//===========================================================================`
`//============================= Mechanical Settings =========================`
`//===========================================================================`

los7702, 2017-01-18:
The problem is that configuration.h is changed all the time during development. I think that it chould be configured for a defined setup with all defaultvalues for that setup. Snd with clear descriptions of all features that can be configured. But all those settings should be overridden by another file that not need  any settings at all to work with the default. 
In that way a specific setup just need to set the settings in that file without fiddling with configuration.h. Therefore also version indipendant. 
Of course you also can choose to change direct in configuration.h whit the same resault.

los7702, 2017-01-18:
It is I rather big deal to make it fully work.
For the defines that are constants itn no problem. But it is more tricky the make the defines undefined in a overridefil that is read before configuration.h 

I think it of a 2 step process. First make it happen for the "constant" defines. And later take defines and make them bool to get them in the same line.

Tannoo, 2017-01-18:
> And change all defines in configuration.h to
#ifndef xyz
#define xyz

But, doing this would cause an error of major porportions. As EVERYTHING cannot be defined.

Tannoo, 2017-01-18:
If this were done, it would have to be a very simple and very basic setup to get printing.
- board type
- endstop info (count, inversion, etc.)
- stepper info
- max movement points.
- LCD type, if any.

Not to include bed leveling, probe info, PID info, etc.

Tannoo, 2017-01-18:
If you start adding all of those....you are just creating another file that we already have.

All of the basic config and a little more is in `configuration.h` file. More advanced setup is in `configuration_adv.h`.

Angel996, 2017-01-18:
Yes, all the basic config is in configuration.h. But these files for some reason are different throughout Marlin development. You cannot use an older config file with a newer version of Marlin. And it makes perfect sense from a programmer's point of view: project is evolving, configuration files are evolving along with it.

But think of it this way: I have the same machine whose physical configuration does not change. What sense does it make to have incompatible configuration files for the SAME HARDWARE device?

My suggestion is to move all hardware settings to a separate file and keep it (as long as it's possible) unchanged _unless_ the hardware is changed / upgraded.

Tannoo, 2017-01-18:
There are so many options...even for the basics. You would end up with almost a duplicate of the current `configuration.h`.

Angel996, 2017-01-18:
I am not talking about options. I am talking about describing hardware. Hardware does not change when Marlin changes. This is just a way of upgrading Marlin easier. All the 'options' should reference the hardware config file. There aren't too many hardware options out there. :)

los7702, 2017-01-18:
I still think you should be able to have your owh preffered option setup that is mot fiddled with when upgrading unles it is a major change in the marlin way of doing stuff... i will fork and star doing so you can see how I think...

And yes not all should be defined but the ones that are defined in configuration.h and configuration _adv.h could be overridden by another .h and the optional defines can just as well being defined elsewhere but described in configuration.h

Tannoo, 2017-01-18:
I know what your are saying. I have upgraded several times since I've had my printer. I did find it somewhat a pain to upgrade. WinMerge saved me LOTS of time doing that.

As far as developing, I have found that the configuration files are the most rarely touched. `marlin_main.cpp` and `ultralcd.cpp` have been the most altered files.

I can see a basic settings file....but where do you stop on the "basic" settings? Because there are other required options to a lot of the "basic" settings.

I've heard of a program that would walk you through setting up marlin with almost no hassle.

emartinez167, 2017-01-18:
Johnny Linden's tool: https://github.com/cabbagecreek/Marlin3DprinterTool/releases

Regards,

Ernesto

Angel996, 2017-01-19:
Tannoo, 

Well, I think I've already made my point. With basic settings you stop where it only reflects the physical features of your printer. I would call it hardware settings however, not basic settings.

Tannoo, 2017-01-19:
Well, in the case of Marlin, PID is not hardware.

Maybe this video, pointed out by @thinkyhead in #4668, will help with setting things up.

los7702, 2017-01-21:
The pid is hardware dependant settings.  All calibrating stuff is... features may not be, but it could. Bl touch for instans is a hardware feature .  The z probe offset is a callibrating setting of that feature.

Tannoo, 2017-01-21:
All PIDs are hardware dependant, but not all hardware is PID dependant.

Tannoo, 2017-01-21:
A PID loop is a better way to run a variable process. But, it's not the only way to run it.

In Marlin, the PIDs are optional.

billyd60, 2017-01-22:
It is eye bleeding torture setting up the config.h and adv.h files every time I upgrade Marlin on my machine.

Be nice if someone wrote a program that asked the user a series of questions about their printer hardware and desired features from Marlin which would automatically edit and update the necessary files accordingly prior to upload.

Is it possible to write an intelligent program that reads the two config.h files and decides on the fly whether or not what it read is important to satisfy the input requests from the user. I am thinking if you required all new code to tag new sections with predefined key words it would simplify this somewhat.

I realize this is probably a pipe dream just thinking out loud.

Angel996, 2017-01-22:
Yep, well, it's called Setup Wizard. :) I was going to offer that too, but I thought it would be a bit lame for us 3D printing enthusiasts. :) In fact, my suggestion has the same effect: there would have been an unchanged config file, tied to **your** printer, that can be used across all future Marlin versions. The Marlin release itself would have its own config file with version-specific settings.

billyd60, 2017-01-22:
lol Yes of course. But my idea is a little different because I don't know that it would be even possible to do a traditional setup wizard without a massive reorganization. I am thinking of a smart wizard that reads the setup files and makes changes (as you would doing it manually) on the fly as it reads the file, in response to the answers you give to hardware and firmware features. This way all that would be needed is perhaps the addition of certain commented key words in each section to help the creator of the smart wizard to easily find pertinent areas to enable, disable or alter. You wouldn't really need to change the traditional (current) setup and format of the config and _adv.h files. Still probably a quite difficult program to write so it's probably a bit of a pipe dream. It's easy to make suggestions when you aren't the one doing the work lol!

Tannoo, 2017-01-23:
Go through the files manually once. You shouldn't have to touch them again if everything is set correctly.

Plus, this gives you the chance to know where everything is should you change something about your printer.

thinkyhead, 2017-02-12:
Perhaps there will be some configuration re-organization for 1.2. But not for this release. We have discussed these kinds of ideas at length, but we always come back to preferring fewer config files for various reasons.

closed 2017-10-31 by thinkyhead