MarlinFirmware/Marlin #5717: Speed change while printing slow

KaiJan57, 2017-01-18:
Looks like the speed is changed by changing new commands which are to be sent to the buffer. Why not change the speed by the motor isr instead?

Bob-the-Kuhn, 2017-01-18:
The stepper ISR only runs when there is a movement command.

All movement commands are sent to the planner.  The planner creates a structure with all the needed parameters and adds the structure to a queue.  It then kicks off the stepper ISR which grabs structures from the queue and uses info from the structure to do the actual movement. 

This method is used to make designing and maintaining the software more manageable.  Isolating functions in separate modules makes it easier to make changes that don't have unintended consequences.

 Hope this answers your question.

psavva, 2017-01-20:
@KaiJan57 can you close this issue based on the explanation from @Bob-the-Kuhn

closed 2017-01-21 by KaiJan57