MarlinFirmware/Marlin #5719: Using an external accurate voltage reference for the analog-to-digital conversions

giuliotamberi, 2017-01-18:
I'd like to use a more accurate voltage reference for the analog-to-digital conversions by providing an external voltage on the pin AREF.

This should be accomplished by:

1) introducing in the Arduino code the proper AREF setting:

....
analogReference(EXTERNAL);
...

2) replacing (everywhere in the code) the default voltage reference value of 5 Volt with the one adopted as extrenal reference. For instance, 4.096 Volt

Unfortunately, I don't see a simple way to implement this in the code. I guess there should be an option to set this, for instance in the "configuration.h" file.

Thanks!
Giulio

Tannoo, 2017-01-18:
Better in the `configuration_adv.h` file.

Bob-the-Kuhn, 2017-01-18:
You'll also need to recalculate your thermistor & thermocouple tables.

The filament width sensor readings may also be affected.

Curiosity - why do you want a better Vref?

Tannoo, 2017-01-18:
> You'll also need to recalculate your thermistor & thermocouple tables.
The filament width sensor readings may also be affected.

...and then there's that.

manianac, 2017-01-19:
I don't think someone like this would be merged, but if you truly wanted to the 168/328 atmega's had a way of reading the internal 1.1v reference, and you could calculate your vcc from there.

Tannoo, 2017-01-19:
From Arduino.cc:

analogReference()
**Description**
Configures the reference voltage used for analog input (i.e. the value used as the top of the input range). 

The options are: 
- DEFAULT: the default analog reference of 5 volts (on 5V Arduino boards) or 3.3 volts (on 3.3V Arduino boards) 
- INTERNAL: an built-in reference, equal to 1.1 volts on the ATmega168 or ATmega328 and 2.56 volts on the ATmega8 (not available on the Arduino Mega) 
- INTERNAL1V1: a built-in 1.1V reference (Arduino Mega only) 
- INTERNAL2V56: a built-in 2.56V reference (Arduino Mega only) 
- EXTERNAL: the voltage applied to the AREF pin (0 to 5V only) is used as the reference.

manianac, 2017-01-19:
But that doesn't help his issue.  He's wanting to measure VCC in arduino, which if you try to use any of those options, doesn't tell you the real reading.  Whether VCC is 5v or 4.80v, it always returns 1023.

What I was referencing earlier (had to go google) was the bandgap reference.
http://jeelabs.org/2012/05/04/measuring-vcc-via-the-bandgap/ By measuring the 1.1v reference (not as the analog reference, but comparing to vcc), you can calculate the actual value of VCC versus 5v.  So 4.89v should come out to ~1001.  This allows you to change any ADC measurements accordingly.

Bob-the-Kuhn, 2017-01-19:
@manianac - I like that method of measuring Vcc. 

The only drawback I can see is the accuracy of the 1.1V reference.  Atmel's min/max "guidelines" would indicate that either it's a terrible reference (+/- 10%) or Atmel just doesn't want to spec it (which could result in chips being thrown away due to being out of spec).

BTW - how do you know that @giuliotamberi is wanting to measure Vcc?

manianac, 2017-01-20:
> BTW - how do you know that @giuliotamberi is wanting to measure Vcc?

I may have misunderstood, but I thought thats what he meant when he wanted "more accurate" voltage measurements for adc conversions.  Hooking VCC to AREF has the effect I described above, so he really should just want to measure VCC.

Tannoo, 2017-01-20:
The AREF pin is not for measuring the ADC voltage, it is for applying voltage for the chip to use as the analog reference.

Tannoo, 2017-01-20:
Apply .5vdc to the AREF pin and set analogreference(EXTERNAL). Now the analog I/O becomes much more sensitive to small voltages.
0 - AREF will be 0 - 1023 from the  ADC.

bobc, 2017-01-20:
Well, who said anything about reading small voltages, or VCC? We just seem
to be inventing requirements here.

We should first clarify if this is a support question or a feature request.
If it is a feature request, the OP must state exactly what he is trying to
do. Presumably, he has non-standard hardware. All boards I've seen use the
5V rail as an analog reference. Does he have a problem with inaccurate
readings, or is he scratching an itch to get the best possible accuracy?

To get the best accuracy from a thermistor, you probably also need to
account for self-heating, which I've never seen anyone do. If you are not
bothering with that, there probably is no point trying to fine tune AREF.

Reading VCC is pretty simple, you use a voltage divider to get it into the
range of the ADC, feed it into a spare ADC port, then adjust the reading
according to the divider.

Tannoo, 2017-01-20:
I agree. @giuliotamberi

Blue-Marlin, 2017-02-01:
As far as i see the absolute voltage is irrelevant. As long the voltage over the voltage divider is the same as for the reference (In case of RAMÜPS both the same VCC ~ 5V).
What we are measuring is only the relation of resistances of the 4K7 resistor to the thermistor.
A better reference voltage will not have a positive influence here.

A better voltage reference will help only when we are measuring absolute voltages from sensors with theyr own reference.

Bob-the-Kuhn, 2017-02-01:
If the DAC within the 2560 is implemented via laser trimmed resistors then I agree with @Blue-Marlin  - Vcc cancels itself out.  If it's implemented with laser trimmed transistors then Vcc may or may not cancel itself out.

@giuliotamberi - I see several options:
- Why bother?  If all you're doing is playing with filament extrude temperatures then what we have now is acceptable.  You need to adjust/tune the temperature for your machine and for each batch of filament so all you really need is something accurate enough to get you in the desired area.  It's more important that the measurement is repeatable.
- Since you're already implementing a custom circuit you could consider using your own precision 4.7K resistors.  You'd have to provide a buffer to power the precision 4.7K resistors from the 4.096 source.  You'd also be using analog inputs on the RAMPS that don't have the built in 1% 4.7K parts.  The best part is that you could use the thermistor tables as is.  That still leaves the thermistor errors.
- Buy better thermistors.  My understanding is the thermistor error will change over the curve so that a 1% unit will most likely read 1% low in one area and 1% high in another area.

giuliotamberi, 2017-02-02:
@Blue-Marlin the absolute voltage of the reference voltage of the ADC is relevant because it has a dicet impact on the voltage from the voltage divider made by the series of the thermisor and the 4.7kOhm pull-up resistor.

@Bob-the-Kuhn unfortunately there is no implementation via laser trimmed resistors.

Blue-Marlin, 2017-02-03:
@giuliotamberi 
> the absolute voltage of the reference voltage of the ADC is relevant because it has a dicet impact on the voltage from the voltage divider made by the series of the thermisor and the 4.7kOhm pull-up resistor.

In general you are right - but not for the special case of RepRap hardware and Marlin.

Example RAMPS on Arduino MEGA.
RAMPS: The 'voltage divider' is in between of GND and VCC
[image]

Arguino Mega: AGND = GND and AVCC = VCC
[image]

Atmel data sheet: REFS0 selects AVCC as the reference.
[image]

Marlin code: Select AVCC as the reference.
```
  #define SET_ADMUX_ADCSRA(pin) ADMUX = _BV(REFS0) | (pin & 0x07); SBI(ADCSRA, ADSC)
```

Reference voltage for the ADC = voltage over the 'voltage divider'
(as long as VCC is not changing too fast (noise - capacitys - inductivities))

So the absolute hight of VCC is not interesting at all. With the 3.3V Arduino DUE you will get the same results.

giuliotamberi, 2017-02-03:
@Blue-Marlin 

Unfortunately that VCC is always noisy. Just monitor this voltage with an oscilloscope and you will notice rapid voltage fluctuations of about +/- 100mV. This ha an impact on the ADC converted value and tehrefore on the closed-loop control of the extruder temperature.

Adding an external 100nF capacitor on the pin AREF (as suggested by Atmel) would improve the stability of the ADC reference voltage but anyway the average value would be a certain value different for the nominal 5V used in the code. No matter of the voltage fluctuations.
In fact, for instance,  if the average voltage of the ADC Vref = 4.9V but you assume in the code that it is 5.0V by default, then you will have a systematic error in the conversion which will decrease the accuracy of the temperature estimation.

Also, you're wrong when stating that the absolute hight of VCC is not interesting at all. Just look at the voltage divider in the Thermistor section of the RAMPS schematic. Let’s focus for clarification just on the Thermistor 0, the  voltage which goes to the input of the ADC is:

THERM0 = VCC * R_THERMISTOR0 / ( R7 + R_THERMISTOR0)

At room temperature:

THERM0 = VCC * 100k / ( 4.7k + 100k) = VCC * 0.955

So the VCC voltage has a direct impact on the voltage THERM0 which goes to the ADC.

As an example, in the nominal best case when VCC = 5V then THERM0 = 4.775V however when VCC = 4.9V then the THERM0 = 4.680V

Giulio

Blue-Marlin, 2017-02-03:
> Adding an external 100nF capacitor on the pin AREF (as suggested by Atmel) would improve the stability of the ADC reference voltage ...

Arduino Mega C6!

You did not get it.
We are not interested in the absolute voltage, just and only in the relation of R7 to  R_THERMISTOR0. 0.955 is all we want.

giuliotamberi, 2017-02-03:
@Blue-Marlin 

Unfortunately on Arduino Mega there is no 100nF capacitor on the AREF pin (and placed close to the AREF pin in the layout).

About the rest you're right and I agree with you.

Bob-the-Kuhn, 2017-02-03:
I think we may be talking apples & oranges here.

@giuliotamberi - question - with this scenario:
with Vref set to Vcc
if Vcc drops by 5%, then the Vin (from the thermistor & 4.7K voltage divider) also drops by 5%
he Vref also is dropping by 5%,

Doesn't the counts seen by the software remain the same?

Bob-the-Kuhn, 2017-02-03:
On the noise issue - the thermistor readings all are oversampled which, hopefully, cancels out the noise.

Blue-Marlin, 2017-02-03:
@giuliotamberi 
I admit C3 for AREF and C6 for AVCC are not idealy placed, but they are far from missing. Near and far are extremely relative.
[image]

Blue-Marlin, 2017-02-03:
@giuliotamberi
Maybe you don't know about '...Marlin\buildroot\share\scripts\createTemperatureLookupMarlin.py'. It's a script to make your own thermistor tables. With the new created tables you can replace any of the existent tables or you can simply add them. 

```
#!/usr/bin/python
"""Thermistor Value Lookup Table Generator

Generates lookup to temperature values for use in a microcontroller in C format based on:
http://en.wikipedia.org/wiki/Steinhart-Hart_equation

The main use is for Arduino programs that read data from the circuit board described here:
http://reprap.org/wiki/Temperature_Sensor_v2.0

Usage: python createTemperatureLookup.py [options]

Options:
  -h, --help        show this help
  --rp=...          pull-up resistor
  --t1=ttt:rrr      low temperature temperature:resistance point (around 25 degC)
  --t2=ttt:rrr      middle temperature temperature:resistance point (around 150 degC)
  --t3=ttt:rrr      high temperature temperature:resistance point (around 250 degC)
  --num-temps=...   the number of temperature points to calculate (default: 36)
"""
...
```

Bob-the-Kuhn, 2017-02-03:
I believe the base issue is that @giuliotamberi thinks that, when Aref is set to Vcc, variations in Vcc will result in changes to the ADC counts seen by the software.

I did an experiment on my 2560 to see if he is correct:
- 10K resistor to an analog input that has a 4.7K pull up
- powered the board from an external 24V supply
- powered the board off the USB bus and attached it to two different PCs
- Measured Vcc using a DVM -  read 4.06V, 4.68V & 4.78V
- Used the M43 I1 command to see the counts - in all cases it was 704

My conclusion is that, when Aref is set to Vcc, variations in Vcc will **NOT** result in changes to the ADC counts seen by the software.

============================================================

To me that means an external reference will **NOT**, in itself, improve the accuracy of the thermocouple readings.

That means the only ways to improve the accuracy, and still use the on board ADC, is to use tighter tolerance 4.7K pull ups and to buy better thermocouples.

============================================================

My suggestions are:
1. Don't use the external reference
2. Use a Pt1000 (or Pt100) RTD - they are more accurate in the 200C area than an EPCOS 100k style thermistor
3. Use an external 0.1% resistor.  This would require using some of the six "non 4.7K" analog inputs on the AUX-1 SERIAL and AUX-2 connectors.

=============================================================

Using the external resistors shouldn't be a big problem.  My suggestion is to build a custom cable that incorporates the external resistors:
- buy a connector housing that matches the header you're using
- crimp all your resistors into the contact for Vcc
- for each sensor, crimp the other end of the resistor and one of the wires from the sensor into the contact for your analog input
- crimp all the other sensor wires into the contact for ground
- use heat shrink tubing to isolate/protect as desired.

==============================================================

I recommend using the Pt1000 rather than the Pt100 because it's less sensitive to wiring/contact resistance.  There's already temperature tables in Marlin if you use 1K or 4.7K pullup resistors.

giuliotamberi, 2017-02-04:
@Bob-the-Kuhn 

Thank you for your post: it clarified all my doubts and provided great suggestions to improve my printer!

Giulio

closed 2017-02-04 by giuliotamberi