MarlinFirmware/Marlin #5720: RC7 Full Graphic Smart Controller problem

DaniDiaz71, 2017-01-18:
I've installed a new Full Graphic Smart Controller on my Prusa i3 (Hephestos by BQ), but after several minutes printing display shows like this:

[image]

At the beginning display shows OK, like this:

[image]

I use IDE 1.6.13 and U8glib 1.19.1, it could be solved at RC8?

Thanks and BR.

  Daniel.

Tannoo, 2017-01-18:
Probably not, but it never hurts to run the lastest firmware.

I belive it is a bad display.

DaniDiaz71, 2017-01-19:
Yes, it seems to be the same problem, i'm to buy a new one.

Thank you very much ;-)

willthiswork89, 2017-01-19:
my new one started doing it again too. I'm started to unconvince my self that the LCD is the issue.... i

Tannoo, 2017-01-19:
Re-download and install the LCD drivers.

psavva, 2017-01-20:
@DaniDiaz71 based on the conclusions, please can you close this incident

DaniDiaz71, 2017-01-20:
Done! Thank you :)

closed 2017-01-20 by DaniDiaz71

DaniDiaz71, 2017-01-20:
Finally my problem was a ribbon cable bad contact, I unplugged both cables and plugged it again and now it's working fine.

Roxy-3D, 2017-03-30:
> Finally my problem was a ribbon cable bad contact, I unplugged both cables and plugged it again and now it's working fine.

I would replace the ribbon cable.  Just because it is making a good connection right now...   I wouldn't trust it...