MarlinFirmware/Marlin #5721: [WIP] Multi MAX6675 and MAX31855 support

ss89, 2017-01-18:
Hello,

i am currently hacking marlins RC8 branch (in my own repo) to support multiple MAX6675 or MAX31855 thermocouple amplifiers.
I am in the state to wait for hardware, but software wise i can compile marlin with support for up to 4 extruder thermocouples (MAX6675).
Whats planned: Support for heated bed with MAX6675 or MAX31855 thermocouple.
If someone is currently doing the same work or wants to support me, please let me know.

kinds regards,
ss89

Grogyan, 2017-01-18:
Might I suggest looking into the MAX31850 which uses the 1-wire bus protocol.
Although you are looking at a lot more characters to stream over the 1-wire, it does allow you to add a lot more thermocouples on the same data line

ss89, 2017-01-18:
@Grogyan Thanks for mentioning that.
My intention is to expand the current possibilities of marlin by expanding the implementation of the MAX6675 and MAX31855.

ss89, 2017-01-21:
Just in case someone wants to try it out:
https://github.com/ss89/Marlin/tree/feature/multipleMAX6675

thinkyhead, 2017-02-15:
The source formatting is horrid 😈 but the implementation looks solid!

ss89, 2017-02-15:
I'm really sorry about that one, i hacked it in textmate and i am not used to contributing to oss projects.
A friend of mine tried it out but it didn't work (nor did the Original implemenation), while repetiers implementation looked similar and worked.
Possibly config fault, but not sure yet.

thinkyhead, 2017-02-15:
No biggie. I've submitted a PR to your branch that fixes all that and a few bugs too!

ss89, 2017-02-15:
Thanks a lot! We will test it as soon aswe have new hardware. He by accident moved the motor too fast and you know what happens then, right? ;)

ss89, 2017-02-23:
The friend of mine who had multiple max6675 which blew up now uses thermistors, so i will order 4 max6675 amplifiers.

ss89, 2017-03-19:
Just to keep you guys updated: i am still waiting on the hardware to arrive :-/

ss89, 2017-04-02:
Finally, 2 of 4 amplifier board have arrived and i have just tested them.
It seems to work nicely with 2 max6675, hopefully in a few days i will be able to test that out with 4 max6675.
Since that seems to work for now, i could look into heated bed max6675 support.
Is there any interest?

Mad-Onion, 2017-06-17:
I'm interested in using a MAX6675 for the heatbed :-)
Do I have this right, that I can use a MAX6675 only for sensor 0 (Marlin 1.1.3)?

ss89, 2017-06-18:
I believe that i currently am the only one 'working' on the max6675 implementation, but i could be wrong.
Haven't had time lately.
@thinkyhead ?

thinkyhead, 2018-02-06:
Yes?

closed 2018-02-06 by thinkyhead

thomaskilian, 2018-06-28:
Funny to see that this is still not realized. In my last Marlin code I hacked that (rather easily) but after modifying tons of other code. I'm now moving to another board but would like to still use the MAX shields - and now staying with standard Marlin. Anyone interested in that? In that case I'd try to do the hack so it could possibly be included for a pull request.

Grogyan, 2018-06-28:
You cannot use a thermocouple shield as it will likely use the CS pin that
is also used for the SD card.
You must use a different I/O pin for the Max31855

Yes the code does work, as I have have it working with no issues along side
the SD card.

thomaskilian, 2018-06-28:
Well, I have to take back my offer anyhow. Now that I looked into the code, this part is anything but modular. So it will be again a just-for-me-hack :-/ I guess that most people have switched to using Octoprint rather than having an SD card sitting on the Arduino. So all that LCD and SD card stuff is for the bin anyway (for me at least). The last (and first) time I adapted Marlin it started by throwing away all that stuff I did not need. That ended up being a cul-de-sac but I understood the code. Would be nice if Marlin were more of a modular thing rather than this monolith.

thinkyhead, 2018-06-30:
> Would be nice if Marlin were more of a modular thing rather than this monolith.

The new file layout in `bugfix-2.0.x` is a step in that direction.