MarlinFirmware/Marlin #5722: [Discussion] Discrepancy between probe_pt() verbose log and G29 bilinear correction matrix log

FHeilmann, 2017-01-18:
When doing a `G29 V4` with bilinear bed leveling, the debug output of `probe_pt` outputs the measured probed height, without the Z-Probe offset applied.

After the command is done, the bilinear leveling matrix is applied, where each point has been corrected using the z-probe offset.

I feel like both outputs should output the same values. It caused me a lot of headache with my printer (0,0 is in the back right and thus the output of the bilinear grid is mirrored) because I could not use the verbose `G29` easily to figure out which output by `probe_pt()` ends up where in the final output.

I wanted to put this up for discussion before submitting a pull request.

manianac, 2017-01-19:
What are you changing in your PR?

FHeilmann, 2017-01-19:
I'd either adjust probe_pt to print the adjusted z-height or G29 to print an unadjusted report. Whichever is more desirable to the Marlin devs

FHeilmann, 2017-01-19:
See #5724

closed 2017-02-15 by thinkyhead