MarlinFirmware/Marlin #5723: MAX31865

Project-J2, 2017-01-18:
Hello, for this moment marlin SPI to read temperature is only compatible with MAX31855 (board for thermocouple, 14bit), but how can we make marlin compatible with MAX31865 (board for PT100, 15bit) ?

Somes ideas ?

Thanks ! :)

manianac, 2017-01-19:
I'm sure the code will be very similar to the 31855, just adapting it to the datasheet

Project-J2, 2017-01-19:
Thanks for your reply but absolutly not :/

MAX31855 need only read by marlin, but MAX31865 need read and write !

See datasheet:
[MAX31865.pdf](https://github.com/MarlinFirmware/Marlin/files/717501/MAX31865.pdf)
[MAX31855.pdf](https://github.com/MarlinFirmware/Marlin/files/717500/MAX31855.pdf)

Somebody had an idea ? :)

Thanks !

comps, 2018-02-22:
Looks like somebody got it working (though it looks like a PoC, not a polished product):
http://forums.reprap.org/read.php?415,809669

To me, this issue seems fairly important and relevant for Marlin v2.0 as it's the only feasible way of getting accurate temperatures on most 32bit boards as their 3.3V logic doesn't work well with the 5V PT100 amplifier boards (without wiring another 5V supply and risking CPU damage).

I'll see if I can get my hands on one of the MAX31865 boards to help with testing.

gabtremblay, 2018-06-04:
Just checking if someone is working / planning to work on that and if I can offer my help.

comps, 2018-07-02:
For somebody who knows the Marlin codebase, the actual implementation will likely be fairly simple thanks to the Adafruit library for MAX31865 (if it works on non-AVR), see [this handy example](https://github.com/adafruit/Adafruit_MAX31865/blob/master/examples/max31865/max31865.ino). The [library itself](https://github.com/adafruit/Adafruit_MAX31865/blob/master/Adafruit_MAX31865.cpp) is also surprisingly simple.

The real challenge might be sharing HW SPI with other things like the SD card, which IIRC switches SPI modes, options like `SPI_SPEED` which will likely come into effect, and other hacks for various devices.

thinkyhead, 2018-11-05:
> Do you plan to pull request it into marlin or you're waiting for some testers before?

Waiting for the miracle. For the miracle to come.

anurag77719, 2018-11-28:
I have used the help available on reprap forum and have used MAX31865 with PT100 ( in 3 WIRE configuration) to read temperature. The 'temperature.cpp' and 'configuration.h' file is attached here. Marlin firmware ( version 1.1.x) was used with RAMPS1.4
 
[Files_for_Marlin_max31865.zip](https://github.com/MarlinFirmware/Marlin/files/2624032/Files_for_Marlin_max31865.zip)

thinkyhead, 2018-11-28:
We’re not updating 1.1.x any more. It would be better if you could fork Marlin, modify a copy of the `bugfix-2.0.x` branch, and submit a pull request with the necessary changes. Test `bugfix-2.0.x` first to see if the thermocouple is already working or not.

boelle, 2019-02-18:
@anurag77719 

We’re not updating 1.1.x any more. It would be better if you could fork Marlin, modify a copy of the `bugfix-2.0.x` branch, and submit a pull request with the necessary changes. Test `bugfix-2.0.x` first to see if the thermocouple is already working or not.

boelle, 2019-07-20:
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

boelle, 2019-08-28:
@anurag77719 could you submit a pr to bugfix 2.0.x ?

smellykid101, 2019-10-28:
Last week I had a chance to try implementing the adafruit library into bugfix 2.0.x, which read temperatures just fine. However, it would randomly error out with min/max temp readings. I haven't pin pointed the exact source of the problem, but I believe it's with the way I integrated it on the SPI bus (ext connectors for sd card/ screen on my skr 1.3 board). I'm still learning GitHub to know how/where to share this information/my edits for further assistance. I have no idea how to submit a pr yet. Recommendations are welcome!

bob808, 2019-11-10:
I can't get the max31865 to play nice with the sd-card.
I get watchdog resets if they are both connected. If I disconnect the sd-card then I get the correct temperature (but no sd-card).
I have both on DUE SPI connector. sd-card has 4 as cs pin, and max31865 has 49 as cs pin.
I added the modified Adafruit library and modified configuration.h/temperature.cpp files.

bob808, 2019-11-17:
Managed to get it working on software SPI. Hardware SPI works for SD card only. max31865 and tmc2130 are on software.
Tried adjusting spi_rate to match to sd card settings but still nothing.
max31865 and tmc2130 both work on hardware spi as long as sd card is not installed on hardware SPI.

Grogyan, 2019-11-17:
Marlin only allows for the Max31855 and Max31856 to work when setting the firmware to use the the SD card on the display

bob808, 2019-11-17:
I managed to make both max31865 and tmc2130 to work on hardware SPI, with the sd-card as well on the hardware SPI.
My problems were due to the fact that I was using a cheap sd-card adapter for testing. This adapter has a HC125 buffer on it, and sadly the CS line is not tied to the OE line of the buffer for MOSI. That meant that the sd-card got hold of the MOSI line and never let it go, so the other devices could not respond on it. I had to use a logic analyzer to figure it out.
Also I've seen that the clock speeds are not the same.
Using SPI_SD_INIT_RATE 5 in SdFatConfig.h sets the init clock at 328KHz for sd-card. While tmc2130/max31865 from JackDaFlow's repo come with the 84000000/pow(2,1+4) format which comes out at around 2.625MHz clock, if using 4 as a speed index.
So both Adafruit_MAX31865.h and TMC2130Stepper.cpp need to have this speed declared to sync all of them: 84000000/pow(2,1+7). That's 328KHz, and all of them work at the same time.

smellykid101, 2019-11-18:
> Using SPI_SD_INIT_RATE 5 in SdFatConfig.h sets the init clock at 328KHz for sd-card. While tmc2130/max31865 from JackDaFlow's repo come with the 84000000/pow(2,1+4) format which comes out at around [...]

These are the same changes I had to enable to get my system stable. I've run a few temperature cycles, let the system idle for a few hours at different temperatures and it seems to work. Might try printing next week when I have time. @bob808 is you system working well now? Have you tested any prints via SD or usb?

bob808, 2019-11-18:
At the moment everything is breadboarded. But I've used a PT100 sensor and it seems to be correctly reporting the temperature, at least for the few hours I had it on.

smellykid101, 2019-11-18:
Awesome! If anyone else has suggestions of what specific situations we should test, let us know. 

What's the procedure to have this implemented in the next Marlin build?

bob808, 2019-11-18:
I made a PR. Just that the Adafruit library has to be added to the project.

bob808, 2019-11-21:
They included my changes but they modified the code to acomodate for both 0/1 temp sensors but it's kinda broken at the moment. 
I have to define both #define MAX6675_SS_PIN   49 and #define CS_PIN 49 or else I get errors, then even so, it works for sensor 0 but with lots of warnings on serial:
`Error:Temp measurement error! MAX6675
Error:Temp measurement error! MAX6675
Error:Temp measurement error! MAX6675
Error:Temp measurement error! MAX6675
Error:Temp measurement error! MAX6675
Error:Temp measurement error! MAX6675
Error:Temp measurement error! MAX6675
Error:Temp measurement error! MAX6675
Error:Temp measurement error! MAX6675`

But the temperature is correctly reported on the display.
Also I can't selectively use it for temp_sensor_1 if temp_sensor_0 is not also max31865. Since I have only one max31865 chip I can't test it for both.
Hope I find some time to see what the problem is with the error on serial console, and why I need to define both variables on the same CS pin.

bob808, 2019-11-27:
Almost done. Just need to solve sanity checks. Next PR should be done.
At the moment it's working but I get lots of:
`Error:Temp measurement error! MAX6675`
I see that this code is responsible for this message:
` if (max6675_temp & MAX6675_ERROR_MASK) {
      SERIAL_ERROR_START();
      SERIAL_ECHOPGM("Temp measurement error! ");
      #if MAX6675_ERROR_MASK == 7
        SERIAL_ECHOPGM("MAX31855 ");
        if (max6675_temp & 1)
          SERIAL_ECHOLNPGM("Open Circuit");
        else if (max6675_temp & 2)
          SERIAL_ECHOLNPGM("Short to GND");
        else if (max6675_temp & 4)
          SERIAL_ECHOLNPGM("Short to VCC");
      #else
        SERIAL_ECHOLNPGM("MAX6675");
      #endif`
I could just remove it as the temperature seems to be correctly reported, but I'd like to know if there's any other issues.
I presume there's MAX6675_ERROR_MASK that has a value. Is that normal? As long as the temperature is correctly on the screen, what would be the issue?

bob808, 2019-11-27:
Ok, with the latest changes this sensor is now usable.
use -5 in Configuration.h
adjust MAX31865_2WIRE if using 3/4 wire sensor (MAX31865_3WIRE/MAX31865_4WIRE) in temperature.cpp
#define MAX6675_SS_PIN in your pins file, for hardware SPI. 
for software SPI:
` #define MAX31865_CS_PIN
#define MAX31865_MOSI_PIN
#define MAX31865_MISO_PIN
#define MAX31865_SCK_PIN `
I only tested for heater0 as I have only one max31865 chip. I'd say this FR is solved.

Zatarines, 2019-12-03:
Hi @Bob808, 
 I will ordered this amplifier [https://www.amazon.com/gp/product/B07PDFNZYX/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1] , to install with a MKS Sgen_L v1.0 because my Ultimaker 2 clone board fried some components. Could you kindly provide me with your Marlin2 FW changes and a wiring diagram that you used in your successful install of the PT100 thermistor. 
regards,

bob808, 2019-12-03:
I have the SSOP version of the chip, I'm doing my own board. But they should be the same. Power on 3V3/GND, and connect to the DUE SPI bus.
[image]
Check online for pinout. Leave RDY disconnected.
The sensor should work with the latest Marlin 2.0 Just select it in Configuration.h, you have all info there. My pins file is relevant to my board only, it will not be of use to you.

Zatarines, 2019-12-06:
@bob808 
  In order to properly connect the amplifier to my MkS gen_lv1.0 board I need four open pins? So CLK, SDO, SDI, and CS should be connected to any pin that I am not using. 
[MKS SGEN_L V1.0_001 PIN.pdf](https://github.com/MarlinFirmware/Marlin/files/3934293/MKS.SGEN_L.V1.0_001.PIN.pdf). For example, in my board I could use TH1 pin p0.23 that is for the ext_0 , and pins for the Max end stops that I am not using. is there another way? please excuse my lack of understanding here. 
thank you,

bob808, 2019-12-06:
This is for Arduino DUE shields. I have no idea if this works on other platforms.

khomouda, 2019-12-23:
@bob808
what about :
#define TEMP_0_PIN what is the pin assigned if you are doing SPI

bob808, 2019-12-23:
If using max31865 I think you can leave it on any of the ADC pins. I think I have it
`#define TEMP_0_PIN          0  // Analog Input`
TEMP_0_PIN may be on the sensitive pins list and it requires you to define it as any of the ADC pins.

AnHardt, 2019-12-23:
Usually the Temp pins have 10µF Kondensators - slowing down the signal change. Not the best choice.

khomouda, 2019-12-23:
here is my setup
[image]

[files.zip](https://github.com/MarlinFirmware/Marlin/files/3995987/files.zip)

bob808, 2019-12-23:
> 
> 
> Usually the Temp pins have 10µF Kondensators - slowing down the signal change. Not the best choice.

I don't understand how this would be a problem. That pin would not be used, but it's necessary to be declared as it's on the sensitive pins list.  I guess ideally there should be a conditional in the sensitive pins file looking for either #define MAX6675_SS_PIN either #define TEMP_0_PIN.

khomouda, 2019-12-23:
what is define MAX6675_SS_PIN for?

bob808, 2019-12-23:
That is the hardware SPI CS pin of the max31865 chip.

khomouda, 2019-12-23:
So should be same as MAX31685_CS_PIN
In my case P0_16?

bob808, 2019-12-23:
MAX31865_CS_PIN is the CS pin for max31865 but for software SPI, on Arduino DUE, that has a different processor than the re-arm board.

khomouda, 2019-12-23:
My above configuration hangs the re-arm at boot screen

Grogyan, 2019-12-23:
> My above configuration hangs the re-arm at boot screen

I checked your Re-Arm pins file, line # 254 does not have the correct pin assignment for the MAX6675_CS

Line # 1153 NEEDS to be uncommented and assigned to LCD in configuration_adv

In your temperature.cpp you do not put pin defines in that file.
And heavens know if you have set the bit shifting correctly to read the data from the MAX31865 properly

khomouda, 2019-12-24:
I am extremely sorry, I am a physician and don't know how to set bit shifting. NB, it is still hanging in the boot screen.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/3997293/Marlin.zip)

AnHardt, 2020-07-08:
Original FR is done.

closed 2020-07-08 by ellensp