MarlinFirmware/Marlin #5725: 1.1.0 RC8 - M605 S2 - DXC_DUPLICATION_MODE - Second extruder not extruding

sdgray, 2017-01-20:
Try as I might I can't find a reason for it. RC7 dual x carriage duplication mode works. RC8 seems to but the second extruder does not move. 

I've found references to the same issue in other branches, but no description of the fix in anything else I've found. 

In any other mode both extruders work as they should. In duplication mode (M605 S2 X50 R0, then home) both T0 and T1 move correctly in X and stay at same temperature. T1 extruder motor does not rotate.

thinkyhead, 2017-02-15:
How does the current `RCBugFix` fare with `DXC_DUPLICATION_MODE`? We've addressed some issues over the last few weeks and might have fixed this. If not, we can look closer.

sdgray, 2017-02-16:
Confirmed. No change. I don't see any commits to RCBugFix like you
mentioned. Are you talking about 32 bit perhaps?

thinkyhead, 2017-02-26:
It seems to be the case, as far back as 1.0.2, that the `extruder_duplication_enabled` flag is set to `true` only when un-parking the carriage while `DXC_DUPLICATION_MODE` is active and the selected extruder is 0. `M605 S2` actually sets `extruder_duplication_enabled` to `false` as the starting state.

Is it always the case that the extruder will be parked when starting duplication mode? Apart from auto-parking the carriage in auto-park mode and on `G28`, Marlin also marks the carriage as "parked" on a `T0` command (if the active extruder isn't already 0).

sdgray, 2017-03-03:
That might explain why it was working for me when running RC7. I had the
default mode as auto park, but not when I was on RC8. I'll be testing
whenever I can get back to it.

closed 2017-10-31 by thinkyhead

emphasize, 2018-03-03:
@thinkyhead After days of narrowing down the persistent problem that is described by the TE, i give up and hope someone can help me with this.

### **_The situation:_** 
Dual X Carriage IDEX-style printer, same observations as described above. Movement - in every mode - ok (T1 start scenario in duplication mode aside). But when it comes to duplicate Extruder(E0)-motor movements - in M605 S2 mode - it fails to do that.

-------------------------------------------

Given that noone in 1 year reacted to that subject, either noone gives a damn about dual/dupli (i dont think thats the case, cause thats the way to go - even more if you want to be more productive) and the problem remains unseen, they work around with some subpar wiring experiments or somethings fishy with my setup/config. Because of this - and for transparancy reasons - i go a little further on the details (but then again, i tweaked almost every config and studied the - dupli-related - code to understand whats happening, that i am pretty confident on the config side)

### _Setup/Config:_ 
_(I'm running a pretty much stripped down version)_
latest bugfix-1.1.x (8)
DUAL EXTRUDER / EXTRUDERS = 2
DUAL_X_CARRIAGE
LIN_ADVANCE
_Maybe good to mention:_ 
DISTINCT_E_FACTORS
No LCD defined (using "independent" one; MKS TFT)

--------------

**_(potential concerns - at least in my view)_**

**_Disabled Extruder:_**   

> **my config:** Configuration.h: [`DISABLE_E false` ](https://github.com/MarlinFirmware/Marlin/blob/63b13588a8a73074c44f82f1eafda8a6b06dc99d/Marlin/Configuration.h#L758) [`DISABLE_INACTIVE_EXTRUDER true`](https://github.com/MarlinFirmware/Marlin/blob/63b13588a8a73074c44f82f1eafda8a6b06dc99d/Marlin/Configuration.h#L759)
Configuration_adv.h [`DISABLE_INACTIVE_E false`](https://github.com/MarlinFirmware/Marlin/blob/63b13588a8a73074c44f82f1eafda8a6b06dc99d/Marlin/Configuration_adv.h#L399)
`DEFAULT_STEPPER_DEACTIVE_TIME 0 // Set to 0 to deactivate.` (This is an observation i made as i wrote this. Doesn't this directly contradict the DISABLE_INACTIVE_EXTRUDER definition?)

>_Observation/Discussion:_ I've ran every possible - and workable - combination. Same outcome every time. 
But what bothers me is that there are some (possible) conflicts in the code. In [Conditionals_post.h](https://github.com/MarlinFirmware/Marlin/blob/63b13588a8a73074c44f82f1eafda8a6b06dc99d/Marlin/Conditionals_post.h#L248) it's `#define DISABLE_INACTIVE_E DISABLE_E` _if_ `DISABLE_INACTIVE_E` _not defined._ If it is defined both existing next to each other. [Planner.cpp](https://github.com/MarlinFirmware/Marlin/blob/2c58060fbf6dcb0a50d5fefea89ba2287c192589/Marlin/planner.cpp#L486) someplace working with `DISABLE_E`. That means it is possible to set false (e.g. `DISABLE_E`) and true (e.g. `DISABLE_INACTIVE_E`) for presumably the same thing at the same time. Why not condense into one, or are there distinct scenarios where you have to shut off (one) / (all) / (all at once) <- and if this is the case the comments are at best misleading
That aside [planner.c](https://github.com/MarlinFirmware/Marlin/blob/2c58060fbf6dcb0a50d5fefea89ba2287c192589/Marlin/planner.cpp#L882) should adress the problem of deactivated Extruders in every case. Given that with `DISABLE_INACTIVE_E=true` the `extruder_duplication_enabled`-Flag is set true at that point in time (in marlin_main.cpp), which brings me to

**_The `extruder_duplication_enabled`-Flag_:**

Since this is crucial as an entry point (and processing throughout the duplication print) you dont want to mix up the sequence.  _Start GCode used:_ (Ideamaker 1.3.1+; Automatic Heatup GCode insertion DISABLED!)
> T0
G21
G90
M605 S2
G28
M140 S{temperature_heatbed}
M104 T0 S{temperature_extruder1}
M109 T0 S{temperature_extruder1}
M190 S{temperature_heatbed}
                                                                    <------- here the unparking(=**extruder_duplication_enabled=true**) of T1 occurs / moving to offset 
G92 E0
M82
M107
G1 F{travel_xy_speed}
M117 Printing..

_(Off Topic: The initial purge pre-print should be hard-coded (function dual_x_carriage_unpark), because the unparking starts with reaching the temperature and i see no possible -noob-easy- solution until recoded elsewhere)_

Serial Communication: "Set extruder_duplication_enabled / Clear active_extruder_parked". No G28, M605, tool_change or -obviously- unparking (which would mean reflagging) since the entry point should remain the flag untouched until it's called by

**_The header of stepper_indirection ([stepper_indirection.h](https://github.com/MarlinFirmware/Marlin/blob/63b13588a8a73074c44f82f1eafda8a6b06dc99d/Marlin/stepper_indirection.h#L476))_**

I dont pretend i know whats exactly going on after the signals _should be_ split into two by

> #elif EXTRUDERS > 1

`extern bool extruder_duplication_enabled;` missing? // but even with that added no extruding
>  #if ENABLED(DUAL_X_CARRIAGE) || ENABLED(DUAL_NOZZLE_DUPLICATION_MODE)
#define E_STEP_WRITE(v) { if (extruder_duplication_enabled) { E0_STEP_WRITE(v); E1_STEP_WRITE(v); } else if (current_block->active_extruder == 0) { E0_STEP_WRITE(v); } else { E1_STEP_WRITE(v); } }

regarding #_APPLY_STEP / #_WRITE_STEP / #_STEP_WRITE (stepper.h/.cpp) but over time there have been numerous recoding on this. Broken there?

As I initiated the procedure with T1 active (T1 -> M605 S2 -> G28 -> ...) it was the other way around (besides obviously no movements x-wise ) T1 extruded, T0 not.  Something's fishy down there.
On the other side and noobish as I am, I question myself why the duplication of the extruders is not coded like the X/X2 in stepper.cpp (if I understand the code correctly)?

----------------------

So, this was an intense one, but i want to be detailed as possible. I'd like to understand the "inner mechanics", besides the point that this is way above my head (and has a crazy steep learning curve). 

How can i proceed or help with the testing?  Cause it seems this part of marlin need some love ;)