MarlinFirmware/Marlin #5727: Bilinear mesh activation problem on rc8bugfix

Sicon76, 2017-01-21:
Hello! I have some problems with the bilinear mesh option. I flashed marlin rc8 bugfix with the fixed probe (I have a capacitive sensor) and bilinear mesh option for 10 points, I did a G28 followed by a G29 and at the end it gave me the mesh points, I stored all with M500 and started a new print using G28 and M420 S1 as starting codes, but the mesh wasn't applied (the Z motors didn't compensate). I did a G29 S0 to see the mesh points but all I had is "G29 Auto Bed Leveling" and no answer, the motors seems energized and did a strange sound but they stand still and the menu seems locked as when the printer is executing some commands (the machine control window of simplify3d give me a "busy: processing" after that command), the only way to exit is to power off, what do you think? How could be sure the mesh is stored and applied to the print?

Thank you!

Sicon76, 2017-01-21:
I did some other tests: I did a print with a a G28 followed by a G29 as gcode start instructions but it is the same, the mesh wasn't enabled (the Z motors didn't move), the same with G28, G29 and M420 S1

radry, 2017-01-23:
You shouldn't do G28 after G29. I think G28 will reset any autoleveling. Just do G28, G29 and start the print. No need for M420.

Sicon76, 2017-01-23:
Never did a G28 after G29, I always do a G28 before (even because G29 doesn't start without homing first), but I'd like to store the G29 results to avoid to do the probing everytime, so after a G29 I do a M500 but I'd like to know how could be sure the mesh is calculated and stored and how to enable it after every power on. I thought a G29 S0 could give me some results, but it doesn't work the ABL (maybe is there another gcode?)

thinkyhead, 2017-04-02:
How's it working today with the current `RCBugFix` code?

closed 2017-10-31 by thinkyhead