MarlinFirmware/Marlin #5731: Support Re-ARM for RAMPS

DeathmannX, 2017-01-22:
Hi there,

i've got a short Question.
Will the RE-ARM for RAMPS support added in the near Future or is avaible now?
Cause i don't realy like the way to set up this thing with Smothieware.

Best regards ;)

JustAnother1, 2017-01-22:
I think @bobc is working on it,...

Tannoo, 2017-01-22:
Read and/or follow this thread: #3851

JustAnother1, 2017-01-22:
the tl;dr is this pull request: https://github.com/MarlinFirmware/Marlin/pull/5357

Roxy-3D, 2017-01-23:
> Will the RE-ARM for RAMPS support added in the near Future or is avaible now?
> Cause i don't realy like the way to set up this thing with Smothieware.

What don't you like about the set up with Smoothieware?  How do you want it setup?

DeathmannX, 2017-01-29:
Sry to let you wait that long. 
I don't like the configuration for Smoothieware and how to transfer the Firmware.
I know the Board got a extra SD Card for that but i don't know.
My Prusa i3 runs with Marlin and you guys got awesome Feature already.
I don't want 10 Printers with 10 different Firmwares i want 1 that work and with new Features upcomming.

I hope i can answer you're Question @Roxy-3D ;)

Fulg, 2017-01-30:
If you don't want to use a different firmware than Marlin, you should not have upgraded to Re-ARM. The whole point of Re-ARM is to keep your existing RAMPS board, ditch the Arduino platform and switch over to Smoothieware...

TGMods, 2017-01-30:
@Fulg My reason for changing to Re-Arm was that my Mega was broken and I was looking for a replacement. Being able to use the power of 32bit processing without changing my wiring too much seemed to be too good to be true. Whilst running Marlin would be perfect, I am willing to use other firmware even if Smoothieware has its limitations such as PT100 support.

DeathmannX, 2017-01-30:
Ouh, i've also got a PT100...
New Reason added. 🌝

Fulg, 2017-01-31:
@TGMods I have a Re-ARM as well and it is an excellent upgrade, indeed without changing your wiring too much, but today it means leaving Marlin behind. I understand this is not what you wanted to hear (nice touch with the downvotes guys!) but it is still true. :)

JustAnother1, 2017-01-31:
@Fulg  I think @bobc and others are currently working on the "leaving Marlin behind" part. So give it some time, or do you see a major reason for not porting Marlin to ARM? 

Feature wise there is a difference between Marlin and Smoothie. So wanting Marlin features on ARM hardware seems to me to be a reasonable request.

Fulg, 2017-01-31:
I am aware of the 32bit porting efforts, but this is "work in progress" and not ready for general release. What @bobc is working on (PR #5357 mentioned at the top) will not work on Re-ARM.

What I said what that **today** Re-ARM means leaving Marlin behind. If you want to stay with Marlin, you have to stay with an 8bit Arduino. This may change in the future but I would not buy new hardware **today** based on plans for future developments, if you want to use that new hardware soon.

I understand that there are good reasons to prefer Marlin over Smoothie, feature-wise Marlin is very much ahead. However firmware is not like most software, you cannot simply recompile it from one platform to another; it takes significant time and effort to support a new target platform.

brainscan, 2017-01-31:
@Fulg maybe it would be helpful to say why it won't work, rather than just saying it won't. Some of us have purchased a re-arm to help with testing and aren't expecting to just use it with Marlin immediately.

Fulg, 2017-01-31:
I am not saying it will never work. I am saying that it is too early to be considered a viable alternative *at this time*. The current porting work still uses the Arduino IDE, which excludes LPC176x boards like Re-ARM. In the future it will likely be possible to support both Arduino and MBED (or add LPC176x support to the Arduino IDE) but this is a non-trivial amount of work that has not been started yet.

If you were using an Arduino Due then you might have better luck to run a 32bit Marlin soon. For now Re-ARM is still a limited-run, closed-source board (no files have been released) so this does not help its case either.

I don't wish to sound negative, I am just stating facts! I would *love* to see Marlin on Re-ARM (and similar platforms), but I am not expecting to see it soon. Eventually yes!

Roxy-3D, 2017-01-31:
> The current porting work still uses the Arduino IDE, which excludes LPC176x boards like Re-ARM. 

It sounds like Bobc is working on it:   https://github.com/MarlinFirmware/Marlin/issues/3851#issuecomment-275672887

JustAnother1, 2017-01-31:
@Fulg If you check bobc's pull request you can see that he has cmake and make support on his list. I wait for that so that I can contribute to the STM32 HAL. With make support for LPC and/or mbed should not be a problem.

I hope that some purchased the re-arm not only for testing but also for developing. Once the HAL API is defined the work on supporting different chips can be done in parallel.  If enough people join re-arm and other platforms might be supported sooner than you think.

brainscan, 2017-01-31:
Saying it won't work does imply that you mean it will never work, or you'd say that it's not working at the moment.

Fulg, 2017-01-31:
@brainscan I believe that is exactly what I said?

> I am not saying it will never work. I am saying that it is too early to be considered a viable alternative *at this time.*

It seems I've been painted into this corner where I am supposedly against Marlin 32bit, but this is not the case. Perhaps I should just stop answering. :)

brainscan, 2017-01-31:
What @bobc is working on (PR #5357 mentioned at the top) will not work on Re-ARM.

This is what you actually said, you then amended it.

psavva, 2017-02-06:
Can we please close this request.

Roxy-3D, 2017-02-06:
> Can we please close this request.

It is important that we get support in place for the Re-ARM board...  But yes, it probably makes sense to close this request for now.

closed 2017-02-06 by Roxy-3D