MarlinFirmware/Marlin #5732: spiral drill moves with G2 ends up in triangular like hole

RedbullVienna, 2017-01-22:
I tried to mill/drill a hole with G2 codes (I using Marlin RC8) but finally it ended up in a hole with triangular like shape.
I simulate the g code with CAMotics and their the hole looks fine. Maybe an issue with the 3d spiral move as it is an hole and not simple a circle with a constant z plane.
I used the following sequence (from EstlCam):
....
G02 Y0.0866 Z-4.5000 I0.0500 J0.0866
G02 X0.1000 Y0.0000 Z-4.6000 I0.0500 J-0.0866
G02 X-0.0500 Y-0.0866 Z-4.7000 I-0.1000 J0.0000
G02 Y0.0866 Z-4.8000 I0.0500 J0.0866
G02 X0.1000 Y0.0000 Z-4.9000 I0.0500 J-0.0866
G02 X-0.0500 Y-0.0866 Z-5.0000 I-0.1000 J0.0000
G02 Y0.0866 Z-5.1000 I0.0500 J0.0866
G02 X0.1000 Y0.0000 Z-5.2000 I0.0500 J-0.0866
G02 X-0.0500 Y-0.0866 Z-5.3000 I-0.1000 J0.0000
G02 Y0.0866 Z-5.4000 I0.0500 J0.0866
G02 X0.1000 Y0.0000 Z-5.5000 I0.0500 J-0.0866
G02 X-0.0500 Y-0.0866 Z-5.6000 I-0.1000 J0.0000
G02 Y0.0866 Z-5.7000 I0.0500 J0.0866
G02 X0.1000 Y0.0000 Z-5.8000 I0.0500 J-0.0866
G02 X-0.0500 Y-0.0866 Z-5.9000 I-0.1000 J0.0000
G02 Y0.0866 Z-6.0000 I0.0500 J0.0866
G02 X0.1000 Y0.0000 Z-6.1000 I0.0500 J-0.0866
G02 X-0.0500 Y-0.0866 Z-6.2000 I-0.1000 J0.0000
G02 Y0.0866 Z-6.3000 I0.0500 J0.0866
G02 X0.1000 Y0.0000 Z-6.4000 I0.0500 J-0.0866
G02 X-0.0500 Y-0.0866 Z-6.5000 I-0.1000 J0.0000
G02 Y0.0866 Z-6.6000 I0.0500 J0.0866
G02 X0.1000 Y0.0000 Z-6.7000 I0.0500 J-0.0866
...

Blue-Marlin, 2017-01-22:
If in sub/mm range

#define MM_PER_ARC_SEGMENT 1
#define N_ARC_CORRECTION 25

and

// Moves (or segments) with fewer steps than this will be joined with the next move
#define MIN_STEPS_PER_SEGMENT 6

become important.

RedbullVienna, 2017-01-22:
THX for the hint!
I have deactivated the G2/3 usage in my CAM SW to came around that issue. 
An automatic dynamic adaptation, e.g. two different values like 0.1mm/1mm for small and big radius, for MM_PER_ARC_SEGMENT maid by a nice feature. 
As soon as I have understood the dependencies with  N_ARC_CORRECTION and MIN_STEPS_PER_SEGMENT I will try to write such a code.

closed 2017-02-15 by thinkyhead