MarlinFirmware/Marlin #5733: Using both Z Dual Endstops and ABL z-probe

tekkydave, 2017-01-22:
I have a Prusa i3 with retractable z-probe on a servo arm which is currently used for both z homing and Linear ABL . I have also recently enabled the **Z_DUAL_STEPPER_DRIVERS** feature which seems to be working fine with 1.1.0-RC8 . The question I have is:
If I also enable the **Z_DUAL_ENDSTOPS** feature and fit the appropriate left & right Z endstops can I still use the retractable z-probe for ABL and it will stay out of the way when z homing is being executed?
Many Thanks.

Bob-the-Kuhn, 2017-01-23:
I believe that using the Z_DUAL_ENDSTOPS feature with a Z_PROBE but not use the Z_PROBE when homing can be accomplished.  Basically you do the Z_PROBE setup & the Z_DUAL_ENDSTOPS setup but **DON'T** enable Z_MIN_PROBE_ENDSTOP.  With Z_MIN_PROBE_ENDSTOP not enabled, G28 will use the Z_DUAL_ENDSTOPS and G29, G30 & M48 will use the probe.

I think I'll try this on my machine.  I wasn't aware of this feature until I started researching this feature.

Setting the physical location of the dual Z_MIN endstops looks like it'll be very tricky.  Marlin always checks the dual Z endstops even when probing.  That means the dual Z endstops have to be below where the Z_PROBE triggers.  The really tricky part comes when doing the print. You'll trigger the Z_MIN endstop at the start of a print if they're not set perfectly.

I was surprised to see that you can't define a Z2_MIN_PIN.  You'll have to dedicate one of the other endstops as the Z2_MIN_PIN.

If you have a Z_MAX_PIN then you'll also need to dedicate one of the other endstops as the Z2_MAX_PIN.  As best I can tell you can have both Z_MAX dual endstops & Z_MIN dual endstops if you  have the xxxx_ENDSTOP_INVERTING flags all set the same way.

As I read the code you'll need to do the following:
- in Configuration.h
-  setup your Z_MIN probe
-  define Z_MIN_PROBE_PIN in your pins_YOUR_BOARD.h file  
-  disable Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN 
- -disable Z_MIN_PROBE_ENDSTOP  

- in Configuration_adv.h
- enable Z_DUAL_ENDSTOPS
- change Z2_USE_ENDSTOP to the desired endstop (_XMAX_, _XMIN_, ...)

I think there's a bug in the latest Marlin RCBugFix.  I got a compiler error indicating that Z2_MIN_ENDSTOP_INVERTING hadn't been defined.  If that happens just add your own define to one of the config files:
#define Z2_MIN_ENDSTOP_INVERTING false

I'll see if I can find anything on setting the location of the Z_MIN sensors.  I'm hoping it's not as nasty as it looks right now.  

Maybe I'll use the Z_MAX option and define Z_HOME_DIR as 1.  This way G28 will first home the Z axis in the positive direction and then do the X & Y axis.  Z_MIN sensors won't be used and setting the Z_MAX sensors should be relatively easy.

tekkydave, 2017-01-23:
Thanks, you answered my question in the first paragraph but the rest is really useful information. I can see there is a lot to consider here.

Bob-the-Kuhn, 2017-01-23:
Another option I've forgotten/missed.  I must have had it enabled early in my build because I definitely made use of endstops in my early stages of learning.

Thanks - this makes thing much easier.

closed 2017-02-07 by tekkydave