MarlinFirmware/Marlin #5734: 1.1.0-rc8 Delta printer - Zprobe

rhinorap, 2017-01-22:
Hya,
Flashed RC8 to my Ramps 1.4 this evening hoping to have bed levelling working using the allen key setup.

No matter what i do it starts to probe nice and swiftly, then slows down it's sampling.

The worst thing is, though that the probe grid is going off the side of the bed and towards the Z tower - despite setting the bed radius.

Suggestions ?

JohnOCFII, 2017-01-22:
@rhinorap I had similar issues with RC8 autolevel behaving weird.  If I defined DELTA_HOME_TO_SAFE_ZONE things seemed to work.  I did have a manually entered negative offset in the Z-access in my bed definition (that worked with RC3) that I had to remove.

After I upgraded to RCBugFix, I no longer had to keep DELTA_HOME_TO_SAFE_ZONE defined.

If you want to read of my issues, and the eventual fix, take a look through #5601 

John

rhinorap, 2017-01-22:
ah, nice one - thanks John.
I've been tinkering and re -enabled Safe_zone - it seems to be working to some extent.

Although i've been getting ZProbe Err now instead now....

rhinorap, 2017-01-22:
Interesting - it now probes but if I re- Enable the EEprom option it shits out again. I can repeat this reliably........

I think the Err:Zprobe condition is because I don't currently have the retract set up....

brainscan, 2017-01-23:
When you switch on eeprom are you updating it or just letting it use the stored values?

rhinorap, 2017-01-23:
This is a bed levelling issue - so i'm not calling on any stored values (at present.).

Roxy-3D, 2017-01-23:
> This is a bed levelling issue - so i'm not calling on any stored values (at present.).

Yes...  But if the EEPROM is enabled you won't pick up new values you put in your Configuration.h file unless you do an M502 / M500.

rhinorap, 2017-01-23:
Ah, yeah - that i've done - sorry - frustration is getting the better of me !

I'm going to re-flash the mega in a while with a script to flush everything and start again - i'm hoping then If i reload it'll sort itself out.

One thing i'm still puzzled about is this error i'm getting when ( with the eeprom disabled) it completes the probing, retracts the allen key, but then reports an error on the LCD of ERR:Zprobe - what am i missing ?

rhinorap, 2017-01-23:
HA ! 
Excellent. I flashed the mega with a basic eeprom wipe code - re-flashed marlin and re-read the eeprom data.

All is now good in my world - I have a Kossel that now homes properly, probes the bed with my manually deployed probe and then parks itself afterwards.

Levelling grid seems to be consistent with a hairline gap across the bed at 0.

Time to print :-p

closed 2017-04-02 by thinkyhead

dtserkanozkan, 2017-08-29:
@rhinorap I am getting the err:ZPROBE message too, at the end of the probing sequence. How did you solve the problem?