MarlinFirmware/Marlin #5736: RC8BugFix - Auto-level - so close I can taste it!

JohnOCFII, 2017-01-22:
After some struggles with G29 probing, I have my Delta configured to probe the bed and create the Bilinear Leveling Grid.

If I use Pronterface and a piece of paper to probe the bed BEFORE autoleveling, my manual level is pretty good.  I can feel resistance on a piece of paper at multiple points around the bed.

If I use Pronterface and check the same locations (center of bed, and about 3 points closer to the towers) AFTER the G29 autolevel, I am already gently grabbing the paper at heights closer to 0.2 to 0.3mm rather than right at 0.0mm.

I'm afraid that if I start a print at that point, I'll be driving into the bed.

Is there some other step I need to complete to get the revised Z-height info to be active?

I followed the instructions here: http://marlinfw.org/docs/features/auto_bed_leveling.html

Related -- I used to use M401 before and M402 just after G29 to manually lower and retract my probe.  I can't see if that is required any more.  I had asked about that on #5600 but received no reply.

Thanks for any advice.  I think I'm close!

John

Roxy-3D, 2017-01-23:
> If I use Pronterface and check the same locations (center of bed, and about 3 points closer to the towers) AFTER the G29 autolevel, I am already gently grabbing the paper at heights closer to 0.2 to 0.3mm rather than right at 0.0mm.

In a perfectly calibrated machine, the nozzle is just barely touching the bed at 0.000 mm.    So shouldn't the nozzle be grabbing the paper at a small height above 0.0000 mm. ?      I think you should press the 'Print'  button!

JohnOCFII, 2017-01-23:
That's the issue -- after running G29, the paper is lightly grabbed at .2mm.  At 0.0, the nozzle is pressed solidly to the bed -- it seems too low after running G29, unless I'm misinterpreting something (which is possible. :) )

brainscan, 2017-01-23:
Sounds like it's set perfectly to print. It should be lightly grabbing the paper at 0.2 and crushing it at 0.0. You don't print at 0.0 or nothing would exit the nozzle. Like Roxy said hit print. If you have baby stepping enabled you can always jog it up a little if it seems too squished.

JohnOCFII, 2017-01-23:
I'll give it a shot later this week, maybe tonight.  Thanks for the input!

brainscan, 2017-01-23:
Good luck

thinkyhead, 2017-04-02:
I'm going to presume you got this working!

closed 2017-04-02 by thinkyhead