MarlinFirmware/Marlin #5738: SCARA fails to link, undefined inverse_kinematics

DouglasPearless, 2017-01-23:
Scenario:
(1) Download latest Arduino for Mac and update boards and processors
(2) Git clone latest Marlin
(3) Compile -> OK
(4) copy the SCARA config files into main Marlin directory.
(5) Compile -> OK
(6) in Configuration.h change
```
// Specify the specific SCARA model
#define MORGAN_SCARA
//#define MAKERARM_SCARA
```
to
```
// Specify the specific SCARA model
//#define MORGAN_SCARA
#define MAKERARM_SCARA
```
(7) Compile -> fails:
```
Arduino: 1.8.1 (Mac OS X), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/douglaspearless/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/douglaspearless/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/douglaspearless/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0042 -ide-version=10801 -build-path /var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445 -warnings=null -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2 -prefs=runtime.tools.avrdude.path=/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8 -prefs=runtime.tools.arduinoOTA.path=/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.0.0 -verbose /Users/douglaspearless/Documents/workspace/Marlin/Marlin/Marlin.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/douglaspearless/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/douglaspearless/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/douglaspearless/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=0X2341_0X0042 -ide-version=10801 -build-path /var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445 -warnings=null -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2 -prefs=runtime.tools.avrdude.path=/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8 -prefs=runtime.tools.arduinoOTA.path=/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/arduinoOTA/1.0.0 -verbose /Users/douglaspearless/Documents/workspace/Marlin/Marlin/Marlin.ino
Using board 'mega' from platform in folder: /Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17
Using core 'arduino' from platform in folder: /Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17
Detecting libraries used...
"/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/variants/mega" "/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/Marlin.ino.cpp" -o "/dev/null"
"/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/variants/mega" "/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/M100_Free_Mem_Chk.cpp" -o "/dev/null"
"/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/variants/mega" "/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/MarlinSerial.cpp" -o "/dev/null"
"/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/variants/mega" "/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/Marlin_main.cpp" -o "/dev/null"
"/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/variants/mega" "/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/Sd2Card.cpp" -o "/dev/null"
"/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/variants/mega" "/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/SdBaseFile.cpp" -o "/dev/null"
"/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/variants/mega" "/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/SdFatUtil.cpp" -o "/dev/null"
"/Users/douglaspearless/Library/Arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/cores/arduino" "-I/Users/douglaspearless/Library/Arduino15/packages/arduino/hardware/avr/1.6.17/variants/mega" "/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/SdFile.cpp" -o "/dev/null"
[... cut ...]
/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/Marlin_main.cpp:1575: undefined reference to `inverse_kinematics(float const*)'
/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T//ccV8KR5w.ltrans2.ltrans.o: In function `Planner::refresh_positioning()':
/var/folders/zf/h23pz4wn2x3f5b3hxlptf15c0000gn/T/arduino_build_537445/sketch/planner.cpp:1439: undefined reference to `inverse_kinematics(float const*)'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

How to fix this??

Cheers
Douglas

Bob-the-Kuhn, 2017-01-25:
This is definitely a bug in Marlin.

"inverse_kinematics" is declared in a section of Marlin_main.cpp that is only included for MORGAN_SCARA.

We'll need to get someone involved that knows what a SCARA printer is.

@thinkyhead , @Roxy-3D - please get this to the proper people.

Bob-the-Kuhn, 2017-01-25:
If I read #5160 correctly the code for MAKERARM_SCARA is not yet released.  

Looking at the website they're still pre-ordering.

thinkyhead, 2017-02-15:
The [MakerArm branch](https://github.com/thinkyhead/Marlin/tree/MakerArmV1) is still a work-in-progress but the kinematics are in working order. It has a somewhat different SCARA implementation than Morgan. We haven't had any developers with a Morgan so it has been hard to make progress in developing the kinematics and debugging the existing code.

closed 2017-02-15 by thinkyhead

rushanthans, 2018-11-23:
hi DouglasPearless Did you find out the code !!! 
You discover share with me, please..