MarlinFirmware/Marlin #5739: Bug: RCBugFix Roxy Edition

timeshell, 2017-01-23:
Hi Roxy.  Using the filament change version that you worked on for us recently, I have just upgraded my NEMA17 from 200 step to 400 step.  It worked fine until I changed the 

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 325.06,325.06,1625.34,837 }

The above values are all doubled from what they were originally with the 200 step motor except for the extruder which is still a 200 step motor.

The X and Y work fine.  However, when trying to home Z the firmware hangs and reboots  (step value 1625.34).  Once on reboot it showed the thermistor having a temperature of 689 degrees until I powered it down and back up again.

Any ideas as to what's going on here?

Bob-the-Kuhn, 2017-01-23:
Does it work if you change the HOMING_FEEDRATE_Z  to 1/4 it's current value?

timeshell, 2017-01-23:
Same thing.  On homing

1.  Homes X/Y
2.  Starts to move Z to SAFEHOME position but stops very quickly and reboots.

timeshell, 2017-01-23:
Seems to work after disabling SAFE_HOMING.

timeshell, 2017-01-23:
But more things seem to be going wrong after doing that.  It isn't accepting commands from Repetier-Host anymore either.  Also G29 kills the printer.

timeshell, 2017-01-23:
Ugh.....  This all worked with the new motors until I reflashed the exact same Marlin with the updated default axis steps.  Now all sorts of things are going haywire.

timeshell, 2017-01-23:
Restoring to:

Bob-the-Kuhn, 2017-01-23:
What does it do if you disable USE_WATCHDOG in Configuration_adv.h?

timeshell, 2017-01-23:
More or less the same thing, except doesn't reboot.  Firmware is hung. Z motors sound like their trying to do something but are locked.

Bob-the-Kuhn, 2017-01-23:
Please attach your config files & I'll see if I can reproduce the problem.

timeshell, 2017-01-23:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/725238/Configuration.zip)

timeshell, 2017-01-23:
[Marlin-RC.zip](https://github.com/MarlinFirmware/Marlin/files/725241/Marlin-RC.zip)

Bob-the-Kuhn, 2017-01-24:
I wasn't able to reproduce your exact problem.

I have dual Y & dual Z enabled on my machine.  When I had either or both of the LCD controllers enabled my Z movements sort of worked.  It didn't matter what the steps per unit were set to.

Try disabling both the ULTIPANEL and the REPRAP_DISCOUNT_SMART_CONTROLLER and see if the problem persists.

What version of U8glib are you using?  I used 1.19.1

Bob-the-Kuhn, 2017-01-24:
I just tried your config with Viki2 enabled.  No problems.

Definitely something funky with ULTIPANEL and the REPRAP_DISCOUNT_SMART_CONTROLLER.

timeshell, 2017-01-24:
I did mention the reboot when SAFE_HOMING.  May not have mentioned the very LONG BEEEEP when I disabled the watchdog.  Sure it's related to the LCD?  It all works fine when I restore the original axis steps.

Bob-the-Kuhn, 2017-01-24:
I'm not sure enabling the REPRAP_DISCOUNT_SMART_CONTROLLER has anything to do with what you're seeing.

If you only move the Z axis - does it work properly?

Since my Z axis movement was screwed up when I had REPRAP_DISCOUNT_SMART_CONTROLLER enabled, I never tried the G28 command.

timeshell, 2017-01-24:
I can move all the axis without any issue.  The problems start occurring when doing a g28.

Bob-the-Kuhn, 2017-01-24:
I think I've got something close.  The problem is that it doesn't appear to be dependent on the steps per unit.

Here's what my system is doing now:

- Do some X, Y & Z movement commands - works as expected
- Issue a G28 command

1. X homes properly
2. Y homes properly
3. Z moves to the probe position
4. Z raises
5. At what appears to be when it should start down my system mostly freezes and issues a fatal error message. If I issue a M999 command it just comes back with fatal error.  Can't do any moves.  Only way out is to disconnect.

The one thing I did notice was my extruder temperature was 0.1C with this configuration instead of the usual 24C.  Using my config it says 24C for the extruder and G28 works.

It's 1AM here so I'll look at it again tomorrow afternoon.

Bob-the-Kuhn, 2017-01-24:
False alarm - had the wrong pins_xxxx.h file.

I've got a couple more things to try tomorrow.

lrpirlet, 2017-01-24:
@timeshell 
Well, G28 does load the EEPROM... You enabled EEPROM, Do you have coherent EEPROM content?

wywywywy, 2017-01-24:
If you enable DEBUG_LEVELING_FEATURE it'll pretty much tell you where it goes wrong.

timeshell, 2017-01-24:
@Bob-the-Kuhn Tjat sounds similar to mine.  Except mine fails just as your step 3 starts and doesn't get to the Probe position before producing the fatal error.

Bob-the-Kuhn, 2017-01-24:
I can't reproduce your problem.

Where do we go from here?  Try the following:
- Does a G0 X10 Y10 command do anything strange?  Maybe it's the simultaneous movement that it's having a problem with.
- Enable DEBUG_LEVELING_FEATURE in configuration.h and then issue a M111 S32 command before doing the G28.  With any luck it'll report something unusual.
- Disable your LCD defines (and babystepping).  Going from 800 to 1600 steps per unit requires one more digit in the display string.  Wouldn't be the first time a string was too long for a buffer.
- Disable the EEPROM.  My understanding is it overrides the settings in the config files.  Maybe there's something strange in there,
- Go to the latest RCBugFix.

If the above doesn't help then we'll see if one of the Marlin support people will step in.

timeshell, 2017-01-24:
I don't think it has anything to do with the 800 to 1600 steps.  I did try just changing that back to 800 and still had the same issue with it going into error on X/Y center for Z probing.  So it seems somehow related to the X/Y movement, maybe together as you stated.  Although, it does work fine with the original X/Y step values.  I have put 1/16 steppers in my machine for now and it is printing happily with them.

I'll try what you suggested.  I do have a vague recollection that someone tried this a couple years ago and it was determined that the 1/32 steppers with 400 step motors was just too much for Marlin to calculate.  (maybe I'm mistaken)

Bob-the-Kuhn, 2017-01-24:
I'm glad you're up & running.

What did you change?  Sounds like you set the jumpers for your driver to 16 microsteps and then changed the steps per unit to something under 800.

I don't think the steps per unit itself is a problem.  My printer was able to run the 1600 steps per unit.

There is one major difference between our printers.  Yours is about 10x faster than mine.  Maybe the limitation is doing the steps fast enough.  Marlin caps the step rate at 40,000 per second.  The resulting max feed rates are:
- 1625.34 steps/mm	24.6mm/s 	1476mm/minute
- 325.06   steps/mm	123mm/s 	7383mm/minute
- 162.53   steps/mm	246mm/s 	14766mm/minute

In your config you have homing speeds of 3,000mm/minute for X/Y & 240mm/minute for Z.  Looks like feedrates are not the issue with G28.

I doubt that the processor speed is getting in the way of the G28 performance.  It only needs to do a few relatively long moves and delays in starting a move won't be noticeable.  I can see it as a possibility when printing.  That usually requires lots of small moves.

I don't have a good theory as to what could be the problem.

timeshell, 2017-01-25:
I didn't change any jumpers.  I took out my DRV8825's and replaced them with A4988's so I could reduce the axis steps to what they were before I upgraded my motors from 200 steps to 400 steps.  Essentially, my original configuration which always worked.  However, my goal was to use the DRV8825's with the 400 step motors which it appears I am unable to do at present due to some limitation in Marlin with that configuration.

Recall I disabled SAFE_HOMING at which point I was able to finish the G28.  However, other things went really wrong afterward as logged in a previous comment.

> But more things seem to be going wrong after doing that. It isn't accepting commands from Repetier-Host anymore either. Also G29 kills the printer.

So, I guess more debugging is required which I'll try to do more of later.  I have other things that are going to take priority since I have a work around for now.

thinkyhead, 2017-04-02:
> when trying to home Z the firmware hangs and reboots

Usually this is due to a too high Z rate, especially after increasing the number of steps-per-unit. The CPU simply can't keep up and the watchdog ends up getting triggered, rebooting the machine.

VanessaE, 2017-04-30:
This bug report seems like the best place to put this, sorry if it's offtopic:

"Reboots" aren't the only thing that can happen.  If you drive E faster than 40ks/sec, Marlin will throw bogus temperature-related errors, either "cold extrusion prevented", or having the bed and hotend temperatures drop briefly and then snap back to normal (without any cool-down/warm-up time), or even kill the print entirely with a MINTEMP error (on the bed, of all things).

In my case, I have E set to 760, and was playing around with retraction testing at 75 mm/sec, which I later realized would be 57ks/sec.  Obviously too fast.  :smile:   Once I slowed it down to 50 mm/sec (given a max of 40k/760 = ~52.632), everything was perfectly fine.

Marlin should probably throw a warning at compile time if the an axis is configured with a max speed and steps/mm that together would exceed 40k.

Roxy-3D, 2017-04-30:
> "Reboots" aren't the only thing that can happen. If you drive E faster than 40ks/sec, Marlin will throw bogus temperature-related errors, either "cold extrusion prevented" (but the print keeps on going), or having the bed and hotend temperatures drop sharply and then snap back to normal after a brief period, without any cool-down/warm-up slope, or even kill the print entirely with a MINTEMP error (on the bed, of all things). Or at least, that's what it does at commit 922c67f.

This sounds a lot like the data corruption problem we had with the compiler inlining functions and setting up too large of a stack frame in UBL's G29.    It would be very helpful to turn on and initialize the M100 Free Memory Watcher and see if you can cause the problem.  Because if the stack is growing into the heap...  You start seeing ridiculous temperatures showing up on the LCD Panel.

VanessaE, 2017-05-01:
I'll try to recreate the problem after I've finished re-calibrating the printer (hardware upgrade).

thinkyhead, 2017-05-04:
@VanessaE We're certainly familiar with both issues. Either the watchdog doesn't get reset, because the main loop is curtailed, or the temperature readings get screwy because the temperature ISR doesn't get enough time. I would be very interested in figuring out a way to pre-calculate the CPU requirements for some given max steps-per-second and sanity-check that at compile time. As more people move to 32x micro-stepping this issue keeps becoming more common.

I vaguely recall that the stepper ISR might have had in the past some code which could detect when it was being run at too high a speed and throttle down. But I sense that any runtime-based solution will only add more overhead. If we can give at least a warning at compile-time then we have at least some indemnity when issues do arise.

VanessaE, 2017-05-04:
In that case, I guess I *don't* need to reproduce the glitch. :smile:

klcjr89, 2017-08-09:
Bumping this issue up, please add it in the Sanity check because I too tried to use DRV8825 drivers with the same bottlenecks, made the machine unusable so now I'm using TMC2100s.

fiveangle, 2017-08-09:
@klcjr89 - it is not possible to guestimate that since it depends on a lot of factors and what other features are enabled so including such a check in SanityCheck.h would be a half solution.  Enabling Linear Advance for example taxes things considerably, but running no bed leveling or other features and a printer can scream.  Although this does give me an idea...

@Bob-the-Kuhn - can you close this thread ?

klcjr89, 2017-08-09:
UBL with DRV8825s in 1/32 is basically unusable, and there is no documentation to explain to people what is actually doable wih the hardware; would be a welcome adddition. Some of us could choose a Duet, but we are so used to and love Marlin.

fiveangle, 2017-08-09:
If you have some useful input on this, I'm sure your PR submission to the wiki would be appreciated: https://github.com/MarlinFirmware/MarlinDocumentation

Roxy-3D, 2017-08-09:
> UBL with DRV8825s in 1/32 is basically unusable

UBL uses the same probe, motion and endstop system that everything else uses.   So what is unusable about it?     AND...   with UBL not turned on, can you use your DRV8825's ?   My guess would be no.

fiveangle, 2017-08-09:
They're alluding that the 8-bit Atmega MCU doesn't have enough juice to run at the level of steps/mm required.  Again, no one is doubting that may be true on this person's system, but to claim that it is "unusable" on every system is quite presumptuous.  Marlin can run at 40mm/s on a cartesian system with 120 microsteps/mm@16 microsteps/step on each axis so someone absolutely can run on a bot that has 60microsteps/mm@32microsteps/step, so it's hard to put much seriousness into the outlandish claim.

Does the available MCU power need to be considered when implementing a 'bot with Marlin with an 8-bit MCU ? Yes.  Are 32 microstepping stepper drivers "unusable" on Marlin ? Uh... no :)

Roxy-3D, 2017-08-09:
Well....   Soon... maybe even the 128 micro-stepping drivers can be run at a high feed rate with the 32-bit processors.

VanessaE, 2017-08-09:
All of this makes me wonder - would it be possible to have the ISR go into a sort of 8x stepping, to complement the 2x and 4x profiles it uses when needed?  Obviously this would only be useful for making super-fast retractions with geared extruders and maybe travel moves (assuming the endpoint is still accurate).  Seems like it might be useful for some folks.

fiveangle, 2017-08-10:
It's certainly possible, but only useful for boards that have the stepper config pins on their drivers tied to controllable pins on the MCU.  That's an extremely small minority of printers, which is surely why no one has tackled it.  Given the limited usefulness of it, time is probably better spent at this point on porting to 32-bit platforms :)

VanessaE, 2017-08-10:
Er, I meant the mode where the ISR steps twice or 4 times per interrupt, not a change in hardware. :smile:

Roxy-3D, 2017-08-10:
The Stepper Motor code is an area I've hardly done anything in.   That is an interesting idea.   But isn't it the Max Feed Rate numbers that limit what values the Stepper chips see?   Having more micro steps causes the feed rate numbers to be higher.   But if the max feed rate (and acceleration) number is high enough, won't even 1x go at a good speed?

VanessaE, 2017-08-10:
It would, I suppose.   However I was thinking more along the lines of an older post of mine in this thread, where I mentioned accidentally exceeding Marlin's limit of 40'000 steps per second (e.g. its 10'000 ISR calls per second, doing 4 step pulses per call).  On my particular machine, that leads to over 50 mm/sec, which is plenty.

On the other hand, one guy I know had a little NEMA-11 motor with a 27:1 gearbox attached to it for his extruder at one time, with a functional limit of less than 15 mm/sec (I don't recall the exact, functional speed he was able to get).  At that speed, the long retract lengths needed for his bowden setup led to sizable retraction times.

Because of its lower mass, I imagine there are others who have tried (or actively use) the same sort of setup.  An 8x mode wouldn't be of any use to me, but it would put the above sort of setup into roughly the same speed class as a more common setup with NEMA-17 and Wade's gears.

closed 2018-02-06 by thinkyhead