MarlinFirmware/Marlin #5740: [FR] Startup script

radry, 2017-01-23:
It would be nice to have a way to automatically run a gcode script after Marlin booted.
Alternatively a way to select predefined scripts over the LCD.

For example I have a belted Z axis which the stepper has to be activated and the nozzle raised slightly before homing because else the nozzle would drag over the bed initially. It is tedious to do that manually with the LCD every time.

Tannoo, 2017-01-23:
Z_HOMING_HEIGHT is already there. You can change the height you need there.

You can create your startup script as a gcode file and place it on an SD card. Run that before anything else.

radry, 2017-01-23:
Does Z_HOMING_HEIGHT raise the Z axis before homing or just prevent homing if it's less than the specified value?

Roxy-3D, 2017-01-23:
> Does Z_HOMING_HEIGHT raise the Z axis before homing or just prevent homing if it's less than the specified value?

Z_HOMING_HEIGHT is used to define the minimum nozzle height when homing.  Usually, when homing you don't know your current height (because the printer could have just been turned on) so you will see a one time raise of that height.   Subsequent homing operations will know the nozzle is at a safe height and not repeat this move.

Tannoo, 2017-01-23:
Well, mine raises Z before homing any axis on the first G28 after startup. Any additional G28s...X and Y home before Z moves at all.

Tannoo, 2017-01-23:
It should be every time, but it's only the first time after powerup or upon connection with a host.

Bob-the-Kuhn, 2017-01-30:
Take a look at [Autostart](https://github.com/MarlinFirmware/Marlin/wiki/Autostart)

It says that if you name a g-code file correctly it'll be automatically executed on startup.

radry, 2017-01-30:
Thank you, that's what I was looking for.

radry, 2017-02-01:
@Bob-the-Kuhn 
Tried it but didn't work. Even manually selecting auto0.g from "print from SD" does nothing. The file containes:  
`G1 Z5`

Bob-the-Kuhn, 2017-02-01:
I got it to work by replacing the line "card.checkautostart(false);" with "card.checkautostart(true);" in Marlin_main.cpp.

I did the same G1 Z5 in my AUTO0.G file.  It went up by 5mm every time the card **powered up**.  It doesn't do it if you just disconnect & then connect.  

There's also some comments about adding an autoload menu item to the LCD.  I can run that down if you want.

boelle, 2019-02-18:
@radry 

Please post your question either on discord: https://discord.gg/n5NJ59y or on facebook: https://www.facebook.com/groups/2080308602206119/ 
The issue list is for bugs and feature requests only
Please close this issue once you have posted it on one of the 2 links
Thanks :-D

radry, 2019-02-19:
@boelle 

> The issue list is for bugs and feature requests only

By title this IS a feature request.

boelle, 2019-03-12:
@thinkyhead this one has the wrong label. should be feature request label and not question label

boelle, 2019-06-22:
need correct label on this one

boelle, 2019-08-22:
i just discovered this new section (for me its new)

/**
 * Startup commands
 *
 * Execute certain G-code commands immediately after power-on.
 */
//#define STARTUP_COMMANDS "M17 Z"

in configuration_adv.h

would this not do the trick?

Vertabreak, 2020-01-19:
@boelle based on the OP i think the new startup commands option would fill this request.

closed 2020-01-24 by boelle

borland1, 2020-05-17:
I would like to insert musical tones on power up using STARTUP_COMMANDS.  The sample example in Configuration_adv.h, only provides one g-code ( "M17 Z" ) inside quotes.  How are multiple g-codes formatted? In quotes, separated by comma?  Or like this?

#define STARTUP_COMMANDS "M300 S440 P200 M300 S660 P250 M300 S880 P300"

Is the comma ignored or required?  Thanks

Startup commands
Execute certain G-code commands immediately after power-on.
*/
//#define STARTUP_COMMANDS "M17 Z"