MarlinFirmware/Marlin #5741: Marlin 1.1.0-RC8 auto bed leveling G28 travelling out of bed and crashing

stiivik, 2017-01-23:
Hi

I am using MarlinRC8, ramps 1.4 board and Anet A8 printer. If I use G28 X Y everything is fine and working and if i use G28 it travels out of bed. Settings should be correct, but apparently they are not as this is crashing beyong max position on X and Y axis. I can't find the mistake. I enabled debugging and logged both commands, I will also post Configuration file. Using the same settings on old Marlin 1.0.2-1 and it's working correct. Is there something I missed or what could it be?

Thanks!

[G28 X Y.txt](https://github.com/MarlinFirmware/Marlin/files/725046/G28.X.Y.txt)
[G28.txt](https://github.com/MarlinFirmware/Marlin/files/725045/G28.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/725047/Configuration.txt)

FHeilmann, 2017-01-23:
When you do G28, when exactly is it going out of bed?

stiivik, 2017-01-23:
Climbed out of bed to test it and it works perfectly. Tried G29 after it and works. Thank You!

closed 2017-01-23 by stiivik